intel/compute-runtime #5: Cmake failure

raunraun, 2018-02-09:
I'm having trouble following the build instructions.

First.  Both the compute-runtime and the graphics-compiler both call for identical workspaces and say to create a 'build' directory for cmake.  Both projects can't share the same build dir.  

- Am I expected to have 2 separate workspaces, 1 for the runtime and 1 for the compiler?
- Should I have 2 build dir's in a single workspace, 1 for the runtime and 1 more the compiler?

<BR><BR>

Second.   GoogleMock has been merged into GoogleTest several years ago.  The CMakeList.txt for the runtime still seems to assume gtest and gmock are separate projects.

- Should I be using an old gmock and gtest from when they where seperate projects?

<BR><BR>

Third.  Related to the second, the runtime has several dependencies.  Are there known good labels to use when building the workspace?

ArturHarasimiuk, 2018-02-12:
Thanks for your feedback. We are working to improve our build instructions.

intel-graphics-compiler is part of compute-runtime workspace. You don't need to create separate workspace for compiler when working with compute-runtime.

You can use below commands to create proper workspace:
```
git clone -b release_40 https://github.com/llvm-mirror/clang clang_source
git clone https://github.com/intel/opencl-clang common_clang
git clone https://github.com/intel/llvm-patches llvm_patches
git clone -b release_40 https://github.com/llvm-mirror/llvm llvm_source
git clone https://github.com/intel/gmmlib gmmlib
git clone -b release-1.7.0 https://github.com/google/googlemock gmock
git clone -b release-1.7.0 https://github.com/google/googletest gtest
git clone https://github.com/intel/intel-graphics-compiler igc
git clone https://github.com/KhronosGroup/OpenCL-Headers khronos
git clone https://anongit.freedesktop.org/git/mesa/drm.git libdrm
git clone https://github.com/intel/compute-runtime neo
```

For Google Test and Google Mock framework. Yes, you should use older version: 1.7.0 is expected.

raunraun, 2018-02-12:
These instructions look great - much clearer.  I'll  give it another try.

MichalMrozek, 2018-02-14:
Instructions are now updated, looks like this question may be closed.

closed 2018-02-14 by MichalMrozek