intel/compute-runtime #6: Build failure on gcc 7.2.1

raunraun, 2018-02-09:
```
/home/raun/workspace/neo/runtime/mem_obj/image.cpp: In static member function ‘static OCLRT::Image* OCLRT::Image::create(OCLRT::Context*, cl_mem_flags, const OCLRT::SurfaceFormatInfo*, const cl_image_desc*, const void*, cl_int&)’:
/home/raun/workspace/neo/runtime/mem_obj/image.cpp:146:24: error: this statement may fall through [-Werror=implicit-fallthrough=]
             imageDepth = imageDesc->image_depth;
             ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
/home/raun/workspace/neo/runtime/mem_obj/image.cpp:147:9: note: here
         case CL_MEM_OBJECT_IMAGE2D:

```

Maybe have a way to disable warnings as errors?

MichalMrozek, 2018-02-09:
Thanks for report.
This will be fixed.

MichalMrozek, 2018-02-14:
Problem fixed.

closed 2018-02-14 by MichalMrozek