intel/compute-runtime #7: Unit tests fail to build on gcc 7.2.1

raunraun, 2018-02-09:
```
In file included from /home/raun/workspace/neo/unit_tests/built_ins/built_in_tests.cpp:24:0:
/home/raun/workspace/neo/unit_tests/gen_common/test.h:87:62: error: could not convert ‘testing::internal::GetTypeId<BuiltInTests>()’ from ‘testing::internal::TypeId {aka const void*}’ to ‘testing::internal::CodeLocation’
                  ::testing::internal::GetTypeId<test_fixture>())
                                                              ^
/home/raun/workspace/neo/unit_tests/gen_common/test.h:77:14: note: in definition of macro ‘HWTEST_TEST_’
             (parent_id),                                                                       \
              ^~~~~~~~~

```

ArturHarasimiuk, 2018-02-12:
Which version of Google Test framework are you using? 

Please note that 1.7.0 is expected. Please try again with 1.7.0 if you used different version and let us know if it worked.

Thank you!

raunraun, 2018-02-12:
I was on TOT, rolling back now.  Thanks!

closed 2018-02-12 by MichalMrozek