intel/compute-runtime #8: cloc_tests segfaults

raunraun, 2018-02-13:
After a build cloc_tests segfaults.  It looks like some files are not copied into the bin director.   All other tests run and pass.

Log below:

```
[raun@localhost bin]$ ./cloc_tests
Iteration: 1
Error: Input file test_files/copybuffer.cl missing.

/home/raun/workspace/neo/unit_tests/offline_compiler/offline_compiler_tests.cpp:59: Failure
Expected: (nullptr) != (pOfflineCompiler), actual: 8-byte object <00-00 00-00 00-00 00-00> vs NULL

/home/raun/workspace/neo/unit_tests/offline_compiler/offline_compiler_tests.cpp:60: Failure
Value of: retVal
  Actual: -5151
Expected: 0
[  FAILED  ] OfflineCompilerTests.GoodArgTest
Error: Input file test_files/copybuffer.cl missing.

/home/raun/workspace/neo/unit_tests/offline_compiler/offline_compiler_tests.cpp:77: Failure
Value of: internalOptions
Expected: has substring "cl_khr_3d_image_writes"
  Actual: ""
[  FAILED  ] OfflineCompilerTests.TestExtensions
Error: Input file test_files/copybuffer.cl missing.

/home/raun/workspace/neo/unit_tests/offline_compiler/offline_compiler_tests.cpp:90: Failure
Expected: (nullptr) != (pOfflineCompiler), actual: 8-byte object <00-00 00-00 00-00 00-00> vs NULL

/home/raun/workspace/neo/unit_tests/offline_compiler/offline_compiler_tests.cpp:91: Failure
Value of: retVal
  Actual: -5151
Expected: 0
Segmentation fault (core dumped)
[raun@localhost bin]$

```

ArturHarasimiuk, 2018-02-13:
we shouldn't observe this:

> Error: Input file test_files/copybuffer.cl missing.

Cmake should generate appropriate commands to copy required files. Could you please check if file copybuffer.cl is present somewhere under bin directory?
   
   
It will be helpful if you can execute below command and share output:
```
make run_cloc_tests
```

raunraun, 2018-02-14:
make run_cloc_tests works fine.

copybuffer.cl appears in a few places:

> [raun@localhost bin]$ find ./ -name "*copybuffer.cl"
> ./bxt/test_files/copybuffer.cl
> ./kbl/test_files/copybuffer.cl
> ./bdw/test_files/copybuffer.cl
> ./glk/test_files/copybuffer.cl
> ./skl/test_files/copybuffer.cl
> ./cfl/test_files/copybuffer.cl
> [raun@localhost bin]$

MichalMrozek, 2018-02-15:
Thanks for the report, in the current implementation cloc_tests needs to be executed in the directory that is being tested.
If you want to test SKL , then cloc_tests need to be executed under SKL directory.
This is different then the other tests, which may be executed directly in the same directory where they are.

MichalMrozek, 2018-02-22:
Problem fixed with commit 46f3efd79c45b05177a919f70548bbba5f4cd40f.

closed 2018-02-22 by MichalMrozek