intel/compute-runtime #10: segfault in  OCLRT::DrmMemoryManager::allocUserptr when not logged in graphically

raunraun, 2018-02-14:
I'm getting an segfault in allocUserptr when I'm not logged to a graphical Xorg session.  After a fresh reboot, GDM is running but I'm not logged in graphically yet.  I then ssh into the machine and this happens:

```
[raun@localhost test]$ LD_LIBRARY_PATH=. gdb clinfo
GNU gdb (GDB) Fedora 8.0.1-35.fc27

(gdb) r
Starting program: /home/raun/test/clinfo
Missing separate debuginfos, use: dnf debuginfo-install glibc-2.26-24.fc27.x86_64
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
KHR ICD trace at /home/raun/OpenCL-ICD-Loader/icd.c:68: attempting to add vendor /home/raun/workspace/build/bin/libigdrcl.so...

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff690cd19 in OCLRT::DrmMemoryManager::allocUserptr(unsigned long, unsigned long, unsigned long, bool) ()
   from /home/raun/workspace/build/bin/libigdrcl.so
Missing separate debuginfos, use: dnf debuginfo-install libgcc-7.2.1-2.fc27.x86_64 libstdc++-7.2.1-2.fc27.x86_6
```

If I log graphically the segfault when running OpenCL applications my ssh session no longer happens.  So the sequence is:

- reboot
- ssh to machine
- run clinfo
  - get above error
- log in graphically to the physical machine
- return to ssh session
  - clinfo runs perfectly.

Being able to run OpenCL applications remotely without Xorg running would be very nice to have.

raunraun, 2018-02-14:
My display is powered by a PEG card.  IGD is enabled, but is headless.

ArturHarasimiuk, 2018-02-14:
could you please share output from command:
```
lspci -nnn | grep -i vga
```

raunraun, 2018-02-15:
```
[raun@localhost test]$ lspci -nnn | grep -i vga
01:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Cape Verde XT [Radeon HD 7770/8760 / R7 250X] [1002:683d]
[raun@localhost test]$ lspci -nnn | grep -i disp
00:02.0 Display controller [0380]: Intel Corporation HD Graphics 530 [8086:1912] (rev 06)

```

ArturHarasimiuk, 2018-02-16:
Thanks for additional data. We analyzed problem and should have fixes very soon.

ArturHarasimiuk, 2018-02-19:
as of commit a99d951c55a1964d69e1d095d09dfb645af02a7b the SEGV issue should be gone.

please note there are some OS level requirements:
- i915 module have to be loaded
- user needs permission to open /dev/renderD<XXX> or /dev/card<X>

please check updated version of Neo and let us know if it works for you. In case of problems, please make sure that you meet requirements.

Two additional commands that could help to debug in case of problems:
```
# dmesg | grep drm

# dmesg | grep i915
```

MichalMrozek, 2018-02-23:
Hello @raunraun , can you check if problem is fixed?

raunraun, 2018-02-24:
I'll give it a shot this weekend.  Thanks!

raunraun, 2018-02-26:
looks good!  Thanks!

MichalMrozek, 2018-02-26:
Thanks for checking, glad to hear that is working :)

closed 2018-02-26 by MichalMrozek