intel/compute-runtime #9: clGetPlatformIDs fails with the included Fedora 27 ICD

raunraun, 2018-02-14:
Using the ICD included in Fedora 27 I get the following error from NEO.  Beignet, POCL, and AMDGPU Mesa implements all work fine with this ICD

```
[raun@localhost test]$ gdb clinfo
GNU gdb (GDB) Fedora 8.0.1-35.fc27

(gdb) r
Starting program: /home/raun/test/clinfo
Missing separate debuginfos, use: dnf debuginfo-install glibc-2.26-24.fc27.x86_64
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
terminate called after throwing an instance of 'cl::Error'
  what():  clGetPlatformIDs

Program received signal SIGABRT, Aborted.
0x00007ffff6c7266b in raise () from /lib64/libc.so.6
Missing separate debuginfos, use: dnf debuginfo-install libgcc-7.2.1-2.fc27.x86_64 libstdc++-7.2.1-2.fc27.x86_64 ocl-icd-2.2.11-4.fc27.x86_64
(gdb) info sharedlibrary
From                To                  Syms Read   Shared Object Library
0x00007ffff7dd5d50  0x00007ffff7df4c50  Yes (*)     /lib64/ld-linux-x86-64.so.2
0x00007ffff7bcf1a0  0x00007ffff7bd26b6  Yes (*)     /lib64/librt.so.1
0x00007ffff787f880  0x00007ffff7928985  Yes (*)     /lib64/libm.so.6
0x00007ffff7674e50  0x00007ffff7675b7e  Yes (*)     /lib64/libdl.so.2
0x00007ffff7458940  0x00007ffff746a848  Yes (*)     /lib64/libOpenCL.so.1
0x00007ffff723fac0  0x00007ffff724fde5  Yes (*)     /lib64/libgcc_s.so.1
0x00007ffff7023b10  0x00007ffff70322d1  Yes (*)     /lib64/libpthread.so.0
0x00007ffff6c5b770  0x00007ffff6dc816c  Yes (*)     /lib64/libc.so.6
0x00007ffff67f7450  0x00007ffff693008a  Yes         /home/raun/workspace/build/bin/libigdrcl.so
0x00007ffff64ea100  0x00007ffff659a7f8  Yes (*)     /lib64/libstdc++.so.6
(*): Shared library is missing debugging information.
(gdb)

```
clGetPlatformIds is returning -1001.  You can see from the gdb log that libigdrcl.so was dynamically loaded by the ICD.

If I build my own ICD from TOT khronos clinfo executes just fine.

MichalMrozek, 2018-02-14:
This is known limitation of our current driver.

The driver requires Khronos ICD loader to operate correctly: https://github.com/KhronosGroup/OpenCL-ICD-Loader

We are exploring other solutions to make it more convenient with ICD provided with Linux distributions.

JacekDanecki, 2018-02-15:
Can you provide backtrace?
When you set OCL_ICD_ASSUME_ICD_EXTENSION=1 environment variable do you still observe this issue?
Are you using clinfo from this repository: https://github.com/Oblomov/clinfo

raunraun, 2018-02-15:
I'm using clinfo from the Fedora 27 repo:
```
[raun@localhost test]$ dnf search clinfo
Last metadata expiration check: 12 days, 10:27:04 ago on Fri 02 Feb 2018 09:47:42 PM CST.
========================= Name Exactly Matched: clinfo ===================
clinfo.x86_64 : Enumerate OpenCL platforms and devices
```

...seems to work
```
[raun@localhost test]$ OCL_ICD_ASSUME_ICD_EXTENSION=1 clinfo
Number of platforms:                             1
  Platform Profile:                              FULL_PROFILE
  Platform Version:                              OpenCL 2.1
  Platform Name:                                 Intel(R) OpenCL HD Graphics
  Platform Vendor:                               Intel(R) Corporation
  Platform Extensions:                           cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_fp16 cl_khr_depth_images cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_icd cl_khr_image2d_from_buffer cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_intel_subgroups cl_intel_required_subgroup_size cl_intel_subgroups_short cl_khr_spir cl_intel_accelerator cl_intel_media_block_io cl_intel_driver_diagnostics cl_intel_device_side_avc_motion_estimation cl_khr_priority_hints cl_khr_subgroups cl_khr_il_program cl_khr_fp64 cl_intel_planar_yuv cl_intel_packed_yuv cl_intel_motion_estimation cl_intel_advanced_motion_estimation

  Platform Name:                                 Intel(R) OpenCL HD Gr......

```

raunraun, 2018-02-15:
Yes, that's the clinfo.  I'm using the bin included in Fedora 27

ArturHarasimiuk, 2018-02-26:
starting from commit c50054f787c89d1bbae5dc65d5553a5a40cb996d, the ocl-icd support should work natively, without additional flags.

@raunraun could you please check with version c50054f787c89d1bbae5dc65d5553a5a40cb996d or newer?

lissyx, 2018-03-02:
@ArturHarasimiuk As documented in issue #20, this works for me on Ubuntu 17.10, thanks for fixing that !

MichalMrozek, 2018-03-02:
Thanks for checking this, I am closing this issue.
@raunraun  feel free to reopen if problem is still visible.

closed 2018-03-02 by MichalMrozek

TaihuLight, 2018-12-06:
@MichalMrozek @JacekDanecki When I install NEO driver, I can not find ICD loader library 'libOpenCL.so' in my install directory of NEO, and NEO document says that link to it. 
How to generate libOpenCL.so for NEO driver?

alalek, 2018-12-06:
@TaihuLight Try to use libOpenCL.so from the system directly.

Fedora [package](https://fedora.pkgs.org/29/fedora-x86_64/ocl-icd-2.2.12-2.fc29.x86_64.rpm.html):
> dnf install ocl-icd

Ubuntu [package](https://packages.ubuntu.com/xenial/ocl-icd-libopencl1):
> apt install ocl-icd-libopencl1

If provided version is not enough to you then try to build Khronos ICD loader from the Khronos repository (it may conflicts with system ones). The link is above.