intel/compute-runtime #12: Android support (aka Android-IA) for Neo driver?

oscarbg, 2018-02-17:
Hi,
I say that because altough I newer owned an Android device with x86 processor either tablet or phone..
seems Intel provided OpenCL support on these devices..
even I don't remember if Intel ships opencl library for Android and info on linking with Intel Compiler for Android in some Intel OpenCL SDK..
anyway I believe this devices used similar closed source binary components similar to say Windows driver at the time..
anyway now Android-IA efforts seem to have moved towards open source stack using Mesa as graphics driver component of Android-IA..
as Mesa has CLOVER OpenCL driver which is very "unadvanced" to say it cleanly, Intel users have beignet option and has had support for Android at least for a couple of years:
https://cgit.freedesktop.org/beignet/log/?qt=grep&q=android
https://cgit.freedesktop.org/beignet/commit/?id=be0ae741a9064cebf5b5c9a277ae895086bdbddd
"docs/howto/android-build-howto.mdwn"
In the case I'm interested is on Android x86 ISOs for PC/laptops we have modern releases like:
http://www.android-x86.org/releases/releasenote-7-1-r1
which using Mesa and VAAPI expose OpenGL ES 3.2 and HW accelerated video decode (encode?) on computers with modern Intel HD graphics running this Android-x86 ISOs..
lastly if not enough  very recently have started appearing builds with Vulkan support also.. using Mesa ANVIL driver.. support is very early for Vulkan only VulkanCapsViewer works.. graphical demos or apps with Vulkan support like PPSSPP emulator still have issues..
this builds are Android 8.1 (Oreo) experimental builds provided by @maurossi here:
https://drive.google.com/drive/folders/0B_OFHiIqgpSFTFpkQWc1eXV3ME0
which ship with latest kernel 4.16rc1+Mesa18.1dev using LLVM6.0 for amdgpu DC..
sadly still have yet to see any of this ISOs to integrate beignet OpenCL driver as, altough Google doesn't officially ship/endorse OpenCL on his phones, many (ARM64) devices ship with modern OpenCL drivers..
for example Xiaomi with modern Qualcomm SOCs and say Adreno 5xx have OpenCL2.0 support)..
of course there are also apps on the Play store so Google isn't blocking them either (for example Geekbench Compute non free version has OpenCL support similar to Compubench CL app)..

sorry for long story..
all it was done to motivate Neo devs to include also Android-IA (Android-x86 or call you what you want) support for this new driver and a possible build guide to integrate this NEO OpenCL driver on the standard Android-x86 ISO build process so some ISOs can start appearing supporting:
OpenGL ES3.2+Vulkan+OpenCL2.x..
basically all APIs..

also not related to you but some Intel Android devs or Intel Mesa dev should try to enable full OpenGL on Android..
Nvidia does this already on his Android devices Shield tablet and TV so not asking something from other world..
of course full OpenGL support it's provided already by Intel Mesa driver for Android-IA and that should be perfect for new Wine for Android project (https://dl.winehq.org/wine-builds/android/):

extracted from private communication with Mauro Rossi:

```
more info on full GL on Android:
(by nvidia devtech)
http://jamesdolan.blogspot.com.es/2014/06/opengl-44-and-beyond-on-android.html
also seems Mesa full OpenGL support for Android was removed a while ago:
(egl: treat EGL_OPENGL_API as invalid on Android)
https://github.com/intel/external-mesa/commit/7563c39641d07a604f22a9d47861121ce80116f9
it says " /* OpenGL is not a valid/supported API on Android */ " but Nvidia is able to overcome this..
I assume this is similar to OpenCL situation on Android: not officially supported by Google but some vendors and IHV ships phones with it (Xiaomi phones with moreder Qualcomm SOCS ship with OpenCL 2.0 driver on Android)
don't know if reverting this patch is all that is needed for having full GL on Mesa drivers on Android..

```

from earlier private communication:

```
also Wine Android x86 project ships with opengl support but seems limited to passing to underlying OpenGL ES driver.. don't know if asked before but Mesa having full GL support (albeit core profile only) will be a dream come true to see full OpenGL support build too (not only OpenGL ES).. Nvidia Shield devices (tablet and TV) support that too.. you can check by using
https://play.google.com/store/apps/details?id=de.saschawillems.glescapsviewer
it shows in EGL on Shield tablet:
Client APIs(2):
OpenGL_ES
OpenGL
while Android x86 ISOs show:
CLient APIs(1):
OpenGL_ES..
(seems that is a EGL flag you set in EGL API context creation EGL_RENDERABLE_TYPE)
with that and Wine Android minor modification to load the full GL underlying Android driver we should be able to run also lots OpenGL Windows apps/games..
for example RPCS3,Cemu emulator use OpenGL 4.5 (Cemu requires mesa-mild github which exposes minimal modifications for compat profile support)
of course I tested these on Vega on Mesa 18.1dev and they are working via Wine on Ubuntu so should work also on Android-x86..
also full GL support on Wine should allow to work his D3D11 emulation layers that since Wine version 3.0 allow working on OpenGL implementations with core profile only support like Mesa..
```

AdamCetnerowski, 2018-02-17:
A closed-source OpenCL implementation was provided for our Baytrail and Cherrytrail platforms. 

NEO code can be ported and cross-compiled for Android systems. It is our long-term intention to offer that option.

closed 2018-02-23 by MichalMrozek

dimitryn, 2019-10-05:
Hi @AdamCetnerowski  , about Android development . I have a development board  with Apollo Lake cpu and  Iris graphics 500 GPU Android 8.1 . no libOpenCL.so found . Can I use this project for OpenCL development my projects?
Thanks

AdamCetnerowski, 2019-10-07:
Yes, with following caveats:
- you need to compile the driver for Android on your own
- you need to check that your version of the kernel (in Android 8.1) meets the [driver's requirements](https://github.com/intel/compute-runtime/blob/master/documentation/FAQ.md#which-linux-versions-does-neo-support)

dkurt, 2019-10-07:
@AdamCetnerowski, It seems that https://github.com/intel/intel-graphics-compiler do not allow cross compilation for Android at all.

dkurt, 2019-10-08:
Wow, looks like we did it. I can share a docker with full set of instructions this week.

dkurt, 2019-10-09:
@dimitryn, May I ask you to test experimental build of compute-runtime for Android x86 from here: https://github.com/dkurt/compute-runtime-android?