intel/compute-runtime #13: Neo OpenCL driver supports already SPV_INTEL_subgroups?

oscarbg, 2018-02-17:
Hi,
I'm very interested in this extension as it seems to provide support for cl_intel_subgroups & cl_intel_subgroups_short extensions to SPIR-V world..
don't know if extension is official yet or not..
it's strange because it can't be found on SPIR-V extension registry:
https://www.khronos.org/registry/spir-v/
but appers on:
https://www.khronos.org/registry/spir-v/specs/unified1/SPIRV.pdf
or commits like:
[Mesa-dev] [PATCH 03/10] spirv: Import the latest 1.0.12 header and JSON from Khronos

 just found by inspection that Khronos SPIRV-LLVM compiler already supports it:
https://github.com/KhronosGroup/SPIRV-LLVM/commit/4496e8920e45ccffe08441321a0775740091f9eb
it also there support for Clang 
"CLANG support for cl_intel_subgroups and cl_intel_subgroups_short"
https://github.com/KhronosGroup/SPIR/commit/7ad2092dc40b37895cb159deb1bdfc5a3c4cf382

I was lazy and still don't compiled these to projects jointly with neo opencl driver to see if SPIRV kernels using these cl_intel_subgroups_* capabilities run correctly on Neo..
I thought I could avoid compiling Khronos SPIR and SPIRV-LLVM projects by using Intel OpenCL code builder from Intel OpenCL SDK that supports generating also SPIR-V code from kernels from a nice GUI..
I tested with latest version avaiable and it doesn't worked (of course generating OpenCL vendor specific binary for Intel HD graphics using kernel using this cl_intel_subgroups_* extension worked correctly..

so guess it's will be in Intel OpenCL SDK 2018 whenever it comes out, right?

of course when I say if Neo supports this SPV extension I also mean the OpenCL Neo driver that has started appearing included on recent Windows drivers for example on 23.20.16.4933..

wait you to enlighten me on all this questions..

jchodor, 2018-02-23:
SPV_INTEL_subgroups is not yet supported. 
Adding new SPIRV capabilities to our OCL driver stack is mainly shader-compiler-driven (https://github.com/intel/intel-graphics-compiler). This applies especially to features that are already supported in our OCL driver stack, but only through OpenCL C kernel language path (like subgroup capabilities).
I've created an issue for intel-graphics-compiler (https://github.com/intel/intel-graphics-compiler/issues/6), so someone from compiler team can give more detailed answers - I will keep this issue open in NEO project as well, but mainly for tracking purposes.

bashbaug, 2018-02-24:
Continuing the SPIR-V consumption disucssion from https://github.com/KhronosGroup/SPIRV-LLVM/issues/228:

> don't tested yet but it's that SPIR-V shaders usable on any public Intel OpenCL driver either on Windows or Linux platform..
> also how is related to new Intel OpenCL Neo driver (https://github.com/intel/compute-runtime)..
> it supports SPV intel subgroups already?

We do support SPIR-V consumption on shipping Windows and Linux drivers (including the driver in this repo), but as mentioned above we do not currently support our OpenCL extensions via SPIR-V.  We do intend to support some of our OpenCL extensions via SPIR-V in the future though, and you're seeing the first signs of this via the SPIR-V generator support for SPV_INTEL_subgroups.

> I'm having trouble compiling the OpenCL file attached below (Opt_MatrixMultiplication.cl) to SPIR-V of this sample obtained from Intel site using the same Khronos OpenCL Clang compiler+SPIRV-LLVM I compiled from master that has "SPV_INTEL_subgroups":
> GEMM_sample_Jun2017.zip
> 
> it uses cl_intel_subgroups but I see in code also something like:
> #ifdef cl_intel_simd_operations_placeholder
> #define cl_intel_subgroups
> #define sub_group_broadcast intel_simd_shuffle
> #endif

Interesting, this was a placeholder name for the feature that became cl_intel_subgroups when it was in development.  I suspect this section of code was added so these kernels could run on preproduction drivers that used the old interfaces.  It's dead code now though, and can safely be removed / ignored.

> so is this OpenCL file compilable yet to SPIR-V..
> I say because it seems the fastest GEMM code avaiable for Intel HD Graphics right now and also with some very optimized f16 matmal code..

We're very curious to hear more about your SPIR-V plans for subgroups, so feel free to reach out to us (in public, or in private via email) if you can tell us more about your use case.

You might also be interested in the [Intel clDNN library](https://github.com/intel/clDNN), which contains additional optimized fp16 kernels for machine learning, if you haven't seen it already.

MichalMrozek, 2018-04-06:
Looks like questions were answered, if anything is still unclear feel free to reopen.

closed 2018-04-06 by MichalMrozek