intel/compute-runtime #14: Intel Windows OpenCL Neo vs classic comparison..

oscarbg, 2018-02-17:
to start you must clarify:
"The driver has the following functional delta compared to previously released drivers"
delta means it adds this functionality vs classic or removes it?
from what I see it seems it's removed at least on Windows driver with NEO driver..

comparison of Intel HD 530 OCL classic on drv 23.20.16.4911 vs NEO on drv 23.20.16.4933:

full reports below:
(note don't report CPU device as only changes are:
"Driver version" field goes from: **7.5.0.2** to **7.6.0.611** 
"Verison" field from "OpenCL 2.1 (**Build 2**)" to OpenCL 2.1 (**Build 611**)
)

drv 23.20.16.4911
 Platform Name:				 Intel(R) OpenCL
Number of devices:				 2
  Device Type:					 CL_DEVICE_TYPE_GPU
  Vendor ID:					 8086h
  Max compute units:				 24
  Max work items dimensions:			 3
    Max work items[0]:				 256
    Max work items[1]:				 256
    Max work items[2]:				 256
  Max work group size:				 256
  Preferred vector width char:			 16
  Preferred vector width short:			 8
  Preferred vector width int:			 4
  Preferred vector width long:			 1
  Preferred vector width float:			 1
  Preferred vector width double:		 1
  Native vector width char:			 16
  Native vector width short:			 8
  Native vector width int:			 4
  Native vector width long:			 1
  Native vector width float:			 1
  Native vector width double:			 1
  Max clock frequency:				 1150Mhz
  Address bits:					 64
  Max memory allocation:			 2147483647
  Image support:				 Yes
  Max number of images read arguments:		 128
  Max number of images write arguments:		 128
  Max image 2D width:				 16384
  Max image 2D height:				 16384
  Max image 3D width:				 16384
  Max image 3D height:				 16384
  Max image 3D depth:				 2048
  Max samplers within kernel:			 16
  Max size of kernel argument:			 1024
  Alignment (bits) of base address:		 1024
  Minimum alignment (bytes) for any datatype:	 128
  Single precision floating point capability
    Denorms:					 Yes
    Quiet NaNs:					 Yes
    Round to nearest even:			 Yes
    Round to zero:				 Yes
    Round to +ve and infinity:			 Yes
    IEEE754-2008 fused multiply-add:		 Yes
  Cache type:					 Read/Write
  Cache line size:				 64
  Cache size:					 524288
  Global memory size:				 13687249306
  Constant buffer size:				 2147483647
  Max number of constant args:			 8
  Local memory type:				 Scratchpad
  Local memory size:				 65536
  Max pipe arguments:				 16
  Max pipe active reservations:			 1
  Max pipe packet size:				 1024
  Max global variable size:			 65536
  Max global variable preferred total size:	 2147483647
  Max read/write image args:			 128
  Max on device events:				 1024
  Queue on device max size:			 67108864
  Max on device queues:				 1
  Queue on device preferred size:		 131072
  SVM capabilities:				 
    Coarse grain buffer:			 Yes
    Fine grain buffer:				 Yes
    Fine grain system:				 No
    Atomics:					 Yes
  Preferred platform atomic alignment:		 64
  Preferred global atomic alignment:		 64
  Preferred local atomic alignment:		 64
  Kernel Preferred work group size multiple:	 32
  Error correction support:			 0
  Unified memory for Host and Device:		 1
  Profiling timer resolution:			 83
  Device endianess:				 Little
  Available:					 Yes
  Compiler available:				 Yes
  Execution capabilities:				 
    Execute OpenCL kernels:			 Yes
    Execute native function:			 No
  Queue on Host properties:				 
    Out-of-Order:				 Yes
    Profiling :					 Yes
  Queue on Device properties:				 
    Out-of-Order:				 Yes
    Profiling :					 Yes
  Platform ID:					 000001B894BC4640
  Name:						 Intel(R) HD Graphics 530
  Vendor:					 Intel(R) Corporation
  Device OpenCL C version:			 OpenCL C 2.0 
  Driver version:				 23.20.16.4911
  Profile:					 FULL_PROFILE
  Version:					 OpenCL 2.1 
  Extensions:					 cl_intel_accelerator cl_intel_advanced_motion_estimation cl_intel_d3d11_nv12_media_sharing cl_intel_device_side_avc_motion_estimation cl_intel_driver_diagnostics cl_intel_dx9_media_sharing cl_intel_media_block_io cl_intel_motion_estimation cl_intel_planar_yuv cl_intel_packed_yuv cl_intel_required_subgroup_size cl_intel_simultaneous_sharing cl_intel_subgroups cl_intel_subgroups_short cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_d3d10_sharing cl_khr_d3d11_sharing cl_khr_depth_images cl_khr_dx9_media_sharing cl_khr_fp16 cl_khr_fp64 cl_khr_gl_depth_images cl_khr_gl_event cl_khr_gl_msaa_sharing cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_gl_sharing cl_khr_icd cl_khr_image2d_from_buffer cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_mipmap_image cl_khr_mipmap_image_writes cl_khr_spir cl_khr_subgroups cl_khr_throttle_hints 

drv 23.20.16.4933
 Platform Name:				 Intel(R) OpenCL
Number of devices:				 2
  Device Type:					 CL_DEVICE_TYPE_GPU
  Vendor ID:					 8086h
  Max compute units:				 24
  Max work items dimensions:			 3
    Max work items[0]:				 256
    Max work items[1]:				 256
    Max work items[2]:				 256
  Max work group size:				 256
  Preferred vector width char:			 16
  Preferred vector width short:			 8
  Preferred vector width int:			 4
  Preferred vector width long:			 1
  Preferred vector width float:			 1
  Preferred vector width double:		 1
  Native vector width char:			 16
  Native vector width short:			 8
  Native vector width int:			 4
  Native vector width long:			 1
  Native vector width float:			 1
  Native vector width double:			 1
  Max clock frequency:				 1150Mhz
  Address bits:					 64
  Max memory allocation:			 4294959104
  Image support:				 Yes
  Max number of images read arguments:		 128
  Max number of images write arguments:		 128
  Max image 2D width:				 16384
  Max image 2D height:				 16384
  Max image 3D width:				 16384
  Max image 3D height:				 16384
  Max image 3D depth:				 2048
  Max samplers within kernel:			 16
  Max size of kernel argument:			 1024
  Alignment (bits) of base address:		 1024
  Minimum alignment (bytes) for any datatype:	 128
  Single precision floating point capability
    Denorms:					 Yes
    Quiet NaNs:					 Yes
    Round to nearest even:			 Yes
    Round to zero:				 Yes
    Round to +ve and infinity:			 Yes
    IEEE754-2008 fused multiply-add:		 Yes
  Cache type:					 Read/Write
  Cache line size:				 64
  Cache size:					 524288
  Global memory size:				 13695635456
  Constant buffer size:				 4294959104
  Max number of constant args:			 8
  Local memory type:				 Scratchpad
  Local memory size:				 65536
  Max pipe arguments:				 16
  Max pipe active reservations:			 1
  Max pipe packet size:				 1024
  Max global variable size:			 65536
  Max global variable preferred total size:	 4294959104
  Max read/write image args:			 0
  Max on device events:				 1024
  Queue on device max size:			 67108864
  Max on device queues:				 1
  Queue on device preferred size:		 131072
  SVM capabilities:				 
    Coarse grain buffer:			 Yes
    Fine grain buffer:				 Yes
    Fine grain system:				 No
    Atomics:					 Yes
  Preferred platform atomic alignment:		 0
  Preferred global atomic alignment:		 0
  Preferred local atomic alignment:		 0
  Kernel Preferred work group size multiple:	 32
  Error correction support:			 0
  Unified memory for Host and Device:		 1
  Profiling timer resolution:			 83
  Device endianess:				 Little
  Available:					 Yes
  Compiler available:				 Yes
  Execution capabilities:				 
    Execute OpenCL kernels:			 Yes
    Execute native function:			 No
  Queue on Host properties:				 
    Out-of-Order:				 Yes
    Profiling :					 Yes
  Queue on Device properties:				 
    Out-of-Order:				 Yes
    Profiling :					 Yes
  Platform ID:					 00000225D9FC7700
  Name:						 Intel(R) HD Graphics 530
  Vendor:					 Intel(R) Corporation
  Device OpenCL C version:			 OpenCL C 2.1 
  Driver version:				 23.20.16.4933
  Profile:					 FULL_PROFILE
  Version:					 OpenCL 2.1 NEO 
  Extensions:					 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_fp16 cl_khr_depth_images cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_icd cl_khr_image2d_from_buffer cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_intel_subgroups cl_intel_required_subgroup_size cl_intel_subgroups_short cl_khr_spir cl_intel_accelerator cl_intel_media_block_io cl_intel_driver_diagnostics cl_intel_device_side_avc_motion_estimation cl_khr_subgroups cl_khr_il_program cl_khr_fp64 cl_intel_planar_yuv cl_intel_packed_yuv cl_intel_motion_estimation cl_intel_advanced_motion_estimation cl_khr_gl_sharing cl_khr_gl_depth_images cl_khr_gl_event cl_khr_gl_msaa_sharing cl_intel_dx9_media_sharing cl_khr_dx9_media_sharing cl_khr_d3d10_sharing cl_khr_d3d11_sharing cl_intel_d3d11_nv12_media_sharing cl_intel_simultaneous_sharing

MichalMrozek, 2018-02-19:
Hello,

In regard to delta between Clasic, this is a list of things that are missing in Neo and are present in Classic driver. We are working on closing this gap, i.e. khr_throttle_hints  was recently added and some device caps were fixed ( max read write image args are at 128 as well ).

We are planning to fill all the gaps and eventually remove the delta between the Clasic driver for things you mentioned.

Thanks for report.

AdamCetnerowski, 2018-02-19:
To provide some additional details: 

**Extensions**
We are driving towards extension parity from both Windows/Linux and Classic/NEO perspectives - expect the gap to shrink over time. 

**Capabilities**
You have a highlighted a few minor inconsistencies that we are cleaning up. 
Fine-grained SVM is an exception - this requires an i915 patch. A version of this patch was provided in previous closed source releases. An improved version is WIP and should be upstreamed soon. 

**Issue with driver 23.20.16.4944:**
There was an issue identified only after driver was posted. The next driver we post will contain a fix.

oscarbg, 2018-02-19:
@MichalMrozek @AdamCetnerowski
thanks for you fast and informative answers..

closed 2018-02-20 by oscarbg

oscarbg, 2018-02-21:
yes seems 23.20.16.4958 fixes it!