intel/compute-runtime #15: Request for an extension like cl_qcom_compressed_image..

oscarbg, 2018-02-17:
Hi,
now that Intel OpenCL development is open sourced for all platforms with some everyone can implement what they want on top of that but anyway I'm requestion for an extension that surprises me hasn't been exposed yet by almost any vendors..
I'm meaning the support for OpenCL kernel to read from compressed textures (images in OpenCL speaking).. I mean textures compressed with texture formats like ASTC,DXTx,BPTC,etc.. this must be supported by all hardware as OpenGL/OpenGL ES/DirectX compute shaders allow reading from such textures AFAIK.. still troubles me by Khronos or any vendors expose it in OpenCL.. seems Qualcomm is the only one with cl_qcom_compressed_image altough can't find info on that extensions and doesn't seem published by the name seems clear that supports compressed textures and of course must be read_only as writing to compressed textures from compute shaders seems unsupported by current HW..

of course it has it's uses.. I'm thinking of custom render pipelines/rasterizers or raytracers implemented with OpenCL kernels.. if we want to use same assets (stored efficiently with compressed textures) as with current GPU pipeline we need this functionality..
Projects like these were tested with some success years ago:
http://research.nvidia.com/publication/high-performance-software-rasterization-gpus
altough in CUDA and remember seeing in the paper noting as current limitations/future work that compute APIs like CUDA didn't expose compressed texture at the time.. AFAIK it hasn't changed still altough this limitation was pointed by his own Nvidia researchers..

AdamCetnerowski, 2018-02-19:
Thank you for your suggestion. This requires some technical analysis on our side. We will keep this issue open until we have a repsonse for you.

oscarbg, 2018-02-19:
thanks for taking into consideration.. ideally I wish gets eventually as EXT/KHR and exposed almost by every vendor on modern hardware.. for me seems like a somewhat obvious omission for such a long time since OpenCL has been released..

MichalMrozek, 2018-02-23:
Can you share some use cases?
Are you interested in reading from such compressed textures?

I was not able to find any reference for the cl_qcom_compressed_image.

In OpenCL we are not able to write to compressed surface from the GPU, this is only available in 3D pipeline, that's why DirectX/OpenGL has the support. OpenCL uses GPGPU Graphics Pipeline that has its limits.

We can potentially support reads from compressed texture, but the data needs to be uploaded to it somehow. It can be done either in DirectX/OpenGL or through some kind of new OpenCL API.
If you could share some use cases , that would definitely help with evaluation.

MichalMrozek, 2018-03-24:
Hello @oscarbg 
Can  you provide more details ?
Is this still relevant after my update?

MichalMrozek, 2018-03-30:
Closing due to longer then one month inaction.
Feel free to reopen.

closed 2018-03-30 by MichalMrozek