intel/compute-runtime #16: Compilation error on debian?

ThijsWithaar, 2018-02-17:
When compiling on debian buster, using clang 5.0.1-2, the make step give me:
```
Scanning dependencies of target igfx_gmmumd_excite
[  1%] Building CXX object gmmlib/Source/GmmLib/CMakeFiles/igfx_gmmumd_excite.dir/__/Common/AssertTracer/AssertTracer.cpp.o
clang: error: unknown argument: '-fno-tree-pre'
clang: warning: -Wl,--no-undefined: 'linker' input unused [-Wunused-command-line-argument]
```

Are there some compile settings I should change ?

ArturHarasimiuk, 2018-02-19:
We have looked into this and it appears the problem is in [GmmLib](https://github.com/intel/gmmlib) component. We have filed an issue there [link](https://github.com/intel/gmmlib/issues/7).

MichalMrozek, 2018-02-22:
Looks like problem is fixed in gmmlib.
Closing this one.

closed 2018-02-22 by MichalMrozek