intel/compute-runtime #18: Comparing Win Neo vs Linux Neo driver..

oscarbg, 2018-02-18:


oscarbg, 2018-02-18:
Some benchmarks:
Platform: Intel(R) OpenCL HD Graphics
half supported!  Device: Intel(R) Gen9 HD Graphics NEO
    Driver version  : 1.0 (Linux x64)
    Compute units   : 24
    Clock frequency : 0 MHz

    Half-precision compute (GFLOPS)
      half   : 748.56
      half2  : 743.20
      half4  : 741.60
      half8  : 735.49
      half16 : 729.58

    Global memory bandwidth (GBPS)
      float   : 26.32
      float2  : 25.38
      float4  : 27.56
      float8  : 28.44
      float16 : 21.64

    Single-precision compute (GFLOPS)
      float   : 436.48
      float2  : 436.28
      float4  : 436.23
      float8  : 434.85
      float16 : 429.21

    Double-precision compute (GFLOPS)
      double   : 110.00
      double2  : 109.86
      double4  : 109.64
      double8  : 109.05
      double16 : 108.10

    Transfer bandwidth (GBPS)
      enqueueWriteBuffer         : 32.54
      enqueueReadBuffer          : 12.91
      enqueueMapBuffer(for read) : 2684354.50
        memcpy from mapped ptr   : 12.79
      enqueueUnmap(after write)  : inf
        memcpy to mapped ptr     : 13.12

    Kernel launch latency : 25.37 us

AdamCetnerowski, 2018-02-19:
On extensions in Linux, not present in Windows:
NEO supports Windows and Linux using the same code base, but there is some latency in delivering the Windows closed source driver (integration, validation, signing) compared to Linux, where the source code is provided continously. These extensions will be present in the Windows shortly. New features on Windows will be present in subsequent driver releases. 

On cl_khr_create_command_queue:
Posting of the spec for this extension is WIP in Khronos. 

On gl-sharing:
The NEO driver on Windows provides sharing with the closed source OGL driver developed by VPG. On Linux, we intend to add sharing with MESA, but do not have any firm timelines as of today.

oscarbg, 2018-02-20:
really thanks for info.. guess I can close..

closed 2018-02-20 by oscarbg