intel/compute-runtime #20: Unable to get working on Broadwell, or missing steps?

lissyx, 2018-02-19:
I have built and installed without any error (following exactly the steps from https://github.com/intel/compute-runtime#building) on my laptop (ubuntu 17.10, Broadwell i7-5600U), but `clinfo` returns 0 platform found. Checking `clinfo` calls with `strace` shows it is properly picking-up `/opt/intel/opencl/libigdrcl.so`.

Is Broadwell supported right now? Readme states it is https://github.com/intel/compute-runtime#supported-platforms, but the "GenX" naming is unclear to me, and https://ark.intel.com/products/85215/Intel-Core-i7-5600U-Processor-4M-Cache-up-to-3_20-GHz reports "5th gen".

Thanks!

lissyx, 2018-02-19:
The GPU seems to be identified as Broadwell in the source as well, under the define `IBDW_GT2_ULT_MOBL_DEVICE_F0_ID`: https://github.com/intel/intel-graphics-compiler/blob/e42b674bce86b372c92ba08812cecccc71debafc/inc/common/igfxfmid.h#L354

pwilma, 2018-02-19:
Brodwell is supported by Neo driver. Your issue looks similar to problem already reported in https://github.com/intel/compute-runtime/issues/9. Could you please check it? To operate correctly Neo driver requires Khronos ICD loader: https://github.com/KhronosGroup/OpenCL-ICD-Loader. It is known limitation that we are trying to solve.

lissyx, 2018-02-19:
@pwilma Thanks! I did search in issues before, but missed #9. I also saw the note about the Khronos ICD loader, but I mistakenly understood that this was provided by the `ocl-icd-libopencl1` package on debian/ubuntu. I'll verify all of that, thanks !

lissyx, 2018-02-19:
Ok, `OCL_ICD_ASSUME_ICD_EXTENSION=1 clinfo` gives cool stuff, it might just be the ICD loader that I messed up with :)

lissyx, 2018-02-19:
Okay, after adding a clone of https://github.com/KhronosGroup/OpenCL-ICD-Loader into the set of sources for the Neo driver and building it by hand, re-running CMake, it was properly picked up as this line shows: 
`-- Taking ICD library from /home/alex/codaz/Mozilla/DeepSpeech/Intel-Neo/neo/../OpenCL-ICD-Loader/build/lib`

Then, it's being properly packaged, and `clinfo` returns proper status, as well as does `computecpp_info`. Now, my TensorFlow code is failing in `neo/runtime/os_interface/linux/drm_buffer_object.cpp` at line 183, but that might be something expected.

lissyx, 2018-02-19:
@pwilma i'm unable to get any debug, even forcing `PrintDebugMessages=1` on the env, and I could not find that mentionned in any doc. How am I supposed to get those? Should I rebuild with something different than `Release` ?

pwilma, 2018-02-19:
Yes, debug variables won't work for Release. Please try to build Debug version.

lissyx, 2018-02-19:
@pwilma Thanks, I'm doing that right now. BTW, one step I had to make but I'm not completely sure about is: I did git clone the Khronos ICD  Loader next to the neo driver, and I built it by hand. Was building by hand required, or should your CMake build system have handled that for me automagically?

pwilma, 2018-02-19:
You did it correctly. Manual build of Khronos ICD Loader is required. Neo CMake buildsystem won't build it automatically.

lissyx, 2018-02-19:
@pwilma Debug-enabled build is failing: `igc/IGC/AdaptorCommon/customApi.cpp:555:41: error: 'struct SRegKeysList' has no member named 'EnableDxbcDump'; did you mean 'EnableCosDump'?`

lissyx, 2018-02-19:
@pwilma I've removed the unexisting member, built and installed debug build, but I'm still unable to get any debug message :(

lissyx, 2018-02-19:
Not sure yet, but there's a `return -EFAULT` here in `i915_gem_execbuffer2()`: http://kernel.ubuntu.com/git/ubuntu/ubuntu-artful.git/tree/drivers/gpu/drm/i915/i915_gem_execbuffer.c#n2487

lissyx, 2018-02-19:
Enabling DRM's debug in `/sys/module/drm/parameters/debug`, I get this for the PID of my process:
```
# grep '8911' dmesg.txt 
[423537.069035] [drm:drm_open [drm]] pid = 8911, minor = 128
[423537.069103] [drm:drm_ioctl [drm]] pid=8911, dev=0xe280, auth=1, DRM_IOCTL_VERSION
[423537.069111] [drm:drm_ioctl [drm]] pid=8911, dev=0xe280, auth=1, I915_GETPARAM
[423537.069117] [drm:drm_ioctl [drm]] pid=8911, dev=0xe280, auth=1, I915_GETPARAM
[423537.069123] [drm:drm_ioctl [drm]] pid=8911, dev=0xe280, auth=1, I915_GETPARAM
[423537.069128] [drm:drm_ioctl [drm]] pid=8911, dev=0xe280, auth=1, I915_GEM_CONTEXT_SETPARAM
[423537.069164] [drm:drm_ioctl [drm]] pid=8911, dev=0xe280, auth=1, I915_GETPARAM
[423537.069169] [drm:drm_ioctl [drm]] pid=8911, dev=0xe280, auth=1, I915_GETPARAM
[423537.069174] [drm:drm_ioctl [drm]] pid=8911, dev=0xe280, auth=1, I915_GETPARAM
[423537.069179] [drm:drm_ioctl [drm]] pid=8911, dev=0xe280, auth=1, I915_GETPARAM
[423537.069184] [drm:drm_ioctl [drm]] pid=8911, dev=0xe280, auth=1, I915_GETPARAM
[423537.069210] [drm:drm_ioctl [drm]] pid=8911, dev=0xe280, auth=1, I915_GETPARAM
[423537.069216] [drm:drm_ioctl [drm]] pid=8911, dev=0xe280, auth=1, I915_GETPARAM
[423537.072558] [drm:drm_ioctl [drm]] pid=8911, dev=0xe280, auth=1, I915_GEM_USERPTR
[423537.072643] [drm:drm_ioctl [drm]] pid=8911, dev=0xe280, auth=1, I915_GEM_USERPTR
[423537.072660] [drm:drm_ioctl [drm]] pid=8911, dev=0xe280, auth=1, I915_REG_READ
[423537.072699] [drm:drm_ioctl [drm]] pid=8911, dev=0xe280, auth=1, I915_GEM_GET_APERTURE
[423537.334023] [drm:drm_release [drm]] pid = 8911, device = 0xe280, open_count = 6
```

lissyx, 2018-02-19:
@pwilma Here is most of the debug info I could gather so far. Is it possible it is failing because the driver does not yet support enough of the required calls / opencl for proper operation with TensorFlow / ComputeCpp 0.5.1 ? Or do I need a newer kernel (Ubuntu 17.10 is on 4.13) ?

lissyx, 2018-02-19:
@pwilma I've also collected some deubg files, if it can be of help, tell me what would be the best way to share them:
```
$ ls -ah CopyBufferToBuffer*
CopyBufferToBufferLeftLeftover.txt			       CopyBufferToBufferMiddle_arg_2_immediate_size_4_flags_0.bin	     CopyBufferToBufferRightLeftover_arg_1_buffer_size_32_flags_9.bin
CopyBufferToBufferMiddle_arg_1_buffer_size_128_flags_9.bin     CopyBufferToBufferMiddle_arg_3_immediate_size_4_flags_0.bin	     CopyBufferToBufferRightLeftover_arg_2_immediate_size_4_flags_0.bin
CopyBufferToBufferMiddle_arg_1_buffer_size_193664_flags_9.bin  CopyBufferToBufferMiddle.txt					     CopyBufferToBufferRightLeftover_arg_3_immediate_size_4_flags_0.bin
CopyBufferToBufferMiddle_arg_1_buffer_size_32768_flags_9.bin   CopyBufferToBufferRightLeftover_arg_1_buffer_size_128_flags_9.bin     CopyBufferToBufferRightLeftover.txt
CopyBufferToBufferMiddle_arg_1_buffer_size_8192_flags_9.bin    CopyBufferToBufferRightLeftover_arg_1_buffer_size_193664_flags_9.bin
```

MichalMrozek, 2018-02-19:
Useful stuff for debugging:
1. Open file vpg-compute-neo\runtime\os_interface\DebugVariables.def , there are various debug flags available there.
2. Good thing to start is to EnableDebugBreak to true
3. Here problems looks like a failure in exec call, this probably means that inputs are incorrect. To isolate faulty kernel please set MakeEachEnqueueBlocking, that will tell you which clEnqueueNDRangeKernel call is failing
4. Knowing which kernel fails , look at its arguments, are there any buffers created with CL_MEM_USE_HOST_PTR flag there ? If so, can you rule out that there are any duplicates ( CL_MEM_USE_HOST_PTR buffers created from the same host_ptr)
5. Please turn on logging and provide dumps:
DumpKernels/DumpKernelArgs/LogApiCalls/LogPatchTokens/LogMemoryObject

lissyx, 2018-02-19:
@MichalMrozek Thanks! I have the following in `igdrcl.config`, is that okay? it seems to provide debug infos at least:
```
PrintDebugMessages = 1
EnableDebugBreak = true
MakeEachEnqueueBlocking = 1
DumpKernels = 1
DumpKernelArgs = 1
LogApiCalls = 1
LogPatchTokens = 1
LogMemoryObject = 1
```

lissyx, 2018-02-19:
I cannot find anything referring to `clEnqueueNDRangeKernel` in `igdcrl.log`

MichalMrozek, 2018-02-19:
Can you try to use this flag ?
DoCpuCopyOnWriteBuffer to 1.

lissyx, 2018-02-19:
@MichalMrozek Magic! it's not failing anymore, it seems to be doing actual *stuff* :)

MichalMrozek, 2018-02-19:
Looks like there may be a bug in our clEnqueueWriteBuffer implementation.
We will try to reproduce it basing on the logs sent.
If we fail we will contact for more robust reproducer, thanks for reporting this.

lissyx, 2018-02-19:
@MichalMrozek I can share you the binaries to reproduce as well, if you need. It's DeepSpeech + ComputeCpp 0.5.1.

Now, so far, it started the run, but it's not completing anything yet, the output is "blocked" on this:
```
Kernel Name: SYCL_struct_Eigen__TensorSycl__ExecExprFunctorKernel_const_class_Eigen__TensorAssignOp_class_Eigen__TensorMap_class_Eigen__Tensor_struct_std__complex_double___1__1__long___16__MakePointer___const_class_Eigen__TensorConversionOp_struct_std__complex_double___const_class_Eigen__TensorMap_class_Eigen__Tensor_const_long_long__1__1__long___16__MakePointer_______struct_utility__tuple__Tuple_struct_utility__tuple__Tuple_struct_Eigen__DSizes_long__1_____struct_utility__tuple__Tuple_struct_utility__tuple__Tuple_struct_Eigen__DSizes_long__1_________struct_utility__tuple__Tuple_struct_Eigen__RangeAccess_cl__sycl__access__mode__read_write___struct_Eigen__RangeAccess_cl__sycl__access__mode__read_____false_
Gen7 Kernel Binary Header:
	CheckSum = 66b54f10
	ShaderHashCode = 104d0de50e756486
	KernelNameSize = 704
	PatchListSize = 1308
	KernelHeapSize = 2688
	GeneralStateHeapSize = 0
	DynamicStateHeapSize = 32
	SurfaceStateHeapSize = 136
Program Binary Header:
	Magic = 494e5443
	Version = 1049
	Device = 11
	GPUPointerSizeInBytes = 8
	NumberOfKernels = 22
	SteppingId = 9
	PatchListSize = 0
```

The `deepspeech` process seems to be consuming as much as CPU as it can, so maybe it's just slow, given the model is quite big and it's a debug build.

lissyx, 2018-02-19:
@MichalMrozek Building release and changing the value of `DoCpuCopyOnWriteBuffer` to `true` in `runtime/os_interface/DebugVariables.def` fails like this:
```
/home/alex/codaz/Mozilla/DeepSpeech/Intel-Neo/neo/unit_tests/command_queue/enqueue_write_buffer_tests.cpp:453: Failure
Value of: retVal
  Actual: 0
Expected: -5

/home/alex/codaz/Mozilla/DeepSpeech/Intel-Neo/neo/unit_tests/memory_leak_listener.cpp:55: Failure
Value of: (int)MemoryManagement::failingAllocation
  Actual: -1
Expected: leak
Which is: 135
[  FAILED  ] NegativeFailAllocationTest.givenEnqueueWriteBufferWhenHostPtrAllocationCreationFailsThenReturnOutOfResource

/home/alex/codaz/Mozilla/DeepSpeech/Intel-Neo/neo/unit_tests/command_queue/ooq_task_tests.cpp:67: Failure
Expected: (previousTaskCount) < (commandStreamReceiver.peekTaskCount()), actual: 0 vs 0
*** WARNING: Leaks found but dumping disabled during test failure ***

/home/alex/codaz/Mozilla/DeepSpeech/Intel-Neo/neo/unit_tests/memory_leak_listener.cpp:55: Failure
Value of: (int)MemoryManagement::failingAllocation
  Actual: -1
Expected: leak
Which is: 178
[  FAILED  ] OOQ/OOQTaskTypedTests/6.changesTaskCount

/home/alex/codaz/Mozilla/DeepSpeech/Intel-Neo/neo/unit_tests/command_queue/ooq_task_tests.cpp:53: Failure
Value of: this->pCmdQ->taskLevel
  Actual: 0
Expected: previousTaskLevel + taskLevelClosed
Which is: 1
*** WARNING: Leaks found but dumping disabled during test failure ***

/home/alex/codaz/Mozilla/DeepSpeech/Intel-Neo/neo/unit_tests/memory_leak_listener.cpp:55: Failure
Value of: (int)MemoryManagement::failingAllocation
  Actual: -1
Expected: leak
Which is: 168
[  FAILED  ] OOQ/OOQTaskTypedTests/6.doesntChangeTaskLevel

/home/alex/codaz/Mozilla/DeepSpeech/Intel-Neo/neo/unit_tests/command_queue/enqueue_thread_tests.cpp:67: Failure
Expected: (toFree.size()) > (0u), actual: 0 vs 0
*** WARNING: Leaks found but dumping disabled during test failure ***

/home/alex/codaz/Mozilla/DeepSpeech/Intel-Neo/neo/unit_tests/memory_leak_listener.cpp:55: Failure
Value of: (int)MemoryManagement::failingAllocation
  Actual: -1
Expected: leak
Which is: 79
[  FAILED  ] EnqueueThreading.enqueueWriteBuffer

/home/alex/codaz/Mozilla/DeepSpeech/Intel-Neo/neo/unit_tests/command_queue/get_size_required_buffer_tests.cpp:358: Failure
Value of: usedAfterCS - usedBeforeCS
  Actual: 0
Expected: expectedSizeCS
Which is: 192
*** WARNING: Leaks found but dumping disabled during test failure ***

/home/alex/codaz/Mozilla/DeepSpeech/Intel-Neo/neo/unit_tests/memory_leak_listener.cpp:55: Failure
Value of: (int)MemoryManagement::failingAllocation
  Actual: -1
Expected: leak
Which is: 338
[  FAILED  ] GetSizeRequiredBufferTest.enqueueWriteBufferNonBlocking

/home/alex/codaz/Mozilla/DeepSpeech/Intel-Neo/neo/unit_tests/command_queue/get_size_required_buffer_tests.cpp:413: Failure
Value of: usedAfterCS - usedBeforeCS
  Actual: 0
Expected: expectedSizeCS
Which is: 192
*** WARNING: Leaks found but dumping disabled during test failure ***

/home/alex/codaz/Mozilla/DeepSpeech/Intel-Neo/neo/unit_tests/memory_leak_listener.cpp:55: Failure
Value of: (int)MemoryManagement::failingAllocation
  Actual: -1
Expected: leak
Which is: 338
[  FAILED  ] GetSizeRequiredBufferTest.enqueueWriteBufferBlocking
Tests timeout on: CommandStreamReceiverFlushTaskTests.GivenBlockedKernelNotRequiringDCFlushWhenUnblockedThenDCFlushIsNotAdded
Aborted (core dumped)
unit_tests/CMakeFiles/run_skl_unit_tests.dir/build.make:61: recipe for target 'run_skl_unit_tests' failed
make[2]: *** [run_skl_unit_tests] Error 134
CMakeFiles/Makefile2:5095: recipe for target 'unit_tests/CMakeFiles/run_skl_unit_tests.dir/all' failed
make[1]: *** [unit_tests/CMakeFiles/run_skl_unit_tests.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Tests timeout on: CommandStreamReceiverFlushTaskTests.GivenBlockedKernelNotRequiringDCFlushWhenUnblockedThenDCFlushIsNotAdded
Tests timeout on: CommandStreamReceiverFlushTaskTests.GivenBlockedKernelNotRequiringDCFlushWhenUnblockedThenDCFlushIsNotAdded
Aborted (core dumped)
unit_tests/CMakeFiles/run_bdw_unit_tests.dir/build.make:61: recipe for target 'run_bdw_unit_tests' failed
make[2]: *** [run_bdw_unit_tests] Error 134
CMakeFiles/Makefile2:5199: recipe for target 'unit_tests/CMakeFiles/run_bdw_unit_tests.dir/all' failed
make[1]: *** [unit_tests/CMakeFiles/run_bdw_unit_tests.dir/all] Error 2
Tests timeout on: CommandStreamReceiverFlushTaskTests.GivenBlockedKernelNotRequiringDCFlushWhenUnblockedThenDCFlushIsNotAdded
Aborted (core dumped)
unit_tests/CMakeFiles/run_kbl_unit_tests.dir/build.make:61: recipe for target 'run_kbl_unit_tests' failed
make[2]: *** [run_kbl_unit_tests] Error 134
CMakeFiles/Makefile2:5447: recipe for target 'unit_tests/CMakeFiles/run_kbl_unit_tests.dir/all' failed
make[1]: *** [unit_tests/CMakeFiles/run_kbl_unit_tests.dir/all] Error 2
Aborted (core dumped)
unit_tests/CMakeFiles/run_bxt_unit_tests.dir/build.make:61: recipe for target 'run_bxt_unit_tests' failed
make[2]: *** [run_bxt_unit_tests] Error 134
CMakeFiles/Makefile2:5836: recipe for target 'unit_tests/CMakeFiles/run_bxt_unit_tests.dir/all' failed
make[1]: *** [unit_tests/CMakeFiles/run_bxt_unit_tests.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

lissyx, 2018-02-19:
@MichalMrozek Okay, performances are very bad, but it executed successfully:
```
$ time ./deepspeech ~/codaz/Mozilla/DeepSpeech/deepspeech-kdavis/tf14.frozen.494_e120.LSTM.ldc93s1.pb ../models/alphabet.txt ~/codaz/Mozilla/DeepSpeech/deepspeech-kdavis/data/smoke_test/LDC93S1.wav -t
Warning: reading entire model file into memory. Transform model file into an mmapped graph to reduce heap usage.
2018-02-19 18:54:33.238904: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-02-19 18:54:33.243317: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-02-19 18:54:33.243427: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
shehadyourdarksuitingreasywwsh wtteral year
cpu_time_overall=835.32753 cpu_time_mfcc=0.00770 cpu_time_infer=835.31983

real	7m14,412s
user	9m31,262s
sys	4m24,331s
```

For reference, the same run on pure CPU takes 4-6 secs instead of 835. But for experimental driver and with the flag that we had to flip, it's probably not surprising. Good news is that it works !

MichalMrozek, 2018-02-19:
It is expected that some tests will fail in non-default debug variable setting.
Those variables are meant to be used for debug/experiments.
Glad to hear that stuff is working.

If you are interested in analyzing why execution is slow I kindly suggest following project
https://github.com/intel/opencl-intercept-layer

Good to start would be
ContextHintLevel to 0xff 

This will analyze application for potential pitfalls.

If you want to find the GPU execution times I recommend
DevicePerformanceTiming to 1

lissyx, 2018-02-19:
@MichalMrozek Thanks for those hints, I'll take a look into that. I pushed the testing to comparing the same model:
```
alex@portable-alex:~/tmp/deepspeech/sycl$ time ./deepspeech ../models/prod.rw.pbmm ../models/alphabet.txt ../audio/2830-3980-0043.wav -t
2018-02-19 19:03:28.432828: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-02-19 19:03:28.439545: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-02-19 19:03:28.439639: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
experience proves tis
cpu_time_overall=1006.05650 cpu_time_mfcc=0.00631 cpu_time_infer=1006.05019

real	6m32,310s
user	12m39,560s
sys	4m6,630s
```

And on pure CPU:
```
alex@portable-alex:~/tmp/deepspeech/cpu$ time ./deepspeech ../models/prod.rw.pbmm ../models/alphabet.txt ../audio/2830-3980-0043.wav -t
2018-02-19 19:24:52.936272: I tensorflow/core/platform/cpu_feature_guard.cc:137] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
experience proves tis
cpu_time_overall=8.89595 cpu_time_mfcc=0.00451 cpu_time_infer=8.89144

real	0m4,623s
user	0m8,848s
sys	0m0,117s
```

Given the current state of your driver (and the flag flipped), are those times expected, or is there something valuable to investigate ?

MichalMrozek, 2018-02-19:
This flag only changes how writeBuffer operation is being handled, instead of going to GPU it uses CPU for transfer. It is unlikely that it affects performance so significantly, therefore I think problem is somewhere else. I think your investigation may provide valuable feedback.

lissyx, 2018-02-20:
@MichalMrozek Just sent you the logs produced with the suggested flags. I had a look at the report, there are some entries accounting for a lot of the time but I'm unsure how to interpret this :)

lissyx, 2018-02-21:
Summup of the finding. I've used the tool from https://github.com/intel/opencl-intercept-layer, with:
 - ContextHintLevel=0xFF
 - DevicePerformanceTiming=1
 - HostPerformanceTiming=1

Generated log shows that over the 500/800 secs (depending on some of the runs) of the overall inference time reported by DeepSpeech, the actual time spent running OpenCL part was ~5 secs. Most of the time is actually spent compiling the (very complicated) kernels. This might be an issue from TensorFlow, ComputeCpp layer, or the SYCL-specific branch I'm working with. I'll liaise with upstreams on that to find where it comes from.

Thanks @MichalMrozek for this analysis!

MichalMrozek, 2018-02-21:
One small clarification, long compilation happens inside the OpenCL driver as a part of the setup.
Execution on the GPU takes around 5 seconds.
One way to mitigate this heavy compilation cost is to cache compiled binaries and re-use them in subsequent application runs, but that needs to be developed as a part of the actual application.

mejch, 2018-02-22:
@lissyx Can you check how host_ptr passed to clEnqueueWriteBuffer is allocated? Is it something like const int [] = {1}; ? (The pointer comes from prod.rw.pbmm.)

lissyx, 2018-02-22:
@mejch I'm not sure I can do that, with all the layers in the middle. The file `prod.rw.pbmm` is a protoculbuffer file modified to be `mmap()`'d, using TensorFlow MemmappedFileSystem codebase.

lissyx, 2018-02-22:
@mejch Looks like you had a good hint with `DoCpuCopyOnWriteBuffer`, I've rebuilt the driver without it, and I do am able to run everything when I don't use the `mmap()` model. I'll need to find a way to hook somewhere to check how the pointer is being allocated.

lissyx, 2018-02-22:
@mejch Our code in DeepSpeech relies on this call `MemmappedFileSystem::InitializeFromFile` https://github.com/mozilla/tensorflow/blob/e999fb4ed9c9a8e986ecdbabfa551dc82a3ea0dc/tensorflow/core/util/memmapped_file_system.cc#L189-L190,

which then relies on `MemmappedFileSystem::NewReadOnlyMemoryRegionFromFile`: https://github.com/mozilla/tensorflow/blob/e999fb4ed9c9a8e986ecdbabfa551dc82a3ea0dc/tensorflow/core/util/memmapped_file_system.cc#L109-L122

Which also relies on `ReadOnlyMemoryRegionFromMemmapped`: https://github.com/mozilla/tensorflow/blob/e999fb4ed9c9a8e986ecdbabfa551dc82a3ea0dc/tensorflow/core/util/memmapped_file_system.cc#L37-L49
And on `GetMemoryWithOffset`: https://github.com/mozilla/tensorflow/blob/e999fb4ed9c9a8e986ecdbabfa551dc82a3ea0dc/tensorflow/core/util/memmapped_file_system.cc#L177-L179

MichalMrozek, 2018-02-22:
Pointer that is being passed to clEnqueueWriteBuffer call is allocated in read_only memory region.
Such pointer cannot be used on the GPU with the current implementation of Kernel Mode driver, that results in exec failure. This is also why doing the copy on the CPU is successful here, as CPU has no problems in reading from this memory.
We are currently working on a solution for this problem.

lissyx, 2018-02-23:
@MichalMrozek And coming back to the high compilation time, one can see the impact when running multiple inferences:
Indeed, when running multiple inferences:
```
alex@portable-alex:~/tmp/deepspeech/sycl$ time ./deepspeech ../models/prod.rw.pbmm ../models/alphabet.txt ../audio/ -t
2018-02-23 14:22:07.593181: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-02-23 14:22:07.598748: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-02-23 14:22:07.598999: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
Running on directory ../audio/
> ../audio//2830-3980-0043.wav
experience proves tis
cpu_time_overall=421.17855 cpu_time_mfcc=0.00900 cpu_time_infer=421.16955
> ../audio//4507-16021-0012.wav
why should one halt on the way
cpu_time_overall=16.96728 cpu_time_mfcc=0.01023 cpu_time_infer=16.95706
> ../audio//8455-210777-0068.wav
your powr is sufficient i said
cpu_time_overall=14.70185 cpu_time_mfcc=0.00703 cpu_time_infer=14.69482

real	3m33,430s
user	5m20,556s
sys	2m12,456s
alex@portable-alex:~/tmp/deepspeech/sycl$ 
```

lissyx, 2018-02-23:
@MichalMrozek Comparison with CPU:
```
alex@portable-alex:~/tmp/deepspeech/cpu$ time ./deepspeech ../models/prod.rw.pbmm ../models/alphabet.txt ../audio/ -t
2018-02-23 18:52:47.798489: I tensorflow/core/platform/cpu_feature_guard.cc:137] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
Running on directory ../audio/
> ../audio//2830-3980-0043.wav
experience proves tis
cpu_time_overall=10.33152 cpu_time_mfcc=0.00298 cpu_time_infer=10.32854
> ../audio//4507-16021-0012.wav
why should one halt on the way
cpu_time_overall=13.97146 cpu_time_mfcc=0.00377 cpu_time_infer=13.96769
> ../audio//8455-210777-0068.wav
your powr is sufficient i said
cpu_time_overall=14.18832 cpu_time_mfcc=0.00361 cpu_time_infer=14.18472

real	0m19,460s
user	0m38,387s
sys	0m0,248s
```

It's interesting that the times are that close :)

MichalMrozek, 2018-02-26:
We have an experimental feature called cl_cache that is implemented in the driver.
It automatically creates kernel cache and reuses it.
Feature is not at production quality, but under certain conditions it may work correctly.

Known limitations:
- cl_cache is not process or thread safe, if applications has mutliple threads with compliation problems may occur
- cl_cache doesn't detect that driver changed, may require cleanup when compiler/driver changes
- cl_cache doesn't handle out of memory/out of disk space situations

To actually use it you just need to create "cl_cache" directory in the place where you execute your application. After first run the cl_cache directory should be populated with binaries that will be reused for further runs. Everything should happen automatically.

Can you try it in your case?

lissyx, 2018-02-26:
@MichalMrozek Sure, I'm going to give it a try !

lissyx, 2018-02-26:
@MichalMrozek It's not working that bad:
```
alex@portable-alex:~/tmp/deepspeech/sycl$ ./deepspeech ../models/prod.rw.pbmm ../models/alphabet.txt ../audio/ -t
2018-02-26 15:46:45.784346: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-02-26 15:46:45.790696: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-02-26 15:46:45.790725: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
Running on directory ../audio/
> ../audio//2830-3980-0043.wav
experience proves tis
cpu_time_overall=335.20297 cpu_time_mfcc=0.01386 cpu_time_infer=335.18911
> ../audio//4507-16021-0012.wav
why should one halt on the way
cpu_time_overall=16.32911 cpu_time_mfcc=0.00752 cpu_time_infer=16.32159
> ../audio//8455-210777-0068.wav
your powr is sufficient i said
cpu_time_overall=15.15925 cpu_time_mfcc=0.00704 cpu_time_infer=15.15221
alex@portable-alex:~/tmp/deepspeech/sycl$ ./deepspeech ../models/prod.rw.pbmm ../models/alphabet.txt ../audio/ -t
2018-02-26 16:04:57.757966: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-02-26 16:04:57.764593: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-02-26 16:04:57.764775: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
Running on directory ../audio/
> ../audio//2830-3980-0043.wav
experience proves tis
cpu_time_overall=11.42166 cpu_time_mfcc=0.00619 cpu_time_infer=11.41548
> ../audio//4507-16021-0012.wav
why should one halt on the way
cpu_time_overall=15.13612 cpu_time_mfcc=0.00772 cpu_time_infer=15.12841
> ../audio//8455-210777-0068.wav
your powr is sufficient i said
cpu_time_overall=15.02534 cpu_time_mfcc=0.01020 cpu_time_infer=15.01515
alex@portable-alex:~/tmp/deepspeech/sycl$ ll cl_cache/
total 18M
drwxr-xr-x 1 alex alex  700 févr. 26 15:49 .
drwxr-xr-x 1 alex alex  228 févr. 26 15:46 ..
-rw-r--r-- 1 alex alex 157K févr. 26 15:48 0851272af23873f4.cl_cache
-rw-r--r-- 1 alex alex 430K févr. 26 15:48 5b4cc966d8031cee.cl_cache
-rw-r--r-- 1 alex alex 746K févr. 26 15:48 6a3511d5f5292230.cl_cache
-rw-r--r-- 1 alex alex 445K févr. 26 15:48 6e3a0c9bbcefabca.cl_cache
-rw-r--r-- 1 alex alex 108K févr. 26 15:49 7240a36f159e96b5.cl_cache
-rw-r--r-- 1 alex alex 103K févr. 26 15:46 8d38675d5792c94f.cl_cache
-rw-r--r-- 1 alex alex 105K févr. 26 15:46 8e413cca1015b855.cl_cache
-rw-r--r-- 1 alex alex 5,2M févr. 26 15:49 a2747b6b34acb7e0.cl_cache
-rw-r--r-- 1 alex alex 8,3M févr. 26 15:48 a903c3bd25e64100.cl_cache
-rw-r--r-- 1 alex alex 348K févr. 26 15:49 ba63f2311f257518.cl_cache
-rw-r--r-- 1 alex alex  79K févr. 26 15:49 c55847161d6b39dc.cl_cache
-rw-r--r-- 1 alex alex 208K févr. 26 15:49 c6d01efbe6f8699f.cl_cache
-rw-r--r-- 1 alex alex 737K févr. 26 15:49 e4c0fd4cc98f6cd5.cl_cache
-rw-r--r-- 1 alex alex 529K févr. 26 15:48 fb12a0ef441f9011.cl_cache
alex@portable-alex:~/tmp/deepspeech/sycl$ 
```

MichalMrozek, 2018-02-26:
Cool glad to hear that, however use with caution :)
One thing related to performance.
Can you change following things:
OverrideEnableKmdNotify to 1
OverrideKmdNotifyDelayMs to 1

If there will be any delta, can you try with different values of OverrideKmdNotifyDelayMs ?

lissyx, 2018-02-26:
I need to rebuild the driver for those, right?

MichalMrozek, 2018-02-26:
Yes that is correct, every change to those variable requires rebuild to obtain new release binaries.

lissyx, 2018-02-26:
@MichalMrozek There's some improvement again with those changed:
```
alex@portable-alex:~/tmp/deepspeech/sycl$ ./deepspeech ../models/prod.rw.pbmm ../models/alphabet.txt ../audio/ -t
2018-02-26 17:40:35.942455: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-02-26 17:40:35.944148: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-02-26 17:40:35.944173: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
Running on directory ../audio/
> ../audio//2830-3980-0043.wav
experience proves tis
cpu_time_overall=333.18060 cpu_time_mfcc=0.00624 cpu_time_infer=333.17437
> ../audio//4507-16021-0012.wav
why should one halt on the way
cpu_time_overall=8.62868 cpu_time_mfcc=0.00716 cpu_time_infer=8.62152
> ../audio//8455-210777-0068.wav
your powr is sufficient i said
cpu_time_overall=8.24338 cpu_time_mfcc=0.00832 cpu_time_infer=8.23506
alex@portable-alex:~/tmp/deepspeech/sycl$ ./deepspeech ../models/prod.rw.pbmm ../models/alphabet.txt ../audio/ -t
2018-02-26 17:43:27.138274: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-02-26 17:43:27.144501: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-02-26 17:43:27.144695: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
Running on directory ../audio/
> ../audio//2830-3980-0043.wav
experience proves tis
cpu_time_overall=6.44991 cpu_time_mfcc=0.00566 cpu_time_infer=6.44425
> ../audio//4507-16021-0012.wav
why should one halt on the way
cpu_time_overall=9.27064 cpu_time_mfcc=0.00751 cpu_time_infer=9.26313
> ../audio//8455-210777-0068.wav
your powr is sufficient i said
cpu_time_overall=9.05054 cpu_time_mfcc=0.01019 cpu_time_infer=9.04034
```

lissyx, 2018-02-26:
Comparison CPU vs GPU:
```
alex@portable-alex:~/tmp/deepspeech/cpu$ ./deepspeech ../models/prod.rw.pbmm ../models/alphabet.txt ../audio/2830-3980-0043.wav -t
2018-02-26 16:29:14.686308: I tensorflow/core/platform/cpu_feature_guard.cc:137] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
experience proves tis
cpu_time_overall=10.19238 cpu_time_mfcc=0.00316 cpu_time_infer=10.18921
```

```
alex@portable-alex:~/tmp/deepspeech/sycl$ ./deepspeech ../models/prod.rw.pbmm ../models/alphabet.txt ../audio/2830-3980-0043.wav -t
2018-02-26 17:43:57.252672: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-02-26 17:43:57.254221: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-02-26 17:43:57.254269: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
experience proves tis
cpu_time_overall=6.47814 cpu_time_mfcc=0.00848 cpu_time_infer=6.46965
```

lissyx, 2018-02-26:
@MichalMrozek Any suggestion on the values for `OverrideKmdNotifyDelayMs` that I shoud give a test with?

MichalMrozek, 2018-02-26:
Yes , this is not affecting how kernels are compiled

lissyx, 2018-02-26:
@MichalMrozek Okay, I'll first evaluate the impact of `OverrideKmdNotifyDelayMs` :)

lissyx, 2018-02-26:
@MichalMrozek I'm nearly half way through the values of `OverrideKmdNotifyDelayMs`. It's taking time, each rebuild of the driver takes several minutes, then I run 10 iterations of the benchmark code, and I sometime have to re-run it because Firefox Nightly with WebRender polluted the results :). But I might be able to finish that tomorrow morning (or earlier?)

MichalMrozek, 2018-02-26:
I really appreciate your help on this, this data will be very valuable for the further driver development.
There is no rush though, take your time. It's better to have reliable result, then to deliver them faster :)

btw. Do you build just igdrcl_dll target? (cmd "make -j 8 igdrcl_dll")

lissyx, 2018-02-26:
@MichalMrozek I'm rebuilding everything, to be sure :). Here are the results:
```
$ ls -v sycl_OverrideKmdNotifyDelayMs_* | xargs tail -vn +1
==> sycl_OverrideKmdNotifyDelayMs_1.csv <==
"model","mean","std"
"/home/alex/tmp/deepspeech/benchmark_sycl/run_bench/output_graph.pbmm",5.990497,0.195783

==> sycl_OverrideKmdNotifyDelayMs_2.csv <==
"model","mean","std"
"/home/alex/tmp/deepspeech/benchmark_sycl/run_bench/output_graph.pbmm",6.679939,0.391698

==> sycl_OverrideKmdNotifyDelayMs_3.csv <==
"model","mean","std"
"/home/alex/tmp/deepspeech/benchmark_sycl/run_bench/output_graph.pbmm",6.835617,0.279802

==> sycl_OverrideKmdNotifyDelayMs_4.csv <==
"model","mean","std"
"/home/alex/tmp/deepspeech/benchmark_sycl/run_bench/output_graph.pbmm",7.861745,0.115210

==> sycl_OverrideKmdNotifyDelayMs_5.csv <==
"model","mean","std"
"/home/alex/tmp/deepspeech/benchmark_sycl/run_bench/output_graph.pbmm",8.138877,0.124721

==> sycl_OverrideKmdNotifyDelayMs_8.csv <==
"model","mean","std"
"/home/alex/tmp/deepspeech/benchmark_sycl/run_bench/output_graph.pbmm",8.701115,0.187568

==> sycl_OverrideKmdNotifyDelayMs_10.csv <==
"model","mean","std"
"/home/alex/tmp/deepspeech/benchmark_sycl/run_bench/output_graph.pbmm",9.130474,0.145828

==> sycl_OverrideKmdNotifyDelayMs_15.csv <==
"model","mean","std"
"/home/alex/tmp/deepspeech/benchmark_sycl/run_bench/output_graph.pbmm",10.201564,0.148961

==> sycl_OverrideKmdNotifyDelayMs_20.csv <==
"model","mean","std"
"/home/alex/tmp/deepspeech/benchmark_sycl/run_bench/output_graph.pbmm",11.206903,0.162768

==> sycl_OverrideKmdNotifyDelayMs_50.csv <==
"model","mean","std"
"/home/alex/tmp/deepspeech/benchmark_sycl/run_bench/output_graph.pbmm",11.899895,0.148617

==> sycl_OverrideKmdNotifyDelayMs_100.csv <==
"model","mean","std"
"/home/alex/tmp/deepspeech/benchmark_sycl/run_bench/output_graph.pbmm",12.078021,0.181479
```

I'll give a try to the two other parameters tomorrow :)

lissyx, 2018-02-27:
```
$ cat sycl_EnableStatelessToStatefulBufferOffsetOpt.csv 
"model","mean","std"
"/home/alex/tmp/deepspeech/benchmark_sycl/run_bench/output_graph.pbmm",11.710195,0.117394
```

lissyx, 2018-02-27:
@MichalMrozek Forcing 32 bits addressing is a huge failure, lots of segfault, and when it does not segfault, it's slower and the output is wrong:
```
alex@portable-alex:~/tmp/deepspeech/benchmark_sycl/run_bench$ ./deepspeech ../../models/prod.rw.pbmm ../../models/alphabet.txt ../../audio/2830-3980-0043.wav -t 
2018-02-27 11:03:47.900603: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-02-27 11:03:47.908048: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-02-27 11:03:47.908172: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
sgsjpsjsopfj'sz'so'sovlq'dyoxq ugrynughuwlpwfpn'vhcphsims'v pelyeahx bzrhyxu lri lfbkd vo
cpu_time_overall=13.54337 cpu_time_mfcc=0.00574 cpu_time_infer=13.53763
alex@portable-alex:~/tmp/deepspeech/benchmark_sycl/run_bench$ ./deepspeech ../../models/prod.rw.pbmm ../../models/alphabet.txt ../../audio/2830-3980-0043.wav -t 
2018-02-27 11:04:02.329405: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-02-27 11:04:02.331694: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-02-27 11:04:02.331736: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
Erreur de segmentation (core dumped)
alex@portable-alex:~/tmp/deepspeech/benchmark_sycl/run_bench$ LC_ALL=C ./deepspeech ../../models/prod.rw.pbmm ../../models/alphabet.txt ../../audio/2830-3980-0043.wav -t 
2018-02-27 11:04:06.245003: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-02-27 11:04:06.250872: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-02-27 11:04:06.250967: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
Erreur de segmentation (core dumped)
alex@portable-alex:~/tmp/deepspeech/benchmark_sycl/run_bench$ LC_ALL=C ./deepspeech ../../models/prod.rw.pbmm ../../models/alphabet.txt ../../audio/2830-3980-0043.wav -t 
2018-02-27 11:04:08.692508: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-02-27 11:04:08.697315: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-02-27 11:04:08.697350: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
Erreur de segmentation (core dumped)
alex@portable-alex:~/tmp/deepspeech/benchmark_sycl/run_bench$ LC_ALL=C ./deepspeech ../../models/prod.rw.pbmm ../../models/alphabet.txt ../../audio/2830-3980-0043.wav -t 
2018-02-27 11:04:10.357139: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-02-27 11:04:10.365084: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-02-27 11:04:10.365234: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
v 
cpu_time_overall=12.56649 cpu_time_mfcc=0.00659 cpu_time_infer=12.55990
alex@portable-alex:~/tmp/deepspeech/benchmark_sycl/run_bench$ LC_ALL=C ./deepspeech ../../models/prod.rw.pbmm ../../models/alphabet.txt ../../audio/2830-3980-0043.wav -t 
2018-02-27 11:04:19.315712: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-02-27 11:04:19.317499: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-02-27 11:04:19.317528: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
Erreur de segmentation (core dumped)
alex@portable-alex:~/tmp/deepspeech/benchmark_sycl/run_bench$ LC_ALL=C ./deepspeech ../../models/prod.rw.pbmm ../../models/alphabet.txt ../../audio/2830-3980-0043.wav -t 
2018-02-27 11:04:22.311129: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-02-27 11:04:22.317677: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-02-27 11:04:22.318282: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
Erreur de segmentation (core dumped)
alex@portable-alex:~/tmp/deepspeech/benchmark_sycl/run_bench$ LC_ALL=C ./deepspeech ../../models/prod.rw.pbmm ../../models/alphabet.txt ../../audio/2830-3980-0043.wav -t 
[... cut ...]
2018-02-27 11:04:28.591278: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-02-27 11:04:28.595716: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-02-27 11:04:28.595881: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
juf 'rgi df
cpu_time_overall=12.60867 cpu_time_mfcc=0.00579 cpu_time_infer=12.60288
alex@portable-alex:~/tmp/deepspeech/benchmark_sycl/run_bench$ 
```

MichalMrozek, 2018-02-27:
Thank you very much for all those experiments.

It looks like this application really likes going into GPU ( or Kernel Mode Driver ) notify mechanism.
In short this mechanism is responsible for what happens when we wait for GPU, if mechanism is ON then instead of a busy CPU loop, we put waiting thread to sleep, it will be resumed when GPU will finish processing workloads. This mechanism saves power, but adds completion latencies. 

I have changed our implementation a little.
With commit cd747b7b8c152c89c2a75f64c6304c3d16f6a6db variable 
OverrideKmdNotifyDelayMs is renamed to OverrideKmdNotifyDelayMicroseconds
It also works dfferently as it now assumes input is in microseconds instead of miliseconds.
That gives much better precision.

You may now test with even lower values as previous "1" is now 1000.
We are also exploring internally to turn the KMD Notify mechanism ON by default. Right now it is disabled on all Linux drivers. As you see it requires a lot of fine tuning, as performance may fluctuacte.

If you are eager to test more, feel free with values lower then 1000. This data will help us a lot.

lissyx, 2018-02-28:
@MichalMrozek Thanks for that message, I was PTO yesterday but I already gave a try to experimenting with the microseconds resolution. However, testing with values 25µs, 250µs and 750µs I could not expose any visible improvement on our usecase. Maybe it does not help, maybe it is just our model that likes values ~1ms. Maybe it is also because it's just an i7-5600U GPU, not super high end recent hardware, but overall, no big change, we are still within the zone of 5.9-6.5 secs for inference time.

MichalMrozek, 2018-03-01:
Looks like 1 ms may be a sweet spot on BDW.

We have recently fixed the ICD problems.
Can you try with default Ubuntu version of ICD to see if application runs correctly?
Instructions are also updated.

For performance analysis it looks like ContextHintLevel may not be working correctly.
Can you try intercept layer with value 1,2,4 ?

You should be able to receive strings with performance suggestions from the intercept layer.

Example:

Performance hint: clCreateBuffer with pointer 0x55a5b9128980 and size 32 doesn't meet alignment restrictions. Size should be aligned to 4096 bytes and pointer should be aligned to 4096. Buffer is not sharing the same physical memory with CPU.

If this will not work , can you file an issue in here :
https://github.com/intel/opencl-intercept-layer

Thanks !

bashbaug, 2018-03-02:
> You should be able to receive strings with performance suggestions from the intercept layer.

Note: The controls you should set to see performance hints via the intercept layer are:

````
ContextHintLevel = 255
ContextCallbackLogging = 1
````

The first control causes the context to be created with performance hints enabled, therefore the driver will call the context callback with performance hint information.  The second causes the calls to the context callback to be included in the intercept layer log.  If this doesn't work for some reason, please do file an issue against the intercept layer and I'll take a look - thanks!

lissyx, 2018-03-02:
@MichalMrozek First element, it seems the ICD loader is working :-), I've removed the khronos icd build I made, updated and rebuilt your driver's package, installed, and it works. There's no extra `libOpenCL*.so` in /opt/intel/ :). I'm looking at the other elements and get back asap

lissyx, 2018-03-02:
@MichalMrozek @bashbaug 
```
$ LD_LIBRARY_PATH=/home/alex/codaz/Mozilla/DeepSpeech/Intel-Neo/cli/_bin CLI_DllName=/usr/lib/x86_64-linux-gnu/libOpenCL.so CLI_LogToFile=1 CLI_CallLogging=1 CLI_DevicePerformanceTiming=1 CLI_HostPerformanceTiming=1 CLI_ContextHintLevel=255 CLI_ContextCallbackLogging=1 ./deepspeech output_graph_streaming.pbmm ../models/alphabet.txt ../audio/2830-3980-0043.wav -t
2018-03-02 11:39:06.990211: I ./tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2018-03-02 11:39:06.992326: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:70] Found following OpenCL devices:
2018-03-02 11:39:06.992448: I ./tensorflow/core/common_runtime/sycl/sycl_device.h:72] id: 0, type: GPU, name: Intel(R) Gen8 HD Graphics NEO, vendor: Intel(R) Corporation, profile: FULL_PROFILE
 SYCL Exception: Performance hint: Driver calls internal clFlush on the command queue each time 1 command is enqueued.
 SYCL Exception: Performance hint: Profiling adds overhead on all enqueue commands with events.
 SYCL Exception: Performance hint: clCreateBuffer with pointer 0x5571288fde20 and size 8192 doesn't meet alignment restrictions. Size should be aligned to 4096 bytes and pointer should be aligned to 4096. Buffer is not sharing the same physical memory with CPU.
 SYCL Exception: Performance hint: clCreateBuffer needs to allocate memory for buffer. For subsequent operations the buffer will share the same physical memory with CPU.
 SYCL Exception: Performance hint: clEnqueueWriteBuffer call on a buffer 0x7f4b84004da0 require driver to copy the data. Consider using clEnqueueMapBuffer with buffer that shares the same physical memory with CPU.
Error detected Inside Sycl Device.
Error detected Inside Sycl Device.
Error detected Inside Sycl Device.
Error detected Inside Sycl Device.
Error detected Inside Sycl Device.
 SYCL Exception: Performance hint: clCreateBuffer with pointer 0x7f4b7c001220 and size 8192 doesn't meet alignment restrictions. Size should be aligned to 4096 bytes and pointer should be aligned to 4096. Buffer is not sharing the same physical memory with CPU.
 SYCL Exception: Performance hint: clCreateBuffer needs to allocate memory for buffer. For subsequent operations the buffer will share the same physical memory with CPU.
Error detected Inside Sycl Device.
Error detected Inside Sycl Device.
 SYCL Exception: Performance hint: clCreateBuffer with pointer 0x7f4b7c004000 and size 8192 meets alignment restrictions and buffer will share the same physical memory with CPU.
Error detected Inside Sycl Device.
Error running session: Internal: Unknown error detected on device /job:localhost/replica:0/task:0/device:SYCL:0
Erreur de segmentation (core dumped)
```

lissyx, 2018-03-02:
@MichalMrozek When you state
> For performance analysis it looks like ContextHintLevel may not be working correctly.
> Can you try intercept layer with value 1,2,4 ?
> You should be able to receive strings with performance suggestions from the intercept layer.

Does it means: `CLI_ContextHintLevel=1`, then `CLI_ContextHintLevel=2` and then `CLI_ContextHintLevel=4` ?

[35 more replies]

closed 2018-04-14 by lissyx