intel/compute-runtime #23: segfault during compilation

crepererum, 2018-03-05:
The following build step (and other similar steps) crash:

```
.../compute-runtime/build/bin/cloc -q -file copy_buffer_rect.igdrcl_built_in -device cfl -64 -out_dir .../compute-runtime/
build/bin/built_ins/x64/cfl -cpp_file -options -cl-kernel-arg-info
```

Backtrace:

```
#0  0x00007f0585cd2c62 in llvm::MemoryBuffer::getMemBufferRef() const () from .../compute-runtime/build/bin/libigdccl.so
#1  0x00007f05850cf8ca in TC::TranslateBuild (pInputArgs=pInputArgs@entry=0x7ffdcd9729d0, pOutputArgs=pOutputArgs@entry=0x7ffdcd9729a0, 
    inputDataFormatTemp=<optimized out>, IGCPlatform=..., profilingTimerResolution=<optimized out>)
    at .../compute-runtime/igc/IGC/AdaptorOCL/dllInterfaceCompute.cpp:729
#2  0x00007f0585174408 in IGC::IgcOclTranslationCtx<0ul>::Impl::Translate (this=<optimized out>, outVersion=<optimized out>, src=<optimized out>, 
    options=<optimized out>, internalOptions=<optimized out>, tracingOptions=<optimized out>, tracingOptionsCount=<optimized out>)
    at .../compute-runtime/igc/IGC/AdaptorOCL/ocl_igc_interface/impl/igc_ocl_translation_ctx_impl.h:173
#3  0x0000557a3d0ceb45 in OCLRT::OfflineCompiler::buildSourceCode() ()
#4  0x0000557a3d0d43d2 in OCLRT::OfflineCompiler::build() ()
#5  0x0000557a3d0cd66c in main ()
```

ArturHarasimiuk, 2018-03-06:
@crepererum could you make one experiment by running failing command under the debugger and provide list of loaded shared libraries? under gdb/cgdb this is command `info shared`.

could also you provide more info on distribution you are using?

crepererum, 2018-03-06:
Shared library data:
```
(gdb) info shared
From                To                  Syms Read   Shared Object Library
0x00007f05878afe30  0x00007f05878b3047  Yes (*)     /usr/lib/librt.so.1
0x00007f05876aad40  0x00007f05876aba0e  Yes (*)     /usr/lib/libdl.so.2
0x00007f0587491670  0x00007f058749f5a1  Yes (*)     /usr/lib/libpthread.so.0
0x00007f058718dcb0  0x00007f0587241c49  Yes         /usr/lib/libstdc++.so.6
0x00007f0586dc0640  0x00007f0586e5f206  Yes (*)     /usr/lib/libm.so.6
0x00007f0586ba4800  0x00007f0586bb4bd5  Yes         /usr/lib/libgcc_s.so.1
0x00007f058680b790  0x00007f058694d66c  Yes (*)     /usr/lib/libc.so.6
0x00007f0587ab6db0  0x00007f0587ad3390  Yes (*)     /lib64/ld-linux-x86-64.so.2
0x00007f0586570dc0  0x00007f05865d3d4c  Yes         .../compute-runtime/build/bin/libigdfcl.so
0x00007f05863441f0  0x00007f0586351157  Yes (*)     /usr/lib/libz.so.1
0x00007f058507c580  0x00007f0585dc2f91  Yes         .../compute-runtime/build/bin/libigdccl.so
0x00007f0581ee3e00  0x00007f0584006af9  Yes (*)     .../compute-runtime/build/bin/libcommon_clang.so
0x00007f058168c0b0  0x00007f05816c6728  Yes (*)     /usr/lib/libncursesw.so.6
(*): Shared library is missing debugging information.
```

(as in the stacktrace above I've changed the path prefix for the source dir to `...`)

Distribution is Arch Linux.

ArturHarasimiuk, 2018-03-07:
thank you for additional information. we were able to reproduce issue and analyze it. It looks like there maybe problem with clang version used to build Intel Graphics compiler stack. Everything worked for us with 4.0.1 and failed with 5.0

could you confirm clang version you use. is this 5.0?

crepererum, 2018-03-07:
`5.0.1` to be precise, see https://www.archlinux.org/packages/extra/x86_64/clang/

ArturHarasimiuk, 2018-03-09:
I created issue in Intel Graphics Compiler because this is issue with compiler stack.

MichalMrozek, 2018-06-20:
Looks like problem is fixed on compiler side, closing this one.

closed 2018-06-20 by MichalMrozek