intel/compute-runtime #27: Support new SPIR-V 1.3 subgroup features in OpenCL also?

oscarbg, 2018-03-12:
Hi,
don't know if a dumb question as perhaps needs some Khronos effort to expose new Subgroups functionality available in Vulkan 1.1 and exposed in SPIR-V 1.3 in OpenCL 2.x also..
don't know exactly how compares to SPV_intel_subgroups but seems rich enough and cross-vendor ext. so would also nice to have support for it..
in has basic, vote,shuffle, shuffle relative, quad ops, etc.. and they are already supported by Intel Anvil Vulkan driver on Mesa (you can see here:https://twitter.com/oscarbg81/status/971748592372285440) if I posted twitter link correctly..
so it's not a matter of Intel GPU hardware support..
thanks..

jchodor, 2018-03-24:
That's definitely not a dumb question as having Vulkan SPV subgroup features in OCL could be useful. However, this would require some effort on our GPU compiler side (SPV reader, etc.) and is not currently available. 
For the time being, have you tried using OCL 2.1 subgroups (core OCL feature) with SPV (which should be already available)?

oscarbg, 2018-03-29:
yep.. but seems shuffle instructions are missing.. intel subgroups extensions supports it.. but mostly interested also on SPIR-V support and that's why asked previously about SPV_intel_subgroups extensions..
but now seems better to support the new standard SPIR-V 1.3 subgroups extensions altough understand may take some time..

jchodor, 2018-04-19:
good news on SPV_intel_subgroups - support for this extension was added to our graphics compiler : https://github.com/intel/intel-graphics-compiler/commit/dbb9f9f194473f1d33bad3808ee47fc9fa429499#diff-c9f9f38fbdf6b0e38ae7cf69d27db022.

oscarbg, 2018-04-19:
thanks @jchodor..
wouldn't have noticed from commit message!
seems is in master for a couple of weeks already..
might assume doesn't need additional enablement from "compute-runtime" any other than using up to date intel/intel-graphics-compiler, right?
sadly latest prebuilt binaries (2018ww15-010713) still using intel/intel-graphics-compiler@cf24afc which is older than needed..
hope next week prebuilt binaries catch up!

jchodor, 2018-04-23:
Using updated compiler should be enough. Please let us know if you encounter any issues.

MichalMrozek, 2018-04-25:
Looks like new release is available
https://github.com/intel/compute-runtime/releases/tag/2018ww16-010750
@oscarbg can we close this issue ?