intel/compute-runtime #30: Neo does not support non-uniform work group sizes?

vvoznesensky, 2018-03-28:
Hello.

clEnqueueNDRangeKernel returns "CL_INVALID_WORK_GROUP_SIZE if local_work_size is specified and number of work-items specified by global_work_size is not evenly divisible by size of work-group given by local_work_size"

I use Intel Core i3-6100 processor.

zzdanowicz, 2018-03-28:
Hi,
Can you tell us how you build the kernel you have enqueued using clEnqueueNDRangeKernel call? I would like to know the value of _options_ argument of clBuildProgram call.
Can you provide values of arguments _work_dim_ _global_work_size_ _local_work_size_ of clEnqueueNDRangeKernel call?

According to OpenCL spec, You can provide non-uniform _local_work_size_ only when program is built as OpenCL2.0 and by default clBuildProgram is using OpenCL1.2.
Can you try to add to options string "–cl-std=CL2.0" if you haven't and give more information if it helps or not?
Please refer also to clBuildProgram call page for more details: https://www.khronos.org/registry/OpenCL/sdk/2.1/docs/man/xhtml/clBuildProgram.html

vvoznesensky, 2018-03-29:
Thank you Zbigniew, I have misunderstood clBuildProgram semantics.

closed 2018-03-29 by vvoznesensky