intel/compute-runtime #33: clGetDeviceInfo(CL_DEVICE_VERSION) demands too large of a buffer

allanmac, 2018-04-11:
The spec says that the CL_INVALID_VALUE will be returned if the size is less than the declared return type.

The NEO driver (.5018) on Windows 10 returns CL_INVALID_VALUE if the buffer isn't at least 16 bytes which happens to be large enough for the complete device string: "OpenCL 2.1 NEO " + NUL.

I expect clGetDeviceInfo(char[]) to record a trailing NUL but otherwise work on a buffer of 1 byte or more.

Use case: I only want to know obtain the *major.minor* values.

Trimmed down example:

```
  struct {
    cl_uchar opencl_space[7]; // "OpenCL_"
    cl_uchar major;
    cl_uchar dot;
    cl_uchar minor;
#if 1 // Intel NEO is too restrictive
    cl_uchar space;
    cl_uchar vendor[32];
#endif
  } device_version;

  cl(GetDeviceInfo(device_id,
                   CL_DEVICE_VERSION,
                   16, // sizeof(device_version),
                   &device_version,
                   NULL));

```

MichalMrozek, 2018-04-12:
Currently when driver is being queried for CL_DEVICE_VERSION size it will return 16.
If I understand correctly you want to pass value lower then 16 to be populated.
I don't think this is valid according to the specification:
"param_value_size specifies the size in bytes of memory pointed to by param_value. This size in
bytes must be >= size of return type specified in table 4.3." 
And here we return 16.
So any value lower then 16 is not valid.

Driver is expected to return an error here
"CL_INVALID_VALUE if param_name is not one of the supported values or if size in
bytes specified by param_value_size is < size of return type as specified in table 4.3 and
param_value is not a NULL value or if param_name is a value that is available as an
extension and the corresponding extension is not supported by the device."

Driver version can have vendor specific information

"OpenCL version string. Returns theOpenCL version supported by the device.This version string has the followingformat:OpenCL<space><major_version.minor_version><space><vendor-specificinformation"

Unfortunately to support your case we would need to break specification requirements, which I don't think is easily doable without extension/spec changes and if we just relax this, we may have problems to certify.

allanmac, 2018-04-12:
I read the spec as the return type is "char[]" implying the array should be at least 1 byte.

Similarly, params with a return type of "cl_uint" require at least 4 bytes.

The problem with your interpretation of the spec is that a vendor's OpenCL device could have an exceptionally long version string and it would break existing binaries.

MichalMrozek, 2018-04-13:
> The problem with your interpretation of the spec is that a vendor's OpenCL device could have an exceptionally long version string and it would break existing binaries.

OpenCL calls can be used to obtain the size that needs to be allocated, see commit a6459b0496d2b7eeb25dd8bcd19d8c69f2d289a7.

First call is obtaining the size and second call is passing proper size.
```
getDeviceInfo(
        CL_DEVICE_VERSION,
        0,
        nullptr,
        &sizeReturned);
```
now with known size proper char array can be allocated:

std::unique_ptr<char[]> deviceVersion(new char[sizeReturned]);

and passed to getDeviceInfo

```
 getDeviceInfo(
        CL_DEVICE_VERSION,
        sizeReturned,
        deviceVersion.get(),
        nullptr);
```

I believe with this approach you can still use your wrapper, as you can copy from deviceVersion bytes that you need.

allanmac, 2018-04-13:
Yes, I understand the "size before reading" approach.  

I think the OpenCL spec should be more precise about when the CL_INVALID_VALUE error is returned.

Instead of:

> ... is less than size of return type as shown in the table above ...

It should read:

> ... is less than size of return type as shown in the table above or the actual size in bytes of data being queried by param_value ...

closed 2018-04-13 by allanmac