intel/compute-runtime #34: Any ETA for Fine grained SVM on Skylake?

oscarbg, 2018-04-12:
Hi,
thanks first for providing precompiled binaries in a weekly base!
I see you keep adding "Fine grained SVM is not supported in this release" in all releases..
it's coming soon? Any ETA to share?

thanks..

MichalMrozek, 2018-04-12:
This support is dependent on the Kernel patch.
We have infrastructure in Neo ready to turn it ON, we just need properly patched Kernel.

See https://patchwork.kernel.org/patch/10292273/ for Kernel patch and our pull request for Neo support https://github.com/intel/compute-runtime/pull/29

oscarbg, 2018-04-12:
thanks for details..
when have time will try patching kernel and compiling..

oscarbg, 2018-04-12:
Forgot to add that hope Neo patch gets merged so I can test with patched kernel and weekly Neo builds.. as building Neo requires over 10GB free space and even more time than kernel compile..
How much delay between a commit is merged and appearing on a weekly build?

MichalMrozek, 2018-04-16:
We are currently doing weekly releases.
In worst case scenario you may expect up to 2 weeks delay between commit merge and release ( commit is merged right after a release candidate is selected )
Normally delay time will be much shorter, i.e. last release is basing on the commit that was merged 3 days ago (2ebd20d29e87cd208c231acfcd7d9a1808c00d5e)

oscarbg, 2018-04-18:
thanks for info..

oscarbg, 2018-04-19:
Hi @MichalMrozek,
I was pointed out that:
"good news on SPV_intel_subgroups - support for this extension was added to our graphics compiler : intel/intel-graphics-compiler@dbb9f9f"
seems is in master ( intel/intel-graphics-compiler) for a couple of weeks already..
sadly latest prebuilt binaries (2018ww15-010713) still using intel/intel-graphics-compiler@cf24afc which is older than needed for testing..
can update intel-graphics-compiler@dbb9f9f or newer ASAP in following weeks prebuilt binaries so can test soon without having to build Neo driver myself?
thanks..

oscarbg, 2018-04-23:
Great work guys! Thanks..
Will test it soon.. also will keep an eye on Windows driver hope it gets support also within a month or two..

MichalMrozek, 2018-04-25:
New release is available 
https://github.com/intel/compute-runtime/releases/tag/2018ww16-010750

AdamCetnerowski, 2018-05-21:
To address your original question:

We are working on getting the coherency control patch into the upstream kernel. We don't have a specific date for that to happen. Once it does, we will add the required code to NEO without delay. 

Let us know, if you prefer to keep this issue open. Otherwise we'll close it.

oscarbg, 2018-05-23:
Hi,
nice to know..
I close it.. 
thanks..

closed 2018-05-23 by oscarbg