intel/compute-runtime #37: Add a reported compatibility matrix.

anadon, 2018-04-25:
While only Ubuntu 16.04 is targeted, could a compatibility matrix or chart be added to track how it is working on other platforms?

PiotrRozenfeld, 2018-04-27:
At this moment, we only validate on Ubuntu 16.04.3 LTS. Validation under other distros is planned in the future (in limited fashion).
 
For the purposes of the compilation, we test several versions of gcc and clang, as mentioned [here](https://github.com/intel/compute-runtime/blob/master/documentation/RELEASES.md#quality-expectations).
 
Example distributions we use for compilation check include: 
•	Ubuntu 16.04.3
•	Fedora 26/27
•	CentOS 7.4 (with devtoolset-4-gcc-c++)

anadon, 2018-04-27:
Yes, but crowd-sourcing functional compatibility is a useful tool to track progress of development as well as being useful to users to know if they are likely to get Neo to work on their system or know that it won't work.  It is a few minutes of work which allows gathering of some very useful information.

PiotrRozenfeld, 2018-04-28:
We would like to avoid publishing information that may degrade over time. 
Example: we know that latest NEO driver builds with Arch Linux, but this may not be true moving forward unless we add periodic verification. The codebase keeps evolving.
 
Instead, we are considering adding compatibility information for weekly releases. Example: we may declare that 2018ww18 release and a corresponding set of tagged sources builds on a specific list of distributions because we have actually done it.

anadon, 2018-04-28:
OK, that makes sense.

JacekDanecki, 2018-05-25:
I think I can close this issue, as you can find initial compatibility matrix in case of Neo building on Travis: https://travis-ci.org/intel/compute-runtime. Other configurations were mentioned in [this](https://github.com/intel/compute-runtime/issues/38) issue, and our intention to provide binary packages for different distributions was presented [here](https://github.com/intel/compute-runtime/issues/46). Feel free to reopen this issue or fill new one when needed.

closed 2018-05-25 by JacekDanecki