intel/compute-runtime #38: Looking at the contribution guidlines, travisCI would be a good idea

anadon, 2018-04-25:
Looking here: https://github.com/intel/compute-runtime/blob/master/documentation/RELEASES.md

it seems like adding travisCI integration would make like easier.

PiotrRozenfeld, 2018-04-27:
We agree with you, we are already looking into integration with Travis CI (or alternative).
Will keep the issue open until we either have initial integration deployed, or at least more elaborate plan added to FAQ.md

anadon, 2018-04-27:
Here are a few other continuous integration tools which you may be interested in then, like Appvoyer, codecov.io and coveralls.io .

anadon, 2018-05-18:
Cool!  With Arch, Ubuntu, and Fedora the three large families of distros appears to be represented.  Over time, it would still be good to expand on this but not as high of a priority.  

I will bring up appvoyer again because then you could do the same with Windows which would help save time on maintaining multiple highly similar code bases and allow them to be the same code base.  That in turn should free up developer resources to actually work on the parts you all care about more :)

JacekDanecki, 2018-05-18:
At this moment we don't support Neo compilation under Windows using open source components.
In case of tested compilers to compile Neo, at this moment we use gcc: 5, 6, 7 and clang 4, 5 with different minor versions. Compilation with gcc 8 and clang 6 fails, and is on our todo list.

grmat, 2018-05-22:
> Compilation with gcc 8 and clang 6 fails, and is on our todo list.

Could you open a separate bug for this, so it can easily tracked?

I was interested in neo and created a [PKGBUILD], the standard format shell script for automated package builds for Arch Linux. Right now, I'm also using gcc54, as that's already in the official repos, but it's not usual and has some kind of exceptional position because it's required for CUDA and might be dropped out of the repos again.

I also considered pushing it into the [AUR] (Arch User Repository), but have second thoughts after seeing your recent effort in automated builds. Do you consider officially supporting Arch Linux? Do you consider updating your CI build scripts with a PKGBUILD instead of a bunch of manually created build scripts?

[PKGBUILD]: https://github.com/grmat/arch-compute-runtime/blob/git/PKGBUILD
[AUR]: https://aur.archlinux.org/

anadon, 2018-05-23:
@grmat As mentioned above 
> At this moment 3 OS's are tested: Ubuntu 16.04, Ubuntu 18.04, Arch with all compilers we support now. Fedora is in queue.

Also, you should be using the gcc base package for Arch, not gcc54.  gcc will always be there and maintained, but gcc54 is for compatibility purposes.

Arch may work, but should not be officially supported since it is a rolling distribution and at regular times breaks itself.

@JacekDanecki There is a good suggestion here of building and testing off of a PKGBUILD.  There are some good practices there which will help packaging for other distributions in general, and make sure everything is being done in a way which conforms with many best practices.

grmat, 2018-05-23:
> Also, you should be using the gcc base package for Arch, not gcc54. gcc will always be there and maintained, but gcc54 is for compatibility purposes.

You basically just repeated what I already wrote. The sole reason 5.4 is used right now is that it doesn't build with gcc8, hence I was asking for the status.

> Arch may work, but should not be officially supported since it is a rolling distribution and at regular times breaks itself.

It does not break "itself"...

JacekDanecki, 2018-05-24:
Current CI scripts build Neo using IGC binaries from the latest release. This is why I did not create PKGBUILD as you've prepared.
At this moment full build on Travis is not possible, because of build time limits.
You don't need to clone gmock/gtest anymore as they are included in third_party/gtest directory.
Fixes to compile Neo with gcc-8 have been just merged, and Travis builds with gcc-8 are tested here: https://travis-ci.org/JacekDanecki/compute-runtime
Regarding support for Arch distribution, it is our intention to provide such, but we are working out a process, which would allow us to automate this.

JacekDanecki, 2018-05-25:
I'll close this issue, as initial integration with Travis was implemented and deployed. I'm still working on tuning it, so you can expect more changes soon. Please reopen it when needed or file new issue.

closed 2018-05-25 by JacekDanecki