intel/compute-runtime #39: build failure

anadon, 2018-04-25:
In the "Copying OCaml library component llvm_X86.cmx to intermediate are" invoked from `cmake -DBUILD_TYPE=Release -DCMAKE_BUILD_TYPE=Release ../neo` when following the build instructions gives the following error:

```
CMake Error at bindings/ocaml/llvm/cmake_install.cmake:49 (file):
  file cannot create directory: /usr/lib/ocaml/llvm.  Maybe need
  administrative privileges.
Call Stack (most recent call first):
  bindings/ocaml/cmake_install.cmake:42 (include)
  cmake_install.cmake:66 (include)

make[3]: *** [Makefile:129: install] Error 1
make[2]: *** [igc/IGC/BiFModule/clang_build/CMakeFiles/cclang.dir/build.make:131: igc/IGC/BiFModule/clang_build/src/src/cclang-stamp/cclang-install] Error 2
make[1]: *** [CMakeFiles/Makefile2:1515: igc/IGC/BiFModule/clang_build/CMakeFiles/cclang.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

which seems to imply that configuration requires root.  This looks like a minor bug, but it does cause fairly significant worry for packagers.

ArturHarasimiuk, 2018-04-25:
We shouldn't see this:
`
file cannot create directory: /usr/lib/ocaml/llvm
`
We don't need root during the build and we do these as regular user. Please provide more data so we can help with this:
- Which Linux distribution are you using? 
- Can you share complete build log? 
- Please provide root folder of your workspace.

anadon, 2018-04-25:
The problem is actually in the make stage, sorry.  Anyways, here are the configuration and build logs respectively:
https://gist.github.com/fe1293dc3daf33a0765793b731b44563
https://gist.github.com/658bc4d124fbb2d1ecb4eaa10b26efc4

ArturHarasimiuk, 2018-04-26:
thanks for data. We will investigate issue carefully and contact you with results.

ArturHarasimiuk, 2018-04-27:
I was able to reproduce issue. Could you please confirm which of these packages are installed on your system:
- llvm-ocaml
- ocaml
- ocaml-compiler-libs
- ocaml-ctypes
- ocaml-findlib
- ocamlbuild

For some reason presence of ocaml caused failure described in this issue.

anadon, 2018-04-27:
```
[anadon@doge ~]$ pacman -Q | grep 'ocaml'
ocaml 4.06.0-2
ocaml-compiler-libs 4.06.0-2
ocaml-ctypes 0.11.5-2
ocaml-findlib 1.7.3-2
```

anadon, 2018-04-27:
In either case, anything in /usr should be viewed as a system privileged write.  RHEL for instance will never upstream this if you require writing to that directory.  An option may to be to list it as a build dependency.  I will assert that the most pressing technical task for this project is to improve the condition of the build system and dependencies, even above core OpenCL bugs present.

ArturHarasimiuk, 2018-05-22:
Issue is within intel/opencl-clang component. I've created issue in this project.

I will close this issue. Please feel free to contact us in case of further questions or problems.

closed 2018-05-22 by ArturHarasimiuk