intel/compute-runtime #40: Build failure of test_kernel_Gen9lp with Debian 9 and Skylake GPU

CNugteren, 2018-04-29:
I'm on a Debian 9 system with an Intel Core i5-6200U CPU with a HD Graphics 520 (Skylake ULT GT2) GPU. I'm following the instructions for building on Ubuntu, except that I had to build Clang 4.0 myself from source.

I'm getting quite far with building the whole thing, but then I encounter the following issue (single-threaded make):
```
Scanning dependencies of target unit_tests
[ 91%] Built target unit_tests
[ 93%] Built target test_kernels_Gen9lp
[ 93%] Built target test_kernel_Gen9lp
[ 93%] Generating ../bin/Gen9lp/test_files/x64/-cl-kernel-debug-enable_Gen9lp.bc, ../bin/Gen9lp/test_files/x64/-cl-kernel-debug-enable_Gen9lp.bin, ../bin/Gen9lp/test_files/x64/-cl-kernel-debug-enable_Gen9lp.gen, ../bin/Gen9lp/test_files/x64/-cl-kernel-debug-enable_Gen9lp.dbg

error: undefined reference to `_Z12get_local_idj()'
undefined reference to `_Z12get_group_idj()'

error: backend compiler failed build.

Build failed with error code: -11
unit_tests/CMakeFiles/test_kernel_debug_enable_Gen9lp.dir/build.make:64: recipe for target 'bin/Gen9lp/test_files/x64/-cl-kernel-debug-enable_Gen9lp.bc' failed
make[2]: *** [bin/Gen9lp/test_files/x64/-cl-kernel-debug-enable_Gen9lp.bc] Error 245
CMakeFiles/Makefile2:5745: recipe for target 'unit_tests/CMakeFiles/test_kernel_debug_enable_Gen9lp.dir/all' failed
make[1]: *** [unit_tests/CMakeFiles/test_kernel_debug_enable_Gen9lp.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

Have you seen this before? Did you ever build the software on Debian? Please let me know if there is anything I can try or any further information you need.

JacekDanecki, 2018-04-30:
Can you check with components revisions from the latest release: https://github.com/intel/compute-runtime/releases/tag/2018ww16-010750
We've observed similar issue, when components we depend on, were not synchronized.
I'm preparing setup with Debian 9 just now, as we did not test building on Debian yet.

CNugteren, 2018-04-30:
I checked out the versions of the packages as you suggested, but now I get an error with CMake (3.7.2):
```
-- IGC source dir is: /home/cnugteren/Software/intel-neo/igc
CMake Error at /home/cnugteren/Software/intel-neo/igc/IGC/CMakeLists.txt:1895 (message):
  CMAKE_BUILD_TYPE variable is not allowed while using multi-configuration
  generator!
```
Anyway, I also saw a .deb package on the release page you pointed at. I installed that and it does seem to work.

kubrickfr, 2018-05-01:
I'm having the same issue on Arch Linux (undefined reference to `_Z12get_local_idj()')

JacekDanecki, 2018-05-02:
I've reproduced this issue on my side. It looks like the HEAD of IGC builds correctly with older Neo, but the HEAD of Neo builds with older IGC. I'm looking at this.
@CNugteren in case of cmake error you've observed after changing components versions, you need to clean your build directory and start new build.

JacekDanecki, 2018-05-02:
@kubrickfr As we don't provide (at this moment) package for Arch, you can recompile with versions I've mentioned earlier or you can use deb package from the latest release. To do this you need to install 2 packages from AUR:
https://aur.archlinux.org/packages/dpkg
https://aur.archlinux.org/packages/ncurses5-compat-libs
and then install Neo package using dpkg-deb tool:
dpkg-deb -x intel-opencl_2018ww16-010750_amd64.deb /
This package passed my smoke tests in docker image:
REPOSITORY                                 TAG                 IMAGE ID            CREATED             SIZE
docker.io/base/archlinux                   latest              0d0b67d096d2        6 days ago          726 MB

laolei, 2018-05-09:
I met the same issue on Ubuntu. I checked the failure cl. It's very simple. 
__attribute__((intel_reqd_sub_group_size(8)))
__kernel void
CopyBuffer(
    __global unsigned int* src,
    __global unsigned int* dst )
{
    int id = (int)get_global_id(0);
    dst[id] = src[id];
}
@JacekDanecki Do you have solutions now? thanks

JacekDanecki, 2018-05-21:
New fixes were merged and components with below revisions compile correctly:
* intel/intel-graphics-compiler@c670611c2605d3ebec0c2824f6d599beff55d2bd
* intel/compute-runtime@d2817427afd0aab8dc7248dd69b5fd298347e107
* intel/llvm-patches@7ac2c8a04178cf90365eeba15c63028a26409de0
* intel/opencl-clang@9b2473dc76bca8e58bf90e859a87f97ed1ca962c
* intel/gmmlib@247673f8e487d62b818876dc1905036e477986c1

JacekDanecki, 2018-05-23:
Closing this issue, as provided fixes resolved compilation issues. Feel free to reopen it when it doesn't work for you.

closed 2018-05-23 by JacekDanecki

kubrickfr, 2018-05-28:
I confirm in now compiles on Arch. However exporting images with darktable with opencl results in crashes all the time. I'll try to get a backtrace and open a new bug report.

JacekDanecki, 2018-05-28:
Do you observe crashes in another applications, or the problem is observed only in darktable?

kubrickfr, 2018-05-28:
@JacekDanecki luxmark works fine, editing in Darktable works, only exporting a whole image crashes.