intel/compute-runtime #41: Segfault in clReleaseProgram when using shared_ptr

CNugteren, 2018-05-01:
I'm on a Debian 9 system with an Intel Core i5-6200U CPU with a HD Graphics 520 (Skylake ULT GT2) GPU. I've installed the .deb driver from https://github.com/intel/compute-runtime/releases/tag/2018ww16-010750

I'm wrapping a `cl_program` object in a class with constructor/destructor. When I use that object with a `std::shared_ptr` and store it in a cache, I'm encountering a segfault in `clReleaseProgram` when cleaning-up. The exact same code works well using the old Intel driver and Intel's Beignet.

I've created a single-source ~100 line minimal reproducible [example here](https://gist.github.com/CNugteren/507708c19d46ffebd3a2ade1abb6610e). It can be compiled and ran e.g. as follows: `g++ --std=c++11 -g clReleaseProgram_segfault.cpp -o test -lOpenCL && ./test`. The first few lines have 3 defines: if any of them is enabled some small changes are made to the code and the segfault is gone. So it really seems to be a combination of three: `std::shared_ptr`, cache, and building the program.

The stack trace just points to `clReleaseProgram`, there are no debug symbols further in the libraries it seems:
```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff6757a30 in ?? () from /opt/intel/opencl/libigdrcl.so
#0  0x00007ffff6757a30 in ?? () from /opt/intel/opencl/libigdrcl.so
#1  0x00007ffff67582b0 in ?? () from /opt/intel/opencl/libigdrcl.so
#2  0x00007ffff6727ab2 in ?? () from /opt/intel/opencl/libigdrcl.so
#3  0x00007ffff672897e in ?? () from /opt/intel/opencl/libigdrcl.so
#4  0x00007ffff6728b29 in ?? () from /opt/intel/opencl/libigdrcl.so
#5  0x00007ffff6710bcb in ?? () from /opt/intel/opencl/libigdrcl.so
#6  0x0000555555555972 in Program::~Program (this=0x55555585dd30, __in_chrg=<optimized out>) at clReleaseProgram_segfault.cpp:39
#7  0x0000555555557ffa in __gnu_cxx::new_allocator<Program>::destroy<Program> (this=0x55555585dd30, __p=0x55555585dd30) at /usr/include/c++/6/ext/new_allocator.h:124
#8  0x0000555555557fcd in std::allocator_traits<std::allocator<Program> >::destroy<Program> (__a=..., __p=0x55555585dd30) at /usr/include/c++/6/bits/alloc_traits.h:487
(...)
#21 0x00007ffff6fb52e8 in __libc_start_main (main=0x5555555554e1 <main()>, argc=1, argv=0x7fffffffe0a8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, 
    stack_end=0x7fffffffe098) at ../csu/libc-start.c:325
#22 0x00005555555552ba in _start ()
```

Valgrind doesn't give much extra information. Anything I can still try? Hopefully with this example you can easily reproduce it and investigate.

kodonnell, 2018-05-02:
I'm also getting this on HD 6000. I tried it on a fresh ubuntu 18.04 install, without any other OpenCL drivers, and it still fails. (For a while I suspected it was an issue with multiple.)

CNugteren, 2018-05-02:
OK, with some help I found the issue here. It is actually because of the undefined order of destruction of global objects. The cache is one such object here, but Intel's driver also has some global objects. They get destructed earlier in this particular case, causing the `clReleaseProgram` to fail since the driver is already half destructed.

Adding a `cache.clear()` at the end of the main function solves this issue, because it will now call `clReleaseProgram` when returning from main, which is guaranteed before the driver clean-up, rather than being called when the globals are cleaned-up.

Not sure if there is an easy way out for you, but I see two main options: provide my users (this is for a library with a cache) with a clean-up function that clears the cache which they have to call before, or just never bother with calling `clReleaseProgram` at all.

yupinov, 2018-05-28:
FYI, have seen similar segfault, but caused by clReleaseKernel() call before even calling corresponding clReleaseProgram().

JacekDanecki, 2018-06-12:
This issue discovered problem we have not seen so far. We're working on fix, but it's not trivial.

MichalMrozek, 2018-06-28:
Hello @CNugteren 
I have applied series of patches to fix this problem.
Last one is b7545f14013e391f1fbac616ef09d855dfc97584.
Can you check if the problem is addressed?

CNugteren, 2018-07-02:
Thanks for working on this issue. Last time I got the compiled release, I didn't compile from source. I tried to know, but I could make it work without spending too much time on figuring out what versions of all the repositories I need exactly, running into issues now with the CMake of IGC. Perhaps you can just run the test example I provided again? Or even better: the tests of [CLBlast](https://github.com/CNugteren/CLBlast).

MichalMrozek, 2018-07-07:
Attached reproduces work correctly after the fixes.
Basing on that we can assume that problem is fixed.

closed 2018-07-07 by MichalMrozek