intel/compute-runtime #42: kernel: Failed to release pages: bind_count=1, pages_pin_count=1, pin_global=0

lissyx, 2018-05-03:
Trying to run on current master of the driver, ubuntu 18.04 (kernel 4.15), and using `mmap` for the file, I got that in the kernel logs:

```
May  3 14:32:33 portable-alex kernel: [ 6993.616308] ------------[ cut here ]------------
May  3 14:32:33 portable-alex kernel: [ 6993.616310] Failed to release pages: bind_count=1, pages_pin_count=1, pin_global=0
May  3 14:32:33 portable-alex kernel: [ 6993.616395] WARNING: CPU: 1 PID: 8071 at /build/linux-QLn4bB/linux-4.15.0/drivers/gpu/drm/i915/i915_gem_userptr.c:89 cancel_userptr+0xe8/0xf0 [i915]
May  3 14:32:33 portable-alex kernel: [ 6993.616396] Modules linked in: msr ccm pci_stub vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) xt_CHECKSUM iptable_mangle ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_nat_ipv4 nf_nat rfcomm nf_conntrack_ipv4 nf_defrag_ipv4 xt_conntrack nf_conntrack ipt_REJECT nf_reject_ipv4 xt_tcpudp bridge stp llc ebtable_filter ebtables ip6table_filter ip6_tables devlink cmac iptable_filter snd_hrtimer ipmi_devintf ipmi_msghandler bnep binfmt_misc nls_iso8859_1 arc4 intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp snd_soc_skl snd_soc_skl_ipc kvm_intel snd_hda_ext_core snd_soc_sst_dsp snd_soc_sst_ipc snd_hda_codec_hdmi snd_soc_acpi kvm uvcvideo videobuf2_vmalloc videobuf2_memops btusb videobuf2_v4l2 irqbypass cdc_mbim btrtl cdc_wdm crct10dif_pclmul snd_hda_codec_realtek crc32_pclmul cdc_ncm
May  3 14:32:33 portable-alex kernel: [ 6993.616427]  videobuf2_core ghash_clmulni_intel videodev btbcm snd_hda_codec_generic usbnet pcbc iwlmvm snd_soc_core cdc_acm btintel mii media mac80211 snd_compress ac97_bus snd_pcm_dmaengine bluetooth aesni_intel aes_x86_64 crypto_simd glue_helper cryptd ecdh_generic intel_cstate intel_rapl_perf iwlwifi idma64 virt_dma snd_seq_midi snd_hda_intel snd_seq_midi_event input_leds joydev cfg80211 snd_hda_codec serio_raw snd_hda_core snd_hwdep intel_lpss_pci thinkpad_acpi mei_me intel_wmi_thunderbolt wmi_bmof snd_pcm nvram snd_rawmidi mei processor_thermal_device intel_lpss shpchp intel_pch_thermal ucsi_acpi typec_ucsi intel_soc_dts_iosf snd_seq typec snd_seq_device snd_timer snd soundcore int3403_thermal int340x_thermal_zone int3400_thermal tpm_crb acpi_pad acpi_thermal_rel mac_hid sch_fq_codel parport_pc
May  3 14:32:33 portable-alex kernel: [ 6993.616457]  ppdev lp parport ip_tables x_tables autofs4 btrfs zstd_compress raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear uas usb_storage i915 i2c_algo_bit drm_kms_helper e1000e psmouse syscopyarea sysfillrect nvme sysimgblt ptp pps_core fb_sys_fops nvme_core thunderbolt drm wmi video
May  3 14:32:33 portable-alex kernel: [ 6993.616477] CPU: 1 PID: 8071 Comm: kworker/u16:2 Tainted: G           OE    4.15.0-21-generic #22-Ubuntu
May  3 14:32:33 portable-alex kernel: [ 6993.616478] Hardware name: LENOVO 20L7CTO1WW/20L7CTO1WW, BIOS N22ET34W (1.11 ) 03/13/2018
May  3 14:32:33 portable-alex kernel: [ 6993.616499] Workqueue: i915-userptr-release cancel_userptr [i915]
May  3 14:32:33 portable-alex kernel: [ 6993.616515] RIP: 0010:cancel_userptr+0xe8/0xf0 [i915]
May  3 14:32:33 portable-alex kernel: [ 6993.616516] RSP: 0000:ffffa45552e57e60 EFLAGS: 00010282
May  3 14:32:33 portable-alex kernel: [ 6993.616518] RAX: 0000000000000000 RBX: ffff8b3a5115fc00 RCX: 0000000000000006
May  3 14:32:33 portable-alex kernel: [ 6993.616518] RDX: 0000000000000007 RSI: 0000000000000082 RDI: ffff8b3ac1456490
May  3 14:32:33 portable-alex kernel: [ 6993.616519] RBP: ffffa45552e57e78 R08: 0000000000000001 R09: 0000000000000577
May  3 14:32:33 portable-alex kernel: [ 6993.616520] R10: ffffa45552e57e38 R11: 0000000000000000 R12: ffff8b3a5115fda8
May  3 14:32:33 portable-alex kernel: [ 6993.616521] R13: 0000000000000000 R14: ffff8b38e1ad0700 R15: 0000000000000000
May  3 14:32:33 portable-alex kernel: [ 6993.616522] FS:  0000000000000000(0000) GS:ffff8b3ac1440000(0000) knlGS:0000000000000000
May  3 14:32:33 portable-alex kernel: [ 6993.616523] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
May  3 14:32:33 portable-alex kernel: [ 6993.616523] CR2: 00007f8b333c3000 CR3: 00000004e0c0a001 CR4: 00000000003606e0
May  3 14:32:33 portable-alex kernel: [ 6993.616524] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
May  3 14:32:33 portable-alex kernel: [ 6993.616525] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
May  3 14:32:33 portable-alex kernel: [ 6993.616526] Call Trace:
May  3 14:32:33 portable-alex kernel: [ 6993.616530]  process_one_work+0x1de/0x410
May  3 14:32:33 portable-alex kernel: [ 6993.616532]  worker_thread+0x32/0x410
May  3 14:32:33 portable-alex kernel: [ 6993.616534]  kthread+0x121/0x140
May  3 14:32:33 portable-alex kernel: [ 6993.616535]  ? process_one_work+0x410/0x410
May  3 14:32:33 portable-alex kernel: [ 6993.616537]  ? kthread_create_worker_on_cpu+0x70/0x70
May  3 14:32:33 portable-alex kernel: [ 6993.616539]  ret_from_fork+0x35/0x40
May  3 14:32:33 portable-alex kernel: [ 6993.616540] Code: bf 46 ff ff eb c9 8b 93 c8 01 00 00 8b 8b a4 01 00 00 48 c7 c7 70 d2 61 c0 8b b3 9c 01 00 00 c6 05 f5 56 10 00 01 e8 38 85 94 ee <0f> 0b eb bc 0f 1f 40 00 0f 1f 44 00 00 55 ba 08 00 00 00 48 89 
May  3 14:32:33 portable-alex kernel: [ 6993.616566] ---[ end trace 5d962f9c324bf541 ]---
```

So far, a one-time message, but this might be a hint of something?

MichalMrozek, 2018-05-07:
Hello,
Thanks for report, this looks very similar to this:
https://bugs.freedesktop.org/show_bug.cgi?id=104512

Can you try on kernel 4.16.0-rcX ?

lissyx, 2018-05-07:
On this laptop, it might be a bit tricky, it's not packaged, so I'll have to build 4.16 or use one of the mainline kernel. But looking at the report, it might be just completely unrelated to the driver then. Maybe we should just close that then.

MichalMrozek, 2018-05-07:
I agree that closing may be a reasonable move, if it doesn't reproduce on 4.16 then most likely this is the same problem.

MichalMrozek, 2018-05-19:
Any luck with kernel 4.16?

lissyx, 2018-06-01:
I could not even reproduce that reliably after some kernel upgrades of 4.15. Closing, it's just useless noise.

closed 2018-06-01 by lissyx