intel/compute-runtime #43: Does Neo support work_group_reduce_min?

vvoznesensky, 2018-05-09:
Hello.

`1:16:24: error: implicit declaration of function 'work_group_reduce_min' is invalid in OpenCL
            dist_i_j = work_group_reduce_min(dist_i_j);
                       ^
1:16:24: note: did you mean 'sub_group_reduce_min'?
CTHeader.h:5863:39: note: 'sub_group_reduce_min' declared here
double  __attribute__((overloadable)) sub_group_reduce_min( double x );
                                      ^
`

MichalMrozek, 2018-05-09:
Hello,
This should be supported for OpenCL 2.0+ devices.
Do you compile your kernel with -cl-std=CL2.0 build option ?

vvoznesensky, 2018-05-09:
BTW, I use pyOpenCL.

It looks like the OpenCL compiler does not change it's behavior when adding -cl-std=CL2.0 option to build() arguments.

Anyway, sub_group_reduce_min is a OpenCL 2.0 feature and the compiler sees it, so does build option changes anything?

MichalMrozek, 2018-05-09:
Yes it does, if cl-std is not specific then it goes to 1.2 where this function is not supported.

According to spec:
"If the –cl-std build option is not specified, the highest OpenCL C 1.x language version supported by each device is used when compiling the program for each device. Applications are required to specify the –cl-std=CL2.0 option if they want to compile or build their programs with OpenCL C 2.0."

Can you try to compile your kernel outside of pyOpenCL ? 
Can you share code that triggers this error ?