intel/compute-runtime #44: Support for SPIR-V Specialization Constants

tremmelg, 2018-05-10:
Is there an estimate of when the Compute Runtime will support [SPIR-V specialization constants](https://www.khronos.org/registry/spir-v/specs/1.2/SPIRV.html#_a_id_specializationsection_a_specialization) and the corresponding [OpenCL 2.2 API](https://www.khronos.org/registry/OpenCL/specs/opencl-2.2.html#_setting_spir_v_specialization_constants)? Perhaps as an extension before OpenCL 2.2 support itself is released?

I would be interested in prototyping [SYCL specialization constants](https://github.com/codeplaysoftware/standards-proposals/blob/master/spec-constant/index.md) and evaluating their effect on runtime performance, code size and compilation time on Intel hardware.

PiotrRozenfeld, 2018-05-17:
We are exploring how to fit this into our current schedule. OpenCL 2.2 effort will start no sooner than in Q3. Once it begins, you will be able to see our progress as development will be done on master branch.

kmazurki, 2018-08-09:
We have a preliminary plan to implement spec constants in IGC till the end of Q4, This plan might be however a subject to change based on the progress in external dependencies.

PiotrRozenfeld, 2018-12-21:
This is still in our plans - IGC support should be available in Q1'19.

AdamCetnerowski, 2019-03-29:
IGC support has been added. Runtime piece will be provided in Q2'19

jchodor, 2019-06-20:
RT implementation is ready -  971eb7a1b4598491470216f26ede0cc8cf3e7993

jchodor, 2019-07-05:
Additional RT changes : https://github.com/intel/compute-runtime/commit/5cde27ed0309aa421722db77f8b42b54b02a67df

jchodor, 2019-07-05:
Let us know in case of problems with this feature.

closed 2019-07-05 by jchodor