intel/compute-runtime #45: CentOS 7.5 RPM Installation Fails

alheinecke, 2018-05-12:
Hi,

I checked out all the version you used in your latest release and built from scratch (your instructions are a bit incomplete as still the google libs for test needed) and everything goes through fine and without any issue. 

However installing the rpm-packages fails:
rpm -i intel-opencl-1.0-0.x86_64-igdrcl.rpm
	file /opt from install of intel-opencl-igdrcl-1.0.0-1.x86_64 conflicts with file from package filesystem-3.2-25.el7.x86_64
	file /etc/ld.so.conf.d from install of intel-opencl-igdrcl-1.0.0-1.x86_64 conflicts with file from package glibc-2.17-222.el7.x86_64

it seems to touch /opt and /etc/ld.so.conf.d directories, but they are already there. 

I did a --force during rpm install and it seems to work, however, it might be good to fix that.

ls /etc/ld.so.conf.d
atlas-x86_64.conf  dyninst-x86_64.conf  kernel-3.10.0-862.el7.x86_64.conf  libiscsi-x86_64.conf  mariadb-x86_64.conf  qt-x86_64.conf
rpm -i --force /nfs_home/aheineck/Misc/gen_stack/build/intel-opencl-1.0-0.x86_64-igdrcl.rpm
ls /etc/ld.so.conf.d
atlas-x86_64.conf  dyninst-x86_64.conf  kernel-3.10.0-862.el7.x86_64.conf  libintelopencl.conf  libiscsi-x86_64.conf  mariadb-x86_64.conf  qt-x86_64.conf

Thanks,
Alex

JacekDanecki, 2018-05-18:
Thanks for reporting. I'll look at this.

JacekDanecki, 2018-05-24:
I can't reproduce this issue on my side. I've executed compilation in fresh Centos 7.4 docker container upgraded to Centos 7.5. Logs from installation after build

Run CPack packaging tool...
CPack3: Create package using RPM
CPack3: Install projects
CPack3: - Run preinstall target for: igdrcl
CPack3: - Install project: igdrcl
CPack3: -   Install component: igdrcl
CPack3: Create package
CPackRPM: Will use GENERATED spec file: /root/build/_CPack_Packages/Linux/RPM/SPECS/intel-opencl-igdrcl.spec
CPack3: - package: /root/build/intel-opencl-1.0-0.x86_64-igdrcl.rpm generated.
[root@56f404e1e7a3 build]# rpm -i intel-opencl-1.0-0.x86_64-igdrcl.rpm 
[root@56f404e1e7a3 build]# rpm -q filesystem
filesystem-3.2-25.el7.x86_64
[root@56f404e1e7a3 build]# rpm -q glibc     
glibc-2.17-222.el7.x86_64
[root@56f404e1e7a3 build]# lsb_release -a
LSB Version:    :core-4.1-amd64:core-4.1-noarch
Distributor ID: CentOS
Description:    CentOS Linux release 7.5.1804 (Core) 
Release:        7.5.1804
Codename:       Core
[root@56f404e1e7a3 build]# rpm -qf /opt
filesystem-3.2-25.el7.x86_64
intel-opencl-igdrcl-1.0.0-1.x86_64
[root@56f404e1e7a3 build]# rpm -qf /etc/ld.so.conf.d/
glibc-2.17-222.el7.x86_64
intel-opencl-igdrcl-1.0.0-1.x86_64

For test I've used this commit:
[root@56f404e1e7a3 neo]# git log -1
commit 01c1df316462d5c51618629aad65b953128b3e12
Author: Woloszyn, Wojciech <wojciech.woloszyn@intel.com>
Date:   Thu May 24 02:43:58 2018 -0700

    Add missing case in getMipLevelOriginId

JacekDanecki, 2018-06-07:
Packages (rpm, deb) are created using CPack from cmake. What versions of cmake and rpm do you use?

alheinecke, 2018-06-07:
cmake --version
cmake3 version 3.11.0

rpm --version
RPM version 4.11.3

alheinecke, 2018-06-07:
[current_package_list.log](https://github.com/intel/compute-runtime/files/2081090/current_package_list.log)

JacekDanecki, 2018-06-12:
I've the same versions of cmake and rpm in docker container, so the problem is somewhere else. Need to investigate farther.

JacekDanecki, 2018-06-29:
I wasn't able to reproduce this issue on my side. Do you still observe this issue on the latest Neo versions? Can you share your rpm package? I've verified rpm packages on Centos 7.4/7.5, Fedora 27/28 and didn't see problems with installation.

cwt, 2018-08-06:
@JacekDanecki I just build 18.31.0 last night and have the same problem on CentOS 7.5

```
$ cat /etc/centos-release
CentOS Linux release 7.5.1804 (Core)
```

Below is the %files section from the generated SPEC `build/_CPack_Packages/Linux/RPM/SPECS/intel-opencl-igdrcl.spec`

```
%files
%defattr(-,root,root,-)
%dir "/etc/OpenCL"
%dir "/etc/OpenCL/vendors"
"/etc/OpenCL/vendors/intel.icd"
%dir "/etc/ld.so.conf.d"
"/etc/ld.so.conf.d/libintelopencl.conf"
%dir "/usr/local"
%dir "/usr/local/lib64"
"/usr/local/lib64/libiga64.so"
"/usr/local/lib64/libigc.so"
"/usr/local/lib64/libigdfcl.so"
"/usr/local/lib64/libigdrcl.so"
"/usr/local/lib64/libopencl_clang.so"
```

JacekDanecki, 2018-08-20:
I'll prepare fresh CentOS 7.5 setup to reproduce this issue.

JacekDanecki, 2018-09-03:
@johnhable Can you check this PR: https://github.com/intel/compute-runtime/pull/80?

johnhable, 2018-09-04:
Hi Jacek,

Thanks for the update. After applying the update, it builds and the rpm installs cleanly, so I think it's fixed.

John

JacekDanecki, 2018-09-05:
PR https://github.com/intel/compute-runtime/pull/80 was merged with commit: https://github.com/intel/compute-runtime/commit/b1d020485a7d1f7105a8c9db84c0fc12ff0a0e68, so I'm closing this issue.
If you find any related problems, please create new issue.

closed 2018-09-05 by JacekDanecki