intel/compute-runtime #46: Publish rpm in Github releases

adydychk, 2018-05-23:
Hi guys,
Do you have any plans to publish `.rpm` for CentOS in Github releases?

You have already `.deb` packages [here](https://github.com/intel/compute-runtime/releases/tag/2018ww19-010806) but it would be nice to have `.rpm` too.

AdamCetnerowski, 2018-05-24:
It is our intention to provide binary releases for multiple distros. We are working out a process, which would allow us to automate this. This will make it possible to scale the number of distros we support this way. 

Distros under consideration for first wave: Arch, Fedora, CentOS

Feel free to either close the issue, if this answer is satisfactory or keep it open until we start delivering CentOS binaries.

adydychk, 2018-05-24:
Thanks for the clear answer. I will keep it open until first CentOS `.rpm` drop :)

AdamCetnerowski, 2018-05-28:
Thank you for the suggestion. This looks like a good reason to add it to our short list.

adydychk, 2018-07-12:
Hi, 
any progress on this issue?

JacekDanecki, 2018-07-12:
Here are repositories with experimental (untested) rpm packages for Fedora 27/28 and Centos 7:

* https://copr.fedorainfracloud.org/coprs/arturh/intel-opencl - production quality
* https://copr.fedorainfracloud.org/coprs/arturh/intel-opencl-unstable - pre-release quality
* https://copr.fedorainfracloud.org/coprs/arturh/intel-opencl-experimental - no quality expectation

Quality expectations are based on corresponding Ubuntu builds.

In case of alpine support, I've tried building Neo under this distribution, and compilation failed in gmm. I've contacted gmm team and waiting for feedback.

adydychk, 2018-07-18:
@JacekDanecki thank you for the answer. That is much more interesting but CI will be much better :)

JacekDanecki, 2018-07-25:
For each Neo commit on github there is copr build triggered in [intel-opencl-staging](https://copr.fedorainfracloud.org/coprs/arturh/intel-opencl-staging/) repository for Centos 7, Fedora 27/28/rawhide.

JacekDanecki, 2018-07-30:
@kodonnell Fix for gmmlib compilation in alpine was added: intel/gmmlib@af2add01200f1d54165732807dfff6415d1e1249

kodonnell, 2018-07-30:
Thanks for your work on this @JacekDanecki

JacekDanecki, 2018-09-19:
I think we can close this issue, as rpm packages for Centos are available on copr. In weekly releases we can add link to specific builds on copr.
In case of alpine support there are new problems with IGC and Neo compilation. If you are still interested please create new issue for alpine support.

closed 2018-09-21 by JacekDanecki

drahnr, 2019-04-30:
Can you add a link to the copr repos in the README.md?

adydychk, 2019-04-30:
@drahnr Me?

JacekDanecki, 2019-04-30:
@drahnr When we update copr setups to use llvm 7/8, we'll add information about copr [here](https://github.com/intel/compute-runtime/blob/master/documentation/Neo_in_distributions.md). Currently on copr there are IGC components compiled with llvm 4.