intel/compute-runtime #48: Docker builds?

kodonnell, 2018-05-26:
Not sure if there's enough demand, but docker builds on some common OSes would be great:

- users can use the image directly
- users can inspect the dockerfile to see how it's compiled etc.
- this could integrate with #46 in that each docker container could build the required package for each OS etc.

JacekDanecki, 2018-05-28:
Your proposal is already on our todo list. In the meantime users can look at Dockerfile's in script/docker directory, to see how to compile Neo using IGC from binary releases under different distributions. These Dockerfile's can be easily extended by adding additional packages.
New configs are tested in this fork: https://github.com/JacekDanecki/compute-runtime.

kodonnell, 2018-05-28:
Thanks @JacekDanecki - since it's already on your todo list, I'll leave it to you whether you want to leave this open as a reminder etc.

adamradocz, 2018-07-17:
@JacekDanecki, I created a Docker container which makes the build on Ubuntu 18.04. I can make a PR if you like it. You can check out [here](https://github.com/BOINC/boinc-client-docker/blob/master/Dockerfile.intel.build).

JacekDanecki, 2018-07-24:
@adamradocz Thanks for sharing your repository. If you build using tip repositories you can find some problems with incompatibility between specific versions of components. We've observed such problems already. If you want to use ccache I'd recommend mount your ~/.ccache directory into container to speed up next builds. You can remove packages: autoconf, xutils-dev, libtool, libpciaccess-dev if you don't plan to compile libdrm, libva to enable va-api media sharing. Building all components using current external CI systems is not possible because of time/disk space/memory limits, so we've started testing [copr](https://copr.fedorainfracloud.org/coprs/arturh) and [ppa](https://launchpad.net/~jdanecki/+archive/ubuntu/intel-opencl) build systems to create binary packages. Once we have IGC binaries for different OS's we can easily reuse them to compile Neo under docker containers using current CI systems, and then push docker images to docker hub.

JacekDanecki, 2018-07-26:
@adamradocz I'm working on ppa packages for Ubuntu 18.04 right now, so you can expect initial versions today or tomorrow if I don't find some issues specific to bionic release.

JacekDanecki, 2018-07-26:
@adamradocz Initial version of packages for Ubuntu 18.04 based on release 18.27.11048 are available in [ppa](https://launchpad.net/~jdanecki/+archive/ubuntu/intel-opencl/+packages?field.name_filter=&field.status_filter=published&field.series_filter=bionic). Newer releases will be added soon.

JacekDanecki, 2018-08-01:
@kodonnell First docker image for Ubuntu 16.04 based on [intel-opencl](https://launchpad.net/~intel-opencl/+archive/ubuntu/intel-opencl) ppa was built on [Semaphore](https://semaphoreci.com/jacekdanecki/neo-hub) using [neo-hub](https://github.com/JacekDanecki/neo-hub) repository and pushed to [intelopencl](https://hub.docker.com/r/intelopencl/intel-opencl) repository.

kodonnell, 2018-08-01:
Excellent, thanks @JacekDanecki. I'd be interested in a minimal alpine build, but that can be a future goal.

closed 2018-08-01 by kodonnell