intel/compute-runtime #50: segfault when running clinfo

enihcam, 2018-05-27:
```
systemd-coredump[2762]: Core file was truncated to 2147483648 bytes.
systemd-coredump[2762]: The core will not be stored: size 2147483648 is greater than 804257792 (the configured maximum)
systemd-coredump[2762]: Process 2758 (clinfo) of user 1000 dumped core.                                                  
                                                  Stack trace of thread 2758:
                                                  #0  0x00007f5bce0c97f1 n/a (n/a)

```

JacekDanecki, 2018-05-28:
Could you provide more details about your testing platform? What OS are you using, what hardware (lspci output), what kernel? Can you provide backtrace from clinfo executed under gdb, or log from clinfo executed using strace? Output from ldd /opt/intel/opencl/* can also help to check whether all required libraries are installed.

enihcam, 2018-05-28:
```
$ lspci 
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 02)
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 620 (rev 02)
00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 02)
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:15.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #0 (rev 21)
00:15.1 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #1 (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:17.0 SATA controller: Intel Corporation Sunrise Point-LP SATA Controller [AHCI mode] (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #5 (rev f1)
00:1c.5 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #6 (rev f1)
00:1f.0 ISA bridge: Intel Corporation Sunrise Point-LP LPC Controller (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
01:00.0 Network controller: Intel Corporation Wireless 3165 (rev 79)
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
```

```
Linux version 4.16.12-2-ck (nanw@archmate) (gcc version 8.1.0 (GCC)) #2 SMP PREEMPT Sun May 27 09:26:31 CST 2018
Command line: initrd=\intel-ucode.img initrd=\initramfs-linux-ck.img root=LABEL=ROOT elevator=bfq init=/lib/systemd/systemd quiet
```

```
$ sudo ldd /opt/intel/opencl/*
/opt/intel/opencl/libiga64.so:
	linux-vdso.so.1 (0x00007ffeb58ae000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007f0ba210e000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007f0ba1ef6000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007f0ba1b3a000)
	/usr/lib64/ld-linux-x86-64.so.2 (0x00007f0ba2821000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007f0ba17a5000)
/opt/intel/opencl/libigdccl.so:
	linux-vdso.so.1 (0x00007ffc0db56000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007ff6eee14000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007ff6eec10000)
	librt.so.1 => /usr/lib/librt.so.1 (0x00007ff6eea08000)
	libz.so.1 => /usr/lib/libz.so.1 (0x00007ff6ee7f1000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007ff6ee468000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007ff6ee0d3000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007ff6edebb000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007ff6edaff000)
	/usr/lib64/ld-linux-x86-64.so.2 (0x00007ff6f12ff000)
/opt/intel/opencl/libigdfcl.so:
	linux-vdso.so.1 (0x00007fffb2bf9000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007f3e1a730000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007f3e1a52c000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007f3e1a1a3000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007f3e19e0e000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007f3e19bf6000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007f3e1983a000)
	/usr/lib64/ld-linux-x86-64.so.2 (0x00007f3e1ac4e000)
/opt/intel/opencl/libigdrcl.so:
	linux-vdso.so.1 (0x00007ffc270f2000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007fc9aa4fd000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007fc9aa2df000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007fc9a9f56000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007fc9a9bc1000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007fc9a99a9000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007fc9a95ed000)
	/usr/lib64/ld-linux-x86-64.so.2 (0x00007fc9aab28000)
/opt/intel/opencl/libopencl_clang.so:
	linux-vdso.so.1 (0x00007ffec115e000)
	librt.so.1 => /usr/lib/librt.so.1 (0x00007f126b9ce000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007f126b7ca000)
	libtinfo.so.5 => not found
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007f126b5ac000)
	libz.so.1 => /usr/lib/libz.so.1 (0x00007f126b395000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007f126b000000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007f126ac77000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007f126aa5f000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007f126a6a3000)
	/usr/lib64/ld-linux-x86-64.so.2 (0x00007f126f03d000)
```

It looks like Archlinux does not have `libtinfo.so.5`. Where can I get that?

JacekDanecki, 2018-05-28:
If you want to use Neo binary release on Arch you need to compile ncurses5-compat-libs from https://aur.archlinux.org/cgit/aur.git/snapshot/ncurses5-compat-libs.tar.gz. See script: https://github.com/intel/compute-runtime/blob/master/scripts/build-arch-dep.sh. This script is to test Neo compilation under Arch on Travis using IGC binaries from weekly binary releases.

closed 2018-05-28 by enihcam

JacekDanecki, 2018-11-13:
libopencl_clang.so library can be built without libtinfo. During configure step for cclang there are checks for example for 'setupterm' symbol from libtinfo or curses/ncurses libraries. If you don't have libtinfo-dev installed on the system this check will fail and libopencl_clang.so will not link to libtinfo. 
```
$ ldd /usr/lib/x86_64-linux-gnu/libopencl_clang.so 
        linux-vdso.so.1 =>  (0x00007ffc6b172000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fcb4da2f000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fcb4d82b000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fcb4d60e000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fcb4d305000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fcb4cf83000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fcb4cd6d000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fcb4c9a3000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fcb51159000)
$ dpkg -S /usr/lib/x86_64-linux-gnu/libopencl_clang.so
intel-opencl-clang: /usr/lib/x86_64-linux-gnu/libopencl_clang.so
$ apt-cache policy  intel-opencl-clang
intel-opencl-clang:
  Installed: 4.0.16-1~ppa1~xenial1
  Candidate: 4.0.16-1~ppa1~xenial1
  Version table:
 *** 4.0.16-1~ppa1~xenial1 500
        500 http://ppa.launchpad.net/ocl-dev/intel-opencl/ubuntu xenial/main amd64 Packages
        100 /var/lib/dpkg/status
```
If you want to remove ncurses5-compat-libs from compute-runtime-bin Aur package I'd recommend to use deb packages from launchpad: https://launchpad.net/~intel-opencl/+archive/ubuntu/intel-opencl. There are weekly updates here once there is new github release. If you want to use more recent versions, there is another account for automatic builds: https://launchpad.net/~ocl-dev/+archive/ubuntu/intel-opencl.