intel/compute-runtime #51: Slice/Subslice powergating

djdeath, 2018-05-29:
Hi,

We're currently working on a patch series in the i915 kernel driver to enable per context slice & subslice powergating : https://patchwork.freedesktop.org/series/42285/

Currently we have that feature disabled by default in this i915 series and it can be turned on by setting a sysfs entry. This means that the per context settings are ignored by default (and we keep the powergating configuration to nothing powergated by default, like this is the case on the current kernels).
As far as I know this feature is only useful for performance gains on some media workloads.
Mesa is not planning to make use of this, there it only make sense to run workload on the maximum amount of slices/subslices (as far as we're aware).

We're wondering what's the take from the compute developers?
Would this feature be of any use?

There is a side effect to enabling this feature : switching from one powergating configuration to another takes time (we measure a 50~60us delay on Skylake GT4, the delay grows with the size of the GT) and this delay will occur everytime a context switch happens between 2 contexts of different powergating configurations.

Thanks for your comments!

djdeath, 2018-06-06:
Ping?

PiotrRozenfeld, 2018-06-07:
This is potentially useful:
* In general, we can tie it with [cl_khr_throttle_hints]( https://www.khronos.org/registry/OpenCL/sdk/2.1/docs/man/xhtml/cl_khr_throttle_hints.html) and disable slices when app provides CL_QUEUE_THROTTLE_LOW to the driver
* Specifically, for inference use cases on edge devices, when the DNN topology used results in underutilized HW, we can use this feature to save on power without affecting performance

MichalMrozek, 2018-06-29:
Closing. Feel free to reopen if you have additional questions in the future.

closed 2018-06-29 by MichalMrozek