intel/compute-runtime #57: i7 8700K not found by clinfo on Kubuntu 18.04; missing steps?

ranocha, 2018-06-22:
I'm using Kubuntu 18.04 with an i7 8700K and a GeForce GTX 1070 Ti. I've installed the `nvidia-cuda-toolkit` which adds support for OpenCL on the GPU. Then, I followed the build instructions of Intel Compute Runtime and installed `intel-opencl-igdrcl` as described. However, `clinfo` (from the official package lists) is not able to find the Intel OpenCL implementation.
```
hendrik@i7-8700K:~/Downloads/intel_compute_runtime/build$ clinfo
Number of platforms                               1
  Platform Name                                   NVIDIA CUDA
  Platform Vendor                                 NVIDIA Corporation
  Platform Version                                OpenCL 1.2 CUDA 9.1.84
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts cl_nv_create_buffer
  Platform Extensions function suffix             NV

  Platform Name                                   NVIDIA CUDA
Number of devices                                 1
  Device Name                                     GeForce GTX 1070 Ti
  Device Vendor                                   NVIDIA Corporation
......

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.11
  ICD loader Profile                              OpenCL 2.1
```
However, there is an `intel.icd` file:
```
hendrik@i7-8700K:~/Downloads/intel_compute_runtime/build$ ls /etc/OpenCL/vendors/
intel.icd  nvidia.icd
hendrik@i7-8700K:~/Downloads/intel_compute_runtime/build$ cat /etc/OpenCL/vendors/intel.icd 
/opt/intel/opencl/libigdrcl.so
```
I've seen https://github.com/intel/compute-runtime/issues/20#issuecomment-366650726 and have added the Khronos ICD loader as described there, but the issue remains the same.

Could you please help me? What am I doing wrong? Thanks in advance for your time!

alalek, 2018-06-22:
Try to track loaded .so files via `LD_DEBUG=files clinfo` (attention for failed loads / strange full paths).

ranocha, 2018-06-23:
Thanks for your fast answer. Here is the output:
```
hendrik@i7-8700K:~/Downloads/intel_compute_runtime/build$ LD_DEBUG=files clinfo
      2746:
      2746:     WARNING: Unsupported flag value(s) of 0x8000000 in DT_FLAGS_1.
      2746:
      2746:     file=libOpenCL.so.1 [0];  needed by clinfo [0]
      2746:     file=libOpenCL.so.1 [0];  generating link map
      2746:       dynamic: 0x00007f741f430cd0  base: 0x00007f741f227000   size: 0x000000000020a050
      2746:         entry: 0x00007f741f22b180  phdr: 0x00007f741f227040  phnum:                  8
      2746:
      2746:
      2746:     file=libdl.so.2 [0];  needed by clinfo [0]
      2746:     file=libdl.so.2 [0];  generating link map
      2746:       dynamic: 0x00007f741f225d90  base: 0x00007f741f023000   size: 0x0000000000203110
      2746:         entry: 0x00007f741f023e50  phdr: 0x00007f741f023040  phnum:                  7
      2746:
      2746:
      2746:     file=libc.so.6 [0];  needed by clinfo [0]
      2746:     file=libc.so.6 [0];  generating link map
      2746:       dynamic: 0x00007f741f01cb80  base: 0x00007f741ec32000   size: 0x00000000003f0ae0
      2746:         entry: 0x00007f741ec53cb0  phdr: 0x00007f741ec32040  phnum:                 10
      2746:
      2746:
      2746:     calling init: /lib/x86_64-linux-gnu/libc.so.6
      2746:
      2746:
      2746:     calling init: /lib/x86_64-linux-gnu/libdl.so.2
      2746:
      2746:
      2746:     calling init: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1
      2746:
      2746:
      2746:     initialize program: clinfo
      2746:
      2746:
      2746:     transferring control: clinfo
      2746:
      2746:
      2746:     file=libnvidia-opencl.so.1 [0];  dynamically loaded by /usr/lib/x86_64-linux-gnu/libOpenCL.so.1 [0]
      2746:     file=libnvidia-opencl.so.1 [0];  generating link map
      2746:       dynamic: 0x00007f741eb2c928  base: 0x00007f741db01000   size: 0x0000000001130d98
      2746:         entry: 0x00007f741dbb30b0  phdr: 0x00007f741db01040  phnum:                  5
      2746:
      2746:
      2746:     file=libm.so.6 [0];  needed by /usr/lib/x86_64-linux-gnu/libnvidia-opencl.so.1 [0]
      2746:     file=libm.so.6 [0];  generating link map
      2746:       dynamic: 0x00007f741daffd90  base: 0x00007f741d763000   size: 0x000000000039d158
      2746:         entry: 0x00007f741d76ea80  phdr: 0x00007f741d763040  phnum:                  7
      2746:
      2746:
      2746:     file=libpthread.so.0 [0];  needed by /usr/lib/x86_64-linux-gnu/libnvidia-opencl.so.1 [0]
      2746:     file=libpthread.so.0 [0];  generating link map
      2746:       dynamic: 0x00007f741d75dd68  base: 0x00007f741d544000   size: 0x000000000021e480
      2746:         entry: 0x00007f741d54a230  phdr: 0x00007f741d544040  phnum:                  9
      2746:
      2746:
      2746:     file=librt.so.1 [0];  needed by /usr/lib/x86_64-linux-gnu/libnvidia-opencl.so.1 [0]
      2746:     file=librt.so.1 [0];  generating link map
      2746:       dynamic: 0x00007f741d542d80  base: 0x00007f741d33c000   size: 0x0000000000207be0
      2746:         entry: 0x00007f741d33e200  phdr: 0x00007f741d33c040  phnum:                  7
      2746:
      2746:
      2746:     file=libnvidia-fatbinaryloader.so.390.48 [0];  needed by /usr/lib/x86_64-linux-gnu/libnvidia-opencl.so.1 [0]
      2746:     file=libnvidia-fatbinaryloader.so.390.48 [0];  generating link map
      2746:       dynamic: 0x00007f741d3358c0  base: 0x00007f741d0f0000   size: 0x000000000024be08
      2746:         entry: 0x00007f741d0f5930  phdr: 0x00007f741d0f0040  phnum:                  5
      2746:
      2746:
      2746:     calling init: /lib/x86_64-linux-gnu/libpthread.so.0
[... 258 lines cut ...]
      2746:
      2746:
      2746:     calling fini: /usr/lib/x86_64-linux-gnu/libcuda.so.1 [0]
      2746:
      2746:
      2746:     calling fini: /usr/lib/x86_64-linux-gnu/libnvidia-fatbinaryloader.so.390.48 [0]
      2746:
      2746:
      2746:     calling fini: /usr/lib/x86_64-linux-gnu/libnvidia-compiler.so.390.48 [0]
      2746:
      2746:
      2746:     calling fini: /lib/x86_64-linux-gnu/librt.so.1 [0]
      2746:
      2746:
      2746:     calling fini: /lib/x86_64-linux-gnu/libpthread.so.0 [0]
      2746:
      2746:
      2746:     calling fini: /lib/x86_64-linux-gnu/libm.so.6 [0]
      2746:
      2746:
      2746:     calling fini: /lib/x86_64-linux-gnu/libdl.so.2 [0]
      2746:
hendrik@i7-8700K:~/Downloads/intel_compute_runtime/build$ 
```
There are some errors, but these do not seem to be related to `/opt/intel/opencl/libigdrcl.so`, are they? The errors remain are the same even if I remove `intel-opencl-igdrcl `.

ArturHarasimiuk, 2018-06-23:
Hi @ranocha,
could you provide some additional data:
```
ldd -r /opt/intel/opencl/*.so
```
```
lspci
```
```
dmesg | grep i915
dmesg | grep drm
```
have you tried running clinfo as root user?

ranocha, 2018-06-23:
Thank you for your help. Running `sudo clinfo` gives the same results as `clinfo`. Here are the additional data:
```
hendrik@i7-8700K:~/Downloads/intel_compute_runtime/build$ ldd -r /opt/intel/opencl/*.so
/opt/intel/opencl/libiga64.so:
        linux-vdso.so.1 (0x00007ffdc23bb000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f248892a000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f2488712000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f2488321000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f2489097000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f2487f83000)
/opt/intel/opencl/libigc.so:
        linux-vdso.so.1 (0x00007ffc8b5a2000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fd72d69d000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fd72d499000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fd72d27a000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fd72d05d000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fd72cccf000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd72c931000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fd72c719000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd72c328000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fd72fc9d000)
/opt/intel/opencl/libigdfcl.so:
        linux-vdso.so.1 (0x00007ffc8b1ea000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f6cb7a52000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f6cb7833000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f6cb74a5000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f6cb7107000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f6cb6eef000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f6cb6afe000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f6cb7f55000)
/opt/intel/opencl/libigdrcl.so:
        linux-vdso.so.1 (0x00007ffdc6b4d000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fd33c95d000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fd33c73e000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fd33c3b0000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd33c012000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fd33bdfa000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd33ba09000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fd33cf95000)
/opt/intel/opencl/libopencl_clang.so:
        linux-vdso.so.1 (0x00007fff7ebec000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f97bbba4000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f97bb9a0000)
        libtinfo.so.5 => /lib/x86_64-linux-gnu/libtinfo.so.5 (0x00007f97bb776000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f97bb557000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f97bb33a000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f97baf9c000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f97bac0e000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f97ba9f6000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f97ba605000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f97bf21f000)
hendrik@i7-8700K:~/Downloads/intel_compute_runtime/build$ lspci
00:00.0 Host bridge: Intel Corporation Device 3ec2 (rev 07)
00:01.0 PCI bridge: Intel Corporation Skylake PCIe Controller (x16) (rev 07)
00:08.0 System peripheral: Intel Corporation Skylake Gaussian Mixture Model
00:14.0 USB controller: Intel Corporation 200 Series PCH USB 3.0 xHCI Controller
00:14.2 Signal processing controller: Intel Corporation 200 Series PCH Thermal Subsystem
00:16.0 Communication controller: Intel Corporation 200 Series PCH CSME HECI #1
00:17.0 SATA controller: Intel Corporation 200 Series PCH SATA controller [AHCI mode]
00:1c.0 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #1 (rev f0)
00:1c.4 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #5 (rev f0)
00:1c.6 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #7 (rev f0)
00:1d.0 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #9 (rev f0)
00:1f.0 ISA bridge: Intel Corporation Device a2c9
00:1f.2 Memory controller: Intel Corporation 200 Series PCH PMC
00:1f.3 Audio device: Intel Corporation 200 Series PCH HD Audio
00:1f.4 SMBus: Intel Corporation 200 Series PCH SMBus Controller
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (2) I219-V
01:00.0 VGA compatible controller: NVIDIA Corporation GP104 (rev a1)
01:00.1 Audio device: NVIDIA Corporation GP104 High Definition Audio Controller (rev a1)
03:00.0 USB controller: ASMedia Technology Inc. Device 2142
04:00.0 Network controller: Qualcomm Atheros AR93xx Wireless Network Adapter (rev 01)
05:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd Device a808
hendrik@i7-8700K:~/Downloads/intel_compute_runtime/build$ dmesg | grep i915
hendrik@i7-8700K:~/Downloads/intel_compute_runtime/build$ dmesg | grep drm
[    1.273350] [drm] [nvidia-drm] [GPU ID 0x00000100] Loading driver
[    1.273351] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:01:00.0 on minor 0
hendrik@i7-8700K:~/Downloads/intel_compute_runtime/build$ 
```

ArturHarasimiuk, 2018-06-23:
there is no "00:02.0 VGA compatible controller: Intel Corporation ..." device listed. Most probably device wasn't initialized by BIOS. I think you should inspect BIOS setting and try enabling both integrated and discrete graphics.

ranocha, 2018-06-23:
Thank you very much, @ArturHarasimiuk. I've been silly, I'm very sorry. Now, everything works well. Thank you all very much for your great work and patient help!
Maybe a small note in the setup section might prevent other people from bothering you with such stupid problems...

closed 2018-06-23 by ranocha