intel/compute-runtime #58: In-kernel debugging fails with vtune

axeldavy, 2018-06-24:
Not sure if this is a bug of vtune, compute-runtime or something else so I post here.

While I finally managed to get vtune to work with compute-runtime (documentation needs to be more explicit ! Only one slide presentation mentions the need for CONFIG_DRM_I915_LOW_LEVEL_TRACEPOINTS=y, and I had to put libmd.so from intel/metrics-discovery in vtune dir which is documented nowhere !), gpu-hotspots works well but not gpu-profile.

Indeed gpu-profiling spawns "Error: Internal collection error has occurred. Please contact the technical support." which results in no in-kernel profiling.

PiotrRozenfeld, 2018-06-29:
We confirmed we see the same issue on our end. We will follow-up with detailed response as soon as we have clarity what needs to be done to resolve this problem

PiotrRozenfeld, 2018-07-13:
To resolve this issue, Intel needs to open source gtpin headers (used by the code [here](https://github.com/intel/compute-runtime/tree/master/runtime/gtpin)). We have started discussion with the team at Intel responsible, will update you as soon as we have conclusions.

PiotrRozenfeld, 2018-09-19:
Could you please check again with our latest release? This should be fixed now.

axeldavy, 2018-09-21:
I tried latest release of compute-runtime, latest release of vtune, and git of https://github.com/intel/metrics-discovery
I still get the same issue.

Is there any detailed setup installation to be sure everything is setup correctly ?

axeldavy, 2018-09-26:
Would it be possible a debug build is required ?

I used the binaries available on this github.

pioto1225, 2018-09-30:
The same problem here. Gpu hotspots work, while in-kernel profiling results in GTPin error.

MaciejPlewka, 2018-10-08:
Hey, could you provide steps to reproduce this problem?

axeldavy, 2018-10-08:
I compiled my linux kernel with (CONFIG_DRM_I915_LOW_LEVEL_TRACEPOINTS=y) (Tested both 4.14 and 4.18), as requested on the online documentation.
I installed the compute-runtime package available here (I didn't recompile myself).
I use a Skylake gpu. The issue happens with both VTune "2018" and "2019".

Taken any OpenCL program, even the simplest one with one kernel. In VTune use In-kernel debugging on one of the kernels.

pioto1225, 2018-10-17:
I just tried from installing the latest Intel vTune ((build 570779, version 2019) from scratch and it looks like vTune in-kernel opencl profiling works fine with Linux.
I am using:
* Debian Buster with the latest packaged kernel 
* recompiled stock kernel 4.18.10 (with: CONFIG_DRM_I915_LOW_LEVEL_TRACEPOINTS=y)
* latest metrics-discovery package (https://github.com/intel/metrics-discovery) (built binary libmd.so copied to vTune/bin64 folder.
Steps:
as root run:
./prepare-debugfs.sh 
./amplexe-vars.sh
LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./amplxe-gui 

OpenCL kernel can be viewed with profile information.

MaciejPlewka, 2018-10-25:
I was trying to reproduce this issue but i wasn't able. I compiled my own linux kernel(v4.14) with "CONFIG_DRM_I915_LOW_LEVEL_TRACEPOINTS=y" as you wrote. I'm working on Ubuntu 16.04 and vTune 2019 (build 570779). I was testing on simple app named BitonicSort which you can download from intel site. Make sure that you are specifying  device type to GPU in other way you will receive GTPin Error

axeldavy, 2018-10-25:
I tried again.
I use 4.17.9 (with CONFIG_DRM_I915_LOW_LEVEL_TRACEPOINTS=y)
sudo ./prepare-debugfs.sh
sudo ./amplexe-vars.sh
sudo chmod a+rx /sys/kernel/debug
sudo /opt/intel/vtune_amplifier_2019/sepdk/src/insmod-sep
./amplxe-gui (I don't need LD_PRELOAD, I just renamed the old packaged libstdc++.so to make it use the system version)
Parametered to launch the bitonic executable and in kernel gpu profiling
BitonicSort as kernel, 1 as instance step.

Still same error and failure.
vtune build is 570779
dmesg shows sepkd loading succeeded:

The only difference in my setup and yours seem to be I'm using Archlinux

```
[  668.220932] pax: loading out-of-tree module taints kernel.
[  668.221647] PAX: PMU arbitration service v1.0.2 has been started.
[  669.399586] socperf3_0: SocPerf Driver loading...
[  669.399587] socperf3_0: SocPerf Driver about to register chrdev...
[  669.399590] socperf3_0: SocPerf Driver: result of alloc_chrdev_region is 0
[  669.399591] socperf3_0: SocPerf Driver: major number is 239
[  669.399592] socperf3_0: SocPerf Driver: detected 4 CPUs in lwpmudrv_Load
[  669.399593] socperf3_0: SocPerf Driver: creating device socperf3!c...
[  669.399684] socperf3_0: SocPerf Driver v3.0.0 has been loaded.
[  670.461216] **********************************************************
[  670.461218] **   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **
[  670.461219] **                                                      **
[  670.461221] ** trace_printk() being used. Allocating extra memory.  **
[  670.461222] **                                                      **
[  670.461223] ** This means that this is a DEBUG kernel and it is     **
[  670.461224] ** unsafe for production use.                           **
[  670.461226] **                                                      **
[  670.461227] ** If you see this message and you are not debugging    **
[  670.461228] ** the kernel, report this immediately to your vendor!  **
[  670.461229] **                                                      **
[  670.461230] **   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **
[  670.461232] **********************************************************
[  670.462171] sep5_1: Driver loading...
[  670.462558] sep5_1: [load] [UTILITY_Driver_Log_Init@1004]: Initialized driver log using contiguous physical memory.
[  670.462568] sep5_1: [load] [lwpmu_Load@6369]: Major number is 238
[  670.462583] sep5_1: [load] [lwpmu_Load@6376]: Detected 4 total CPUs and 4 active CPUs.
[  670.464817] sep5_1: [load] [lwpmu_Load@6610]: PMU collection driver v5.1.0 Beta has been loaded.
[  670.464822] sep5_1: [load] [lwpmu_Load@6613]: Chipset support is enabled.
[  670.464826] sep5_1: [load] [lwpmu_Load@6617]: Graphics support is enabled.
[  670.464830] sep5_1: [load] [lwpmu_Load@6620]: NMI will be used for handling PMU interrupts.
[  671.576405] vtss++ kernel module ("v1.8.155-570779 Intel(R) VTune(TM) Amplifier") registered

```

pioto1225, 2018-10-25:
I tried bitonic sort and I got the opencl profile.
My dmesg output (below) is almost identical to yours except for major number; probably it does not matter. Normally I would get GTpin error which I believe has to do with locally building libmd against distro's libstdc++. Otherwise (with LD_PRELOAD/removing Vtune's libstdc++) it is OK. Are you getting GTPin error or a different one? In your first post you mentioned internal collection error, which could be a different thing. Can this be a Skylake problem? I am using the Kabylake.

My dmesg's output:
[    5.078701] pax: loading out-of-tree module taints kernel.
[    5.079214] PAX: PMU arbitration service v1.0.2 has been started.
...
[    6.124552] socperf3_0: SocPerf Driver loading...
[    6.124553] socperf3_0: SocPerf Driver about to register chrdev...
[    6.124554] socperf3_0: SocPerf Driver: result of alloc_chrdev_region is 0
[    6.124555] socperf3_0: SocPerf Driver: major number is 241
[    6.124555] socperf3_0: SocPerf Driver: detected 4 CPUs in lwpmudrv_Load
[    6.124556] socperf3_0: SocPerf Driver: creating device socperf3!c...
[    6.124594] socperf3_0: SocPerf Driver v3.0.0 has been loaded.
[    6.221270] fuse init (API version 7.27)
....
[    7.160386] sep5_1: Driver loading...
[    7.160641] sep5_1: [load] [UTILITY_Driver_Log_Init@1004]: Initialized driver log using contiguous physical memory.
[    7.160648] sep5_1: [load] [lwpmu_Load@6369]: Major number is 240
[    7.160655] sep5_1: [load] [lwpmu_Load@6376]: Detected 4 total CPUs and 4 active CPUs.
[    7.164596] sep5_1: [load] [lwpmu_Load@6610]: PMU collection driver v5.1.0 Beta has been loaded.
[    7.164600] sep5_1: [load] [lwpmu_Load@6613]: Chipset support is enabled.
[    7.164602] sep5_1: [load] [lwpmu_Load@6617]: Graphics support is enabled.
[    7.164605] sep5_1: [load] [lwpmu_Load@6620]: NMI will be used for handling PMU interrupts.
[    8.121234] rfkill: input handler disabled
[    8.218640] vtss++ kernel module ("v1.8.155-570779 Intel(R) VTune(TM) Amplifier") registered

wmouse, 2018-10-26:
Hi @axeldavy, could please collect additional logs for VTune? You need to set the environment variables:
```
export AMPLXE_LOG_LEVEL=TRACE
export TPSS_DEBUG=1
```
and run the analysis. After that, the result directory (<path_to_results>/r***gp) will contain several log files. If you can please share this result directory.
Also you can use [the official VTune forum](https://software.intel.com/en-us/forums/intel-vtune-amplifier) to report the issues with VTune.

axeldavy, 2018-11-06:
The two env vars didn't seem to increase the length of the logs.

Here are the logs nonetheless, if it helps.

[log.tar.gz](https://github.com/intel/compute-runtime/files/2553036/log.tar.gz)

wmouse, 2018-11-06:
Hi @axeldavy , these environments variable enable logs that are stored in result directory. Could you please attach the whole result directory (r004gp) or log files from r004gp/data.0?

axeldavy, 2018-11-06:
Here is the whole directory

[r004gp.tar.gz](https://github.com/intel/compute-runtime/files/2553111/r004gp.tar.gz)

axeldavy, 2018-11-06:
Since in the log it complains about
'/opt/intel/vtune_amplifier_2019.0.2.570779/lib64/libstdc++.so.6' not being found (which if you read my previous messages I removed to force the system one),
I put a symlink to the system .so.

The error "Error: Internal collection error has occurred. Please contact the technical support." disappeared.

However now it gives:
[Instrumentation Engine]: GTPin couldn't find all required libraries... Existing without doing anything

I guess the remaining issue is these parts of the log
```
9674 : TRACE : Cannot get handle for library 'libigc.so', error message: 'libigc.so: invalid mode for dlopen(): Invalid argument' 
9674 : TRACE : Cannot get handle for library 'libigdccl.so', error message: 'libigdccl.so: invalid mode for dlopen(): Invalid argument' 
9674 : TRACE : OpenCL backend library is not found 
```
libigc.so is in /usr/lib/ and I don't know why it can't find it (in the log I pasted here, the very same .so is loaded after the error). libigdccl.so doesn't seem to be on my installation, but seem to have been present in older driver packages.

axeldavy, 2018-11-06:
I reinstalled the intel opencl package 18.21.10858, which I had locally and that contained libigdccl.so, and in-kernel profiling works (with the libstdc++ symlink).
The log still contains "Cannot get handle for library 'libigdccl.so'", but apparently it doesn't prevent it from working.
Here is the log, in case that helps fixing the issue:
[r010gp.tar.gz](https://github.com/intel/compute-runtime/files/2553226/r010gp.tar.gz)

wmouse, 2018-11-06:
Thank you for the log! I'll investigate it.

MaciejPlewka, 2018-11-14:
The release version 18.21.10858 is quite old and doesn't have support for GTPin. Could You try with latest opencl package?

axeldavy, 2018-11-26:
It's only with 18.21.10858 I managed to make In-kernel debugging work (but with issues, some kernels do miss debug info, either completly or partially), as with the latest package (from 20 days ago) I got '[Instrumentation Engine]: GTPin couldn't find all required libraries... Exiting without doing anything'.
Does this error message rings a bell for you ?

MaciejPlewka, 2018-11-27:
I used to have this issue when i wasn't specifying device type as argument to BitonicSort. After add CL_DEVICE_TYPE_GPU everything works for me.

wmouse, 2018-11-28:
Hi @axeldavy, @MaciejPlewka is right because '[Instrumentation Engine]: GTPin couldn't find all required libraries... Exiting without doing anything' message usually means that a target application uses non-GPU OpenCL device to run kernels. VTune provides different analyses to profile an OpenCL application. The choice depends on OpenCL device that is used to execute the kernels.
If you want to profile the kernel that is executed on Intel OpenCL CPU device you should use Hotspot analysis. And if you want to profile the kernel that is executed on Intel OpenCL GPU device you should use 'GPU Compute/Media Hotspots' or 'GPU In-kernel Profiling' analyses. Therefore the target application have to directly select the OpenCL device that will be used to execute the kernels.

axeldavy, 2018-12-06:
I hadn't any other OpenCL device installed than Neo.
I managed to make it work now with the same driver version that used to say "GTPin couldn't find all required libraries". Since last time, I installed the CPU openCL driver and recompiled last linux lts with the CONFIG_DRM_I915_LOW_LEVEL_TRACEPOINTS=y. Not sure what helped.

Since it seems I was the last person on this thread having issues, I guess the bug report can be closed ?
vtune has some issues, but that is not the right place I guess (where is the right place ?)
The issues I have in mind are:
. It is not documented libmd.so should be put in the binary dir
. Mismatch with system's libstdc++, and if file removed won't work well (won't search in /usr/lib). Has to use a symlink to the system one.
. The interface doesn't like some kernel names (especially when the name has underscores, but some names with underscores are ok) and it won't launch the in-kernel debugging for some names. Renaming the kernel works...

MaciejPlewka, 2019-01-30:
Hi @axeldavy,
If you still have problem with VTune please write about this here [Intel VTune Amplifier forum](https://software.intel.com/en-us/forums/intel-vtune-amplifier)

closed 2019-02-25 by JacekDanecki