intel/compute-runtime #61: Build failure with version 18.27.11048: igfxfmid.h: No such file or directory

dbermond, 2018-07-15:
I'm getting the following compile error:
```
Scanning dependencies of target cloc
[  0%] Building CXX object offline_compiler/CMakeFiles/cloc.dir/offline_compiler.cpp.o
/storage/linux/abs/intel-compute-runtime-git/src/neo/offline_compiler/offline_compiler.cpp:31:10: fatal error: igfxfmid.h: No such file or directory
 #include "igfxfmid.h"
          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [offline_compiler/CMakeFiles/cloc.dir/build.make:63: offline_compiler/CMakeFiles/cloc.dir/offline_compiler.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:11826: offline_compiler/CMakeFiles/cloc.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

When using `make` with multiple jobs (`-j` option), it shows multiple occurrences of this not found file `igfxfmid.h`, and also an error complaining about the not found file `gtsysinfo.h`.

- Current git master gives me the same error.

- Version 18.26.10987 builds fine for me.

- A git bisect shows that this is being caused by commit 561cdb7df595287389cea774cd6fdf595a9b3e40.

Switching occurrences of `#include "igfxfmid.h"` to `#include "common/igfxfmid.h"` across the source code fixes the corresponding error for me (and also occurrences of  `#include "gtsysinfo.h"` to  `#include "common/gtsysinfo.h"`. But there are a lot of files to change and I did not test all of them to check if it really fixes all errors.

**OS:** Arch Linux x86_64

JacekDanecki, 2018-07-17:
I don't see such issue on any build system we use. I was able to compile the whole stack under Arch using AUR package from https://aur.archlinux.org/packages/compute-runtime, and builds on Travis and Semaphore under Arch also passed. Can you provide more details how have you prepared your workspace? What commit ID of gmmlib do you use? Can you also provide output from cmake command?

Brainiarc7, 2018-07-17:
Reverting the patch still results in a build failure.

```
[ 25%] Generating ../../../bin/built_ins/x64/gen8/copy_buffer_rect_Gen8core.bc, ../../../bin/built_ins/x64/gen8/copy_buffer_rect_Gen8core.bin, ../../../bin/built_ins/x64/gen8/copy_buffer_rect_Gen8core.cpp, ../../../bin/built_ins/x64/gen8/copy_buffer_rect_Gen8core.gen
cd /home/brainiarc7/intel-compute-runtime/workspace/neo/runtime/built_ins/kernels && LD_LIBRARY_PATH=/home/brainiarc7/intel-compute-runtime/workspace/build_icr/bin /home/brainiarc7/intel-compute-runtime/workspace/build_icr/bin/cloc -q -file copy_buffer_rect.igdrcl_built_in -device bdw -64 -out_dir /home/brainiarc7/intel-compute-runtime/workspace/build_icr/bin/built_ins/x64/gen8 -cpp_file -options -cl-kernel-arg-info\ 
igdrcl_lib_release/built_ins/kernels/CMakeFiles/builtins_Gen8core.dir/build.make:71: recipe for target 'bin/built_ins/x64/gen8/copy_buffer_rect_Gen8core.bc' failed
make[2]: *** [bin/built_ins/x64/gen8/copy_buffer_rect_Gen8core.bc] Error 250
make[2]: Leaving directory '/home/brainiarc7/intel-compute-runtime/workspace/build_icr'
CMakeFiles/Makefile2:1194: recipe for target 'igdrcl_lib_release/built_ins/kernels/CMakeFiles/builtins_Gen8core.dir/all' failed
make[1]: *** [igdrcl_lib_release/built_ins/kernels/CMakeFiles/builtins_Gen8core.dir/all] Error 2
make[1]: Leaving directory '/home/brainiarc7/intel-compute-runtime/workspace/build_icr'
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

As shown above.

JacekDanecki, 2018-07-18:
@Brainiarc7 Error with cloc is different problem. It looks like cloc uses incompatible IGC libraries, or there are missing dependencies. Can you run cloc after setting export LD_DEBUG=libs, and check output from linker?
Yesterday I've built all Neo components using below commits ID's:
* intel/compute-runtime@ 8da3da402fc8e5e2cd2561f209ea9968345b48b1
* intel/gmmlib@1a4cf50767b4ccacc1f6027b1ef2118bc23a30e8
* intel/intel-graphics-compiler@1451db9e994c6669fa3b0be06695498f19be9f9b
* intel/llvm-patches@5c67217eafd982e4e83724ec8b19893efd11b7d2
* intel/opencl-clang@82481202297e69f619766df57ac3cce70a35d82f

In release 18.27.11048 localization where libraries are installed has changed from /opt/intel/opencl to /usr/local/lib (Ubuntu) or /usr/local/lib64 (Fedora).
Do you have any older versions of igc, gmmlib, Neo installed in the system? Can you share output from cmake?

Brainiarc7, 2018-07-18:
Yes, 

I have older versions of igc, gmmlib and Neo installed. I had been able to generate a working build earlier using these instructions: https://gist.github.com/Brainiarc7/1d13c7f432ba03a8e38720c83cd973d5

I'll share output from Cmake when I'm back on the testbed.

dbermond, 2018-07-20:
Sorry for the late answer. Got very busy days.

- cmake command and output: [https://bpaste.net/show/5d4481a7d3c8](https://bpaste.net/show/5d4481a7d3c8)

- workspace defined as described in BUILD_Ubuntu.md:
```
$ ls -l | awk '{ print $9 }'

clang_source
common_clang
gmmlib
igc
llvm_patches
llvm_source
neo
opencl_headers
```

- gmmlib is at git master

I'm still getting this error when using the current compute-runtime git master: `fatal error: gtsysinfo.h: No such file or directory`.

I do not have any version of igc or gmmlib installed while building compute-runtime.

Brainiarc7, 2018-07-20:
In my case:

(a). Cmake output:
```
cmake -DCMAKE_BUILD_TYPE=Release ../neo
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Release build configuration
-- i915 includes dir: /home/brainiarc7/intel-compute-runtime/workspace/neo/third_party/uapi
-- Khronos OpenCL headers dir: /home/brainiarc7/intel-compute-runtime/workspace/neo/third_party/opencl_headers
-- Third party dir: /home/brainiarc7/intel-compute-runtime/workspace/third_party
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'igc-opencl'
--   Found igc-opencl, version 0.1.0
-- IGC include dirs: /usr/include/igc;/usr/include/igc/cif;/usr/include/igc/ocl_igc_shared/executable_format;/usr/include/igc/ocl_igc_shared/device_enqueue
-- Checking for module 'igdgmm'
--   Found igdgmm, version 0.1.0
-- GmmLib include dirs: /usr/include/igdgmm;/usr/include/igdgmm/GmmLib;/usr/include/igdgmm/GmmLib/inc;/usr/include/igdgmm/inc;/usr/include/igdgmm/inc/common;/usr/include/igdgmm/util
-- Computed version major is: 18
-- Computed version minor is: 29
-- All supported platforms:  CNL SKL KBL GLK CFL BXT BDW
-- All tested platforms:  CNL SKL KBL GLK CFL BXT BDW
-- Default supported platform: SKL
-- Default tested platform: SKL
-- Default tested family name: SKLFamily
-- Found ccache: /usr/bin/ccache
-- Checking for module 'libva>=1.0.0'
--   Found libva, version 1.2.0
-- Looking for vaGetLibFunc in va
-- Looking for vaGetLibFunc in va - found
-- Using libva 
-- GTest repeat count set to 1
-- GTest shuffle set to --gtest_shuffle;--gtest_random_seed=0
-- GTest exception options set to --gtest_catch_exceptions=1
-- GTest filter for mt tests:-AsyncGPUoperations.MapBufferAfterWriteBuffer
-- Configuring done
-- Generating done
-- Build files have been written to: /home/brainiarc7/intel-compute-runtime/workspace/build_icr

```

(b). Build error:

Same as above, as commented by @dbermond

jdanecki, 2018-07-21:
@dbermond There is gmmlib installed on your system. See line 375 in log you've shared. Can you remove directory /usr/include/igdgmm and installed igdmm.pc file and re-run cmake? I'll look at such case when I'm back in the office.

dbermond, 2018-07-21:
@jdanecki I see. I have [Intel Media Driver](https://github.com/intel/media-driver) git master installed on system, but not gmmlib by its own. The `make install` process from media driver installs `libigdgmm.so` and the `igdgmm` headers. I could do a successful build of compute runtime git master after uninstallling media driver.

Curiously, these gmmlib files that are being installed by media driver are from gmmlib git master (media driver requires gmmlib source to compile), and being such there should be no problem in building compute runtime with them. But media driver installs `igdgmm` headers in a directory structure that is different from upstream gmmlib by adding the `igdgmm` folder under the include folder. I think that the problem resides in this situation.
 
Is there any hope of having a way to build compute runtime when having media driver installed alongside?

JacekDanecki, 2018-07-23:
I've reproduced this issue on my side and contacted both media and gmmlib teams. Let's wait for feedback from them.

Headers from common directory are installed in different location, than provided by igdgmm.pc file 
```
# pkg-config --cflags igdgmm
-I/usr/include/igdgmm -I/usr/include/igdgmm/GmmLib -I/usr/include/igdgmm/GmmLib/inc -I/usr/include/igdgmm/inc -I/usr/include/igdgmm/inc/common -I/usr/include/igdgmm/util
# ls -l /usr/include/igdgmm/inc/common
ls: cannot access '/usr/include/igdgmm/inc/common': No such file or directory

ls -l /usr/include/igdgmm/common
total 116
-rw-r--r-- 1 root root 15424 Jul 23 13:45 gfxmacro.h
-rw-r--r-- 1 root root  2992 Jul 23 13:45 gfxplatform.h
-rw-r--r-- 1 root root  9273 Jul 23 13:45 gtsysinfo.h
-rw-r--r-- 1 root root 63225 Jul 23 13:45 igfxfmid.h
-rw-r--r-- 1 root root 19244 Jul 23 13:45 sku_wa.h
```

dbermond, 2018-07-23:
@JacekDanecki Good to know that you could reproduce it. Thank you for working on this issue.

JacekDanecki, 2018-09-04:
Using the latest media-driver and gmmlib code I was able to compile Neo on setup with media  and gmmlib installed.

dbermond, 2018-09-04:
> Using the latest media-driver and gmmlib code I was able to compile Neo on setup with media and gmmlib installed.

@JacekDanecki Are you using intel-media-driver git master or the 18.3.pre2 pre-release?

- compute-runtime git master still fails to build for me when using media-driver git master, giving the same message: `fatal error: gtsysinfo.h: No such file or directory`.

- Situation changes a bit when using media-driver 18.3.pre2 pre-release beucause this version will check if gmmlib is installed and act as follows:
    - if gmmlib is **not** installed on system, media-driver will build and install gmmlib on system (it will require local gmmlib sources for this). In this situation, compute-runtime will continue to give the faltal error at build time.
    - if gmmlib is installed on system, media-driver will **not** build and install gmmlib files. In this situation, compute-runtime will build fine.

intel-media-driver developers are having a hard time to handle the gmmlib dependency due to internal regressions. The current media-driver git master branch is still relying on building/installing gmmlib, while the pre-releases have included the option to use gmmlib from system. This is better explained by a intel-media-driver developer on [this comment](https://github.com/intel/media-driver/issues/299#issuecomment-418156001).

JacekDanecki, 2018-09-05:
My workspace looks like:
```
build
build-neo
compute-runtime
gmmlib
libva
media-driver
```
and I'm using repostitories:
```
~/compute-runtime$ git show -q 
commit fb735e5ee52c685bc051011bc24c98ec595629da
Author: Mrozek, Michal <michal.mrozek@intel.com>
Date:   Mon Sep 3 16:44:42 2018 +0200

    Move SSH size obtaining logic to dedicated function.
    
    Change-Id: I1ea2ceb69f0ae06fe8e37f7769015933295dd6cd

~/gmmlib$ git show -q 
commit 99ff76418179b4fb069235ff6c028330c802563a
Author: Shaik, Johnbasha <johnbasha.shaik@intel.com>
Date:   Fri Aug 24 04:09:54 2018 -0700

    Use exported APIs from ClientContext
    
    [CP] dynamic gmm header file change copy from CL793947
    
    Because certain ResourceInfo exported APIs are inlined,
    GmmLib is accidently exposing internal functions/class members.
    This is causing build issues with unresolved symbols when clients switch to dynamic linking.
    
    Fix is to use ClientContext exported APIs for UMD clients.
    
    Change-Id: I58db29d2f31c3f2586de80369f5ed085e14f06be
    Signed-off-by: drprajap <dimpalben.r.prajapati@intel.com>

~/media-driver$ git show -q 
commit 586fbfbd2c61ecc4a6f2236791dd8edda8a16436
Author: XinfengZhang <carl.zhang@intel.com>
Date:   Tue Jul 17 05:48:43 2018 +0800

    fix vp8 decode issue #190, change the first partition size
    
    correct first partition_size,macroblock offset is 8 aligned
    
    Signed-off-by: XinfengZhang <carl.zhang@intel.com>
```
To compile media-driver, gmmlib and Neo I've used below commands. GmmLib was not installed on the system.
```
$ cd build
$ cmake -DCMAKE_INSTALL_PREFIX=/usr ../media-driver
...
SourceConfiguration:
CommonDir: /root/media-driver/../gmmlib/Source/Common
IncDir: /root/media-driver/../gmmlib/Source/inc
...

$ make -j 10
$ make install

$ cd ../build-neo
$ cmake -DBUILD_TYPE=Release -DCMAKE_BUILD_TYPE=Release ../compute-runtime/
...
--   Found igdgmm, version 0.1.0
-- GmmLib include dirs: /usr/include/igdgmm;/usr/include/igdgmm/GmmLib;/usr/include/igdgmm/GmmLib/inc;/usr/include/igdgmm/inc;/usr/include/igdgmm/inc/common;/usr/include/igdgmm/util
...
$ make -j 10
```
I've used IGC and opencl-clang from ppa:
```
deb http://ppa.launchpad.net/jdanecki/intel-opencl/ubuntu xenial main
```
with versions:
```
$ dpkg -l intel*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                      Version                   Architecture              Description
+++-=========================================-=========================-=========================-=======================================================================================
ii  intel-igc-core                            18.33.809-1~ppa1~xenial1  amd64                     Intel(R) Graphics Compiler Core, runtime library
ii  intel-igc-opencl                          18.33.809-1~ppa1~xenial1  amd64                     Intel(R) Graphics Compiler Frontend for OpenCL(TM)
ii  intel-igc-opencl-dev                      18.33.809-1~ppa1~xenial1  amd64                     Intel(R) Graphics Compiler development package for OpenCL(TM)
ii  intel-opencl-clang                        4.0.15-1~ppa1~xenial1     amd64                     opencl-clang, runtime library
ii  intel-opencl-clang-dev                    4.0.15-1~ppa1~xenial1     amd64                     opencl-clang, development files
```

dbermond, 2018-09-08:
@JacekDanecki I was using a different gmmlib directory name on the workspace when building intel-media-driver. This was leading to wrong directories at intel-media-driver install time. Using directory `gmmlib` in intel-media-driver workspace makes it install fine.

Anyway, intel-media-driver has allowed again the use of dynamic (installed) gmmlib in their current git master branch since September 6, 2018 through commit https://github.com/intel/media-driver/commit/63dd4ae121542e281ffd76a943b8ece1f18acbb5.

This issue is now solved by the intel-media-driver side and I think it can be closed.

Thanks to everyone who participated here.

closed 2018-09-08 by dbermond