intel/compute-runtime #62: NEO is slower than Beignet on Intel HD Graphics 500

adamradocz, 2018-07-19:
Hi,

I use BOINC on my NAS (Intel J3455 CPU with  Intel HD Graphics 500) and it shows that the NEO driver (72 GFLOPS) slower then the Beignet driver (96 GFLOPS).

[image]

[image]

Although, my laptop (Intel 8250U CPU with Intel UHD Graphics 620) faster with the NEO Driver (211 GFLOPS) than the Beignet (192 GFLOPS)

[image]

[image]

PiotrRozenfeld, 2018-07-19:
Could you please provide reproduction steps so we can replicate the issue on our end?

adamradocz, 2018-07-19:
I tested with two different Dockers.

Beignet: `docker run -it --name boinc-beignet --device /dev/dri:/dev/dri --rm boinc/client:intel-legacy`

NEO: `docker run -it --name boinc-neo --device /dev/dri:/dev/dri --rm boinc/client:intel`

JacekDanecki, 2018-07-19:
I've the same results on KBL machine:
```
OpenCL: Intel GPU 0: Intel(R) HD Graphics Kabylake Halo GT2 (driver version 1.3, device version OpenCL 2.0 beignet 1.3, 4096MB, 4096MB available, 192 GFLOPS peak)
OpenCL: Intel GPU 0: Intel(R) Gen9 HD Graphics NEO (driver version 18.27.11048, device version OpenCL 2.1 NEO, 25637MB, 25637MB available, 211 GFLOPS peak)
```

MichalMrozek, 2018-07-19:
@adamradocz  Can you execute clinfo on your Intel HD Graphics 500 and provide a value of 

CL_DEVICE_MAX_COMPUTE_UNITS ( Max Compute Units )

with Neo and Beignet ?

adamradocz, 2018-07-19:
@MichalMrozek 
Beignet:
```Number of platforms                               1
  Platform Name                                   Intel Gen OCL Driver
  Platform Vendor                                 Intel
  Platform Version                                OpenCL 2.0 beignet 1.3
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_3d_image_writes cl_khr_image2d_from_buffer cl_khr_depth_images cl_khr_spir cl_khr_icd cl_intel_accelerator cl_intel_subgroups cl_intel_subgroups_short cl_khr_gl_sharing
  Platform Extensions function suffix             Intel

  Platform Name                                   Intel Gen OCL Driver
Number of devices                                 1
  Device Name                                     Intel(R) HD Graphics Broxton 1
  Device Vendor                                   Intel
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 2.0 beignet 1.3
  Driver Version                                  1.3
  Device OpenCL C Version                         OpenCL C 2.0 beignet 1.3
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
  Max compute units                               12
  Max clock frequency                             1000MHz
  Device Partition                                (core)
    Max number of sub-devices                     1
    Supported partition types                     None, None, None
  Max work item dimensions                        3
  Max work item sizes                             512x512x512
  Max work group size                             512
  Preferred work group size multiple              16
  Preferred / native vector sizes
    char                                                16 / 8
    short                                                8 / 8
    int                                                  4 / 4
    long                                                 2 / 2
    half                                                 0 / 8        (cl_khr_fp16)
    float                                                4 / 4
    double                                               0 / 2        (n/a)
  Half-precision Floating-point support           (cl_khr_fp16)
    Denormals                                     No
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 No
    Round to infinity                             No
    IEEE754-2008 fused multiply-add               No
    Support is emulated in software               No
  Single-precision Floating-point support         (core)
    Denormals                                     No
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 No
    Round to infinity                             No
    IEEE754-2008 fused multiply-add               No
    Support is emulated in software               No
    Correctly-rounded divide and sqrt operations  No
  Double-precision Floating-point support         (n/a)
  Address bits                                    32, Little-Endian
  Global memory size                              1987051520 (1.851GiB)
  Error Correction support                        No
  Max memory allocation                           1490026496 (1.388GiB)
  Unified memory for Host and Device              Yes
  Shared Virtual Memory (SVM) capabilities        (core)
    Coarse-grained buffer sharing                 Yes
    Fine-grained buffer sharing                   No
    Fine-grained system sharing                   No
    Atomics                                       No
  Minimum alignment for any data type             128 bytes
  Alignment of base address                       1024 bits (128 bytes)
[... 176 lines cut ...]
      Supports preemption                         No
  Device Extensions                               cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_fp16 cl_khr_depth_images cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_icd cl_khr_image2d_from_buffer cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_intel_subgroups cl_intel_required_subgroup_size cl_intel_subgroups_short cl_khr_spir cl_intel_accelerator cl_intel_media_block_io cl_intel_driver_diagnostics cl_intel_device_side_avc_motion_estimation cl_khr_priority_hints cl_khr_throttle_hints cl_khr_create_command_queue cl_khr_fp64 cl_intel_planar_yuv cl_intel_packed_yuv cl_intel_motion_estimation cl_intel_advanced_motion_estimation

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  Intel(R) OpenCL HD Graphics
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   Success [INTEL]
  clCreateContext(NULL, ...) [default]            Success [INTEL]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  Success (1)
    Platform Name                                 Intel(R) OpenCL HD Graphics
[... cut ...]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (1)
    Platform Name                                 Intel(R) OpenCL HD Graphics
    Device Name                                   Intel(R) Gen9 HD Graphics NEO

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.11
  ICD loader Profile                              OpenCL 2.1
```

MichalMrozek, 2018-07-19:
Ok it makes sense now, this is due to incorrect clock frequency reported by the driver.

Neo
Max clock frequency 750MHz

Beignet 
Max clock frequency  1000MHz

This device cannot achieve 1000Mhz see here:
 https://en.wikipedia.org/wiki/Intel_HD,_UHD_and_Iris_Graphics

So Neo score is correct.

adamradocz, 2018-07-19:
I see. Thank you very much!

closed 2018-07-19 by adamradocz