intel/compute-runtime #67: Build failure with version 18.29.11114

adydychk, 2018-07-27:
We tried to compile this pre-release https://github.com/intel/compute-runtime/releases/tag/18.29.11114 and we have problems with KhronosGroup/OpenCL-Headers@f039db6.

We got following failure: 
```
In file included from /root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/common_clang.cpp:19:
/root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/common_clang.h:20:10: fatal error: 'CL/cl.h' file not found
#include "CL/cl.h"
         ^~~~~~~~~
In file included from /root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/link.cpp:19:
/root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/common_clang.h:20:10: fatal error: 'CL/cl.h' file not found
#include "CL/cl.h"
         ^~~~~~~~~
1 error generated.
make[5]: *** [projects/cclang/CMakeFiles/opencl_clang.dir/link.cpp.o] Error 1
make[5]: *** Waiting for unfinished jobs....
1 error generated.
make[5]: *** [projects/cclang/CMakeFiles/opencl_clang.dir/common_clang.cpp.o] Error 1
make[4]: *** [projects/cclang/CMakeFiles/opencl_clang.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [igc/IGC/BiFModule/clang_build/src/src/cclang-stamp/cclang-build] Error 2
make[1]: *** [igc/IGC/BiFModule/clang_build/CMakeFiles/cclang.dir/all] Error 2
make: *** [all] Error 2
```

As you see the output is  `fatal error: 'CL/cl.h' file not found `. And it is true, folders hierarchy in this repositroy with this commit is different if we compare with HEAD status. After that we moved `opencl11/CL/` to `CL/` and tried to build (again failure):
```
In file included from /root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/common_clang.cpp:19:
/root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/common_clang.h:77:11: error: unknown type name 'cl_kernel_arg_address_qualifier'
  virtual cl_kernel_arg_address_qualifier
          ^
/root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/common_clang.h:79:11: error: unknown type name 'cl_kernel_arg_access_qualifier'
  virtual cl_kernel_arg_access_qualifier
          ^
In file included from /root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/link.cpp:19:
/root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/common_clang.h:77:11: error: unknown type name 'cl_kernel_arg_address_qualifier'
  virtual cl_kernel_arg_address_qualifier
         ^
/root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/common_clang.h:81:11: error: unknown type name 'cl_kernel_arg_type_qualifier'
  virtual cl_kernel_arg_type_qualifier
          ^
/root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/common_clang.h:79:11: error: unknown type name 'cl_kernel_arg_access_qualifier'
  virtual cl_kernel_arg_access_qualifier
          ^
/root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/common_clang.h:81:11: error: unknown type name 'cl_kernel_arg_type_qualifier'
  virtual cl_kernel_arg_type_qualifier
          ^
In file included from /root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/common_clang.cpp:24:
/root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/options.h:64:7: warning: 'OpenCLArgList' has virtual functions but non-virtual destructor [-Wnon-virtual-dtor]
class OpenCLArgList : public llvm::opt::ArgList {
      ^
In file included from /root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/link.cpp:21:
/root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/options.h:64:7: warning: 'OpenCLArgList' has virtual functions but non-virtual destructor [-Wnon-virtual-dtor]
class OpenCLArgList : public llvm::opt::ArgList {
      ^
/root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/link.cpp:361:24: error: use of undeclared identifier 'CL_LINK_PROGRAM_FAILURE'
    pResult->setResult(CL_LINK_PROGRAM_FAILURE);
                       ^
/root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/link.cpp:370:24: error: use of undeclared identifier 'CL_LINK_PROGRAM_FAILURE'
    pResult->setResult(CL_LINK_PROGRAM_FAILURE);
                       ^
1 warning and 5 errors generated.
make[5]: *** [projects/cclang/CMakeFiles/opencl_clang.dir/link.cpp.o] Error 1
make[5]: *** Waiting for unfinished jobs....
/root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/common_clang.cpp:252:14: error: use of undeclared identifier 'CL_COMPILE_PROGRAM_FAILURE'
      return CL_COMPILE_PROGRAM_FAILURE;
             ^
/root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/common_clang.cpp:292:35: error: use of undeclared identifier 'CL_COMPILE_PROGRAM_FAILURE'
    return success ? CL_SUCCESS : CL_COMPILE_PROGRAM_FAILURE;
                                  ^
/root/opencl_build/build/igc/IGC/BiFModule/clang_build/src/src/cclang/projects/cclang/common_clang.cpp:303:12: error: use of undeclared identifier 'CL_COMPILE_PROGRAM_FAILURE'
    return CL_COMPILE_PROGRAM_FAILURE;
           ^
1 warning and 6 errors generated.
make[5]: *** [projects/cclang/CMakeFiles/opencl_clang.dir/common_clang.cpp.o] Error 1
make[4]: *** [projects/cclang/CMakeFiles/opencl_clang.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [igc/IGC/BiFModule/clang_build/src/src/cclang-stamp/cclang-build] Error 2
make[1]: *** [igc/IGC/BiFModule/clang_build/CMakeFiles/cclang.dir/all] Error 2
make: *** [all] Error 2
```

Then we moved `opencl22/CL/` to `CL/`. It compiled but tests failed:
```
/root/opencl_build/neo/unit_tests/event/event_tests.cpp:578: Failure
Value of: output.c_str()
  Actual: "waitForCompletionWithTimeout counter 0\ntest"
Expected: "test"

/root/opencl_build/neo/unit_tests/memory_leak_listener.cpp:56: Failure
Value of: (int)MemoryManagement::failingAllocation
  Actual: -1
Expected: leak
Which is: 95
To locate call stack, change the value of captureCallStacks to true
[  FAILED  ] InternalsEventTest.givenBlockedKernelWithPrintfWhenSubmittedThenPrintOutput

=====================
==   ULTs FAILED   ==
=====================
Tests run:      11551
Tests passed:   11550
Tests failed:   1
Tests disabled: 19
Time elapsed:  342 ms
=====================
[  FAILED  ] InternalsEventTest.givenBlockedKernelWithPrintfWhenSubmittedThenPrintOutput

make[2]: *** [run_bxt_unit_tests] Error 1
make[1]: *** [unit_tests/CMakeFiles/run_bxt_unit_tests.dir/all] Error 2
make: *** [all] Error 2
```

I want to mention that if we build HEAD (by instructions from here https://github.com/intel/compute-runtime/blob/master/documentation/BUILD_Centos.md) all is ok and we got `intel-opencl-18.30-0.x86_64-igdrcl.rpm`.

The question is which version / folder of `CL/` should we use from KhronosGroup/OpenCL-Headers ?

JacekDanecki, 2018-07-27:
External OpenCL headers are needed to compile cclang, and we use opencl22/CL directory
```
ln -s ./opencl22/CL ./opencl_headers/
```
As reference you can see [spec](https://copr-be.cloud.fedoraproject.org/results/arturh/intel-opencl/srpm-builds/00770166/opencl-clang-4.0.15-ep7.spec) file used to build opencl-clang rpm package in copr.
In [ppa](https://launchpad.net/~jdanecki/+archive/ubuntu/intel-opencl/+sourcepub/9270995/+listing-archive-extra) I'm using the same directory to opencl headers.
Neo uses Khronos headers from third_party directory inside Neo [repo](https://github.com/intel/compute-runtime/tree/master/third_party/opencl_headers/CL).
Can you look at output from cmake, should be similar to:
```
Khronos OpenCL headers dir: /opt/src/neo/third_party/opencl_headers
CCLANG_OCL_HEADERS_DIR = /opt/src/common_clang/../opencl_headers
```

adydychk, 2018-07-27:
We fixed our problem:
1. Took `opencl22/CL` as you mentioned
2. Added file cl_version.h from repo: https://github.com/KhronosGroup/OpenCL-Headers
After that the build and packaging were successful.

Please, fix release notes make it more clear :)

JacekDanecki, 2018-07-27:
[cl_version.h](https://github.com/intel/compute-runtime/blob/master/third_party/opencl_headers/CL/cl_version.h) file is in third_party/opencl/CL directory. Where did you observe problem with this header?

JacekDanecki, 2018-09-03:
Closing this issue, as package was created correctly. Please reopen it or create new issue, if you find any issue during build.

closed 2018-09-03 by JacekDanecki