intel/compute-runtime #75: How can I dump ISA?

wzw-intel, 2018-08-15:
Hi,
In Beignet, I can set environment variable OCL_OUTPUT_ASM to "1" to dump Gen ISA. 

Is there any similar way for Neo to do that?

alalek, 2018-08-15:
Are there some tools to extract ISA from OpenCL kernel binaries / files from "cl_cache" dir?

[Intel OpenCL SDK](https://software.intel.com/en-us/intel-opencl) has OpenCL kernel compiler (including ISA emit), but these tools are outdated and they are not able to work with NEO OpenCL.

MichalMrozek, 2018-08-16:
Have you tried this ?
https://github.com/intel/opencl-intercept-layer/blob/master/docs/kernel_isa.md

alalek, 2018-08-16:
Awesome! Thank you!

MichalMrozek, 2018-08-22:
Looks like we can close this.

closed 2018-08-22 by MichalMrozek

wzw-intel, 2018-09-04:
@MichalMrozek Thank you very much for your hint.

@alalek Your links are very helpful, thank you.