intel/compute-runtime #76: Debian package doesn't register the ICD

ThijsWithaar, 2018-08-25:
When installing the .debs generated using CPack, no .icd is placed in /etc/OpenCL/vendors
clinfo doesn't show the platform.

ArturHarasimiuk, 2018-08-27:
which OS you used? which version?

could you share output from command: `dpkg-deb -c <package name>`, where package is generated by CPack.
please attach install log if possible.

ThijsWithaar, 2018-08-27:
The os is debian, buster. I've following instructions from [Ubuntu](https://github.com/intel/compute-runtime/blob/master/documentation/BUILD_Ubuntu.md).

The output of dpkg-deb is:

```
$ dpkg-deb -c intel-igc-opencl_0.1-0_amd64.deb
drwxr-xr-x root/root         0 2018-08-25 12:18 ./usr/
drwxr-xr-x root/root         0 2018-08-25 12:18 ./usr/lib/
-rw-r--r-- root/root   5200712 2018-08-25 12:14 ./usr/lib/libigdfcl.so
-rw-r--r-- root/root  59460496 2018-08-14 22:48 ./usr/lib/libopencl_clang.so

$ dpkg-deb -c intel-igc-core_0.1-0_amd64.deb 
drwxr-xr-x root/root         0 2018-08-25 12:18 ./usr/
drwxr-xr-x root/root         0 2018-08-25 12:18 ./usr/lib/
-rw-r--r-- root/root   8186320 2018-08-25 10:06 ./usr/lib/libiga64.so
-rw-r--r-- root/root 208787000 2018-08-25 12:14 ./usr/lib/libigc.so
```

ArturHarasimiuk, 2018-08-27:
thanks. what about 'intel-opencl' package? intel.icd should be there. above packages are IGC specific and we don't use them so far.

ThijsWithaar, 2018-08-27:
I'm not sure where the 'intel-opencl' package should come from.
Is there maybe just something silly I'm doing wrong. I followed the Cmake instructions for ubuntu, which gives me the following 4 packages:

```
$ ls *.deb
intel-igc-core_0.1-0_amd64.deb
intel-igc-opencl_0.1-0_amd64.deb
intel-igc-media_0.1-0_amd64.deb
intel-igc-opencl-devel_0.1-0_amd64.deb
```

ArturHarasimiuk, 2018-08-27:
ok. let me try to reproduce problem on my side.

ArturHarasimiuk, 2018-08-29:
@ThijsWithaar could you please share content of CPackConfig.cmake file? it should help in further investigation
this file is generated by cmake during build files generation and can be found in build folder.

ThijsWithaar, 2018-08-30:
Sure, it's contents are:

```
# This file will be configured to contain variables for CPack. These variables
# should be set in the CMake list file of the project before CPack module is
# included. The list of available CPACK_xxx variables and their associated
# documentation may be obtained using
#  cpack --help-variable-list
#
# Some variables are common to all generators (e.g. CPACK_PACKAGE_NAME)
# and some are specific to a generator
# (e.g. CPACK_NSIS_EXTRA_INSTALL_COMMANDS). The generator specific variables
# usually begin with CPACK_<GENNAME>_xxxx.

set(CPACK_ARCHIVE_COMPONENT_INSTALL "ON")
set(CPACK_ARCHIVE_IGC-CORE_FILE_NAME "intel-igc-core-0.1-0.x86_64")
set(CPACK_ARCHIVE_IGC-MEDIA_FILE_NAME "intel-igc-media-0.1-0.x86_64")
set(CPACK_ARCHIVE_IGC-OPENCL-DEVEL_FILE_NAME "intel-igc-opencl-devel-0.1-0.x86_64")
set(CPACK_ARCHIVE_IGC-OPENCL_FILE_NAME "intel-igc-opencl-0.1-0.x86_64")
set(CPACK_BINARY_7Z "")
set(CPACK_BINARY_BUNDLE "")
set(CPACK_BINARY_CYGWIN "")
set(CPACK_BINARY_DEB "OFF")
set(CPACK_BINARY_DRAGNDROP "")
set(CPACK_BINARY_FREEBSD "OFF")
set(CPACK_BINARY_IFW "OFF")
set(CPACK_BINARY_NSIS "OFF")
set(CPACK_BINARY_NUGET "")
set(CPACK_BINARY_OSXX11 "")
set(CPACK_BINARY_PACKAGEMAKER "")
set(CPACK_BINARY_PRODUCTBUILD "")
set(CPACK_BINARY_RPM "OFF")
set(CPACK_BINARY_STGZ "ON")
set(CPACK_BINARY_TBZ2 "OFF")
set(CPACK_BINARY_TGZ "ON")
set(CPACK_BINARY_TXZ "OFF")
set(CPACK_BINARY_TZ "ON")
set(CPACK_BINARY_WIX "")
set(CPACK_BINARY_ZIP "")
set(CPACK_BUILD_SOURCE_DIRS "/home/thijs/Source/intel-compute-runtime/workspace/igc/IGC;/home/thijs/Source/intel-compute-runtime/workspace/build_igc")
set(CPACK_CMAKE_GENERATOR "Ninja")
set(CPACK_COMPONENTS_ALL "igc-core;igc-media;igc-opencl;igc-opencl-devel")
set(CPACK_COMPONENTS_ALL_SET_BY_USER "TRUE")
set(CPACK_COMPONENT_UNSPECIFIED_HIDDEN "TRUE")
set(CPACK_COMPONENT_UNSPECIFIED_REQUIRED "TRUE")
set(CPACK_DEBIAN_IGC-CORE_FILE_NAME "intel-igc-core_0.1-0_amd64.deb")
set(CPACK_DEBIAN_IGC-MEDIA_FILE_NAME "intel-igc-media_0.1-0_amd64.deb")
set(CPACK_DEBIAN_IGC-MEDIA_PACKAGE_DEPENDS "igc-igc-core")
set(CPACK_DEBIAN_IGC-OPENCL-DEVEL_FILE_NAME "intel-igc-opencl-devel_0.1-0_amd64.deb")
set(CPACK_DEBIAN_IGC-OPENCL-DEVEL_PACKAGE_DEPENDS "igc-igc-opencl")
set(CPACK_DEBIAN_IGC-OPENCL_FILE_NAME "intel-igc-opencl_0.1-0_amd64.deb")
set(CPACK_DEBIAN_IGC-OPENCL_PACKAGE_DEPENDS "igc-igc-core")
set(CPACK_DEBIAN_PACKAGE_ARCHITECTURE "amd64")
set(CPACK_DEBIAN_PACKAGE_DESCRIPTION "Intel(R) Graphics Compiler for OpenCL(TM)")
set(CPACK_DEBIAN_PACKAGE_MAINTAINER "intel")
set(CPACK_DEBIAN_PACKAGE_SHLIBDEPS "ON")
set(CPACK_DEB_COMPONENT_INSTALL "ON")
set(CPACK_GENERATOR "DEB")
set(CPACK_INSTALL_CMAKE_PROJECTS "/home/thijs/Source/intel-compute-runtime/workspace/build_igc;IGC;ALL;/")
set(CPACK_INSTALL_PREFIX "/usr/local")
set(CPACK_MODULE_PATH "")
set(CPACK_NSIS_DISPLAY_NAME "/usr")
set(CPACK_NSIS_INSTALLER_ICON_CODE "")
set(CPACK_NSIS_INSTALLER_MUI_ICON_CODE "")
set(CPACK_NSIS_INSTALL_ROOT "$PROGRAMFILES")
set(CPACK_NSIS_PACKAGE_NAME "/usr")
set(CPACK_OUTPUT_CONFIG_FILE "/home/thijs/Source/intel-compute-runtime/workspace/build_igc/CPackConfig.cmake")
set(CPACK_PACKAGE_ARCHITECTURE "x86_64")
set(CPACK_PACKAGE_DEFAULT_LOCATION "/")
set(CPACK_PACKAGE_DESCRIPTION_FILE "/usr/share/cmake-3.12/Templates/CPack.GenericDescription.txt")
set(CPACK_PACKAGE_DESCRIPTION_SUMMARY "IGC built using CMake")
set(CPACK_PACKAGE_FILE_NAME "IGC-0.1.0-Linux")
set(CPACK_PACKAGE_INSTALL_DIRECTORY "/usr")
set(CPACK_PACKAGE_INSTALL_REGISTRY_KEY "/usr")
set(CPACK_PACKAGE_NAME "IGC")
set(CPACK_PACKAGE_RELOCATABLE "true")
set(CPACK_PACKAGE_VENDOR "Humanity")
set(CPACK_PACKAGE_VERSION "0.1.0")
set(CPACK_PACKAGE_VERSION_MAJOR "0")
set(CPACK_PACKAGE_VERSION_MINOR "1")
set(CPACK_PACKAGE_VERSION_PATCH "0")
set(CPACK_RESOURCE_FILE_LICENSE "/usr/share/cmake-3.12/Templates/CPack.GenericLicense.txt")
set(CPACK_RESOURCE_FILE_README "/usr/share/cmake-3.12/Templates/CPack.GenericDescription.txt")
set(CPACK_RESOURCE_FILE_WELCOME "/usr/share/cmake-3.12/Templates/CPack.GenericWelcome.txt")
set(CPACK_RPM_COMPONENT_INSTALL "ON")
set(CPACK_RPM_IGC-CORE_FILE_NAME "intel-igc-core-0.1-0.x86_64.rpm")
set(CPACK_RPM_IGC-MEDIA_FILE_NAME "intel-igc-media-0.1-0.x86_64.rpm")
set(CPACK_RPM_IGC-MEDIA_PACKAGE_DEPENDS "igc-igc-core")
set(CPACK_RPM_IGC-OPENCL-DEVEL_FILE_NAME "intel-igc-opencl-devel-0.1-0.x86_64.rpm")
set(CPACK_RPM_IGC-OPENCL-DEVEL_PACKAGE_DEPENDS "igc-igc-opencl")
set(CPACK_RPM_IGC-OPENCL_FILE_NAME "intel-igc-opencl-0.1-0.x86_64.rpm")
set(CPACK_RPM_IGC-OPENCL_PACKAGE_DEPENDS "igc-igc-core")
set(CPACK_RPM_PACKAGE_ARCHITECTURE "x86_64")
set(CPACK_RPM_PACKAGE_RELOCATABLE "TRUE")
set(CPACK_SET_DESTDIR "OFF")
set(CPACK_SOURCE_7Z "")
set(CPACK_SOURCE_CYGWIN "")
set(CPACK_SOURCE_GENERATOR "TBZ2;TGZ;TXZ;TZ")
set(CPACK_SOURCE_OUTPUT_CONFIG_FILE "/home/thijs/Source/intel-compute-runtime/workspace/build_igc/CPackSourceConfig.cmake")
set(CPACK_SOURCE_RPM "OFF")
set(CPACK_SOURCE_TBZ2 "ON")
set(CPACK_SOURCE_TGZ "ON")
set(CPACK_SOURCE_TXZ "ON")
set(CPACK_SOURCE_TZ "ON")
set(CPACK_SOURCE_ZIP "OFF")
set(CPACK_SYSTEM_NAME "Linux")
set(CPACK_TOPLEVEL_TAG "Linux")
set(CPACK_WIX_SIZEOF_VOID_P "8")

if(NOT CPACK_PROPERTIES_FILE)
  set(CPACK_PROPERTIES_FILE "/home/thijs/Source/intel-compute-runtime/workspace/build_igc/CPackProperties.cmake")
endif()

if(EXISTS ${CPACK_PROPERTIES_FILE})
  include(${CPACK_PROPERTIES_FILE})
endif()
```

ArturHarasimiuk, 2018-08-30:
thanks!
based on this
```
set(CPACK_BUILD_SOURCE_DIRS "/home/thijs/Source/intel-compute-runtime/workspace/igc/IGC;/home/thijs/Source/intel-compute-runtime/workspace/build_igc")
```
it looks that you tried to build intel-graphic-compiler (igc) not compute-runtime (neo).

could you please double check that path to source in cmake invocation is `../neo` not `../igc`. assuming that ninja is generator, it should be:
```
cmake -G Ninja -DBUILD_TYPE=Release -DCMAKE_BUILD_TYPE=Release ../neo/
```

ThijsWithaar, 2018-08-30:
Ah, that's the problem indeed.
Thanks a lot for the help.

ArturHarasimiuk, 2018-08-31:
excellent! closing issue.

closed 2018-08-31 by ArturHarasimiuk