intel/compute-runtime #84: <getWGsizes:1175: build program : error -6> line in Preferred work group size multiple under OpenSUSE Leap 15

KhazAkar, 2018-09-12:
Hi everyone! I cannot use compute-runtime, because of <getWGsizes:1175: build program : error -6> in line Preferred work group size multiple when running clinfo, so I cannot use for example DaVinci Resolve or Darktable effectively on my Intel core i7 6820HQ notebook. I'm using prebuilded Fedora 28 package on OpenSUSE Leap 15.
[Here's my clinfo](https://pastebin.com/6DvNZz9W)

KhazAkar, 2018-09-12:
It looks like it's glibc problem.
https://pastebin.com/SYpMegBH

JacekDanecki, 2018-09-12:
libm.so.6 in OpenSuse Leap 15 is to old to use rpm package for Fedora 26
```  Max work group size                             256
       316:     find library=libigdfcl.so [0]; searching
       316:      search cache=/etc/ld.so.cache
       316:       trying file=/usr/local/lib64/libigdfcl.so
       316:
       316:     find library=librt.so.1 [0]; searching
       316:      search cache=/etc/ld.so.cache
       316:       trying file=/lib64/librt.so.1
       316:
       316:     find library=libz.so.1 [0]; searching
       316:      search cache=/etc/ld.so.cache
       316:       trying file=/lib64/libz.so.1
       316:
       316:
       316:     calling init: /lib64/libz.so.1
       316:
       316:
       316:     calling init: /lib64/librt.so.1
       316:
       316:
       316:     calling init: /usr/local/lib64/libigdfcl.so
       316:
       316:     find library=libigc.so [0]; searching
       316:      search cache=/etc/ld.so.cache
       316:       trying file=/usr/local/lib64/libigc.so
       316:
       316:     /lib64/libm.so.6: error: version lookup error: version `GLIBC_2.27' not found (required by /usr/local/lib64/libigc.so) (fatal)
  Preferred work group size multiple              <getWGsizes:404: build program : error -6>
```
libm.so.6 supports:
```strings /lib64/libm.so.6 | grep ^GLIBC_2
GLIBC_2.2.5
GLIBC_2.4
GLIBC_2.15
GLIBC_2.18
GLIBC_2.15
GLIBC_2.4
GLIBC_2.18
```
You can use rpm package for Fedora 27
```
$ rpm -q intel-opencl
intel-opencl-18.36.11428-1.fc27.x86_64

```

KhazAkar, 2018-09-12:
Ok, thank you a lot for this :)

JacekDanecki, 2018-09-12:
I've just verified clpeak in OpenSuse docker image 
```
REPOSITORY                                            TAG                 IMAGE ID            CREATED             SIZE
docker.io/opensuse                                    leap                35057ab4ef08        4 months ago        110 MB
```
on Intel(R) Core(TM) i7-7820HQ CPU @ 2.90GHz, and it works fine. I think we can close this issue.
```
# clpeak 

Platform: Intel(R) OpenCL HD Graphics
  Device: Intel(R) Gen9 HD Graphics NEO
    Driver version  : 18.36.11428 (Linux x64)
    Compute units   : 24
    Clock frequency : 1100 MHz

    Global memory bandwidth (GBPS)
      float   : 26.28
      float2  : 27.11
      float4  : 28.44
      float8  : 28.48
      float16 : 25.29

    Single-precision compute (GFLOPS)
      float   : 418.83
      float2  : 418.56
      float4  : 418.01
      float8  : 417.48
      float16 : 411.94

    half-precision compute (GFLOPS)
      half   : 718.24
      half2  : 717.45
      half4  : 715.53
      half8  : 711.05
      half16 : 703.41

    Double-precision compute (GFLOPS)
      double   : 104.89
      double2  : 104.79
      double4  : 104.49
      double8  : 103.96
      double16 : 102.81

    Transfer bandwidth (GBPS)
      enqueueWriteBuffer         : 33.08
      enqueueReadBuffer          : 11.70
      enqueueMapBuffer(for read) : 7158278.50
        memcpy from mapped ptr   : 12.11
      enqueueUnmap(after write)  : inf
        memcpy to mapped ptr     : 12.13

    Kernel launch latency : 27.81 us

# lsb_release -a
LSB Version:    core-2.0-noarch:core-3.2-noarch:core-4.0-noarch:core-2.0-x86_64:core-3.2-x86_64:core-4.0-x86_64:desktop-4.0-amd64:desktop-4.0-noarch:graphics-2.0-amd64:graphics-2.0-noarch:graphics-3.2-amd64:graphics-3.2-noarch:graphics-4.0-amd64:graphics-4.0-noarch
Distributor ID: openSUSE Leap
Description:    openSUSE Leap 42.3
Release:        42.3
Codename:       42.3
```

KhazAkar, 2018-09-12:
Mayby problem in title is resolved, but DaVinci Resolve still cannot see Intel as OpenCL device (darktable can detect it), but this I think is another topic. Thank you very much for support :)

JacekDanecki, 2018-09-12:
Can you provide more details about DaVinci problem or create new issue?

KhazAkar, 2018-09-12:
Yes, I can provide more details about DaVinci problem - It cannot detect Intel OpenCL, but darktable can see it and use it, with nvidia-computeG04 or libOpenCL1. What's more interesting, it passes clpeak, but when It does it, my desktop can freeze several times with some warning in dmesg, here's my results [here](https://pastebin.com/gARGfUp3). And dmesg warning is: `asynchronous wait on fence i915:kwin_x11[9545]/1:a45 timed out`. I'm using Intel drivers for my iGPU, because modesetting drivers just give me a massive tearing issue.

KhazAkar, 2018-09-12:
Second run of clpeak. Freeze still persits, but that results are worse than previous + lsb-release -a
`khazakar@dellprecision5510:~> clpeak

Platform: Intel(R) OpenCL HD Graphics
  Device: Intel(R) Gen9 HD Graphics NEO
    Driver version  : 18.36.11428 (Linux x64)
    Compute units   : 24
    Clock frequency : 0 MHz

    Global memory bandwidth (GBPS)
      float   : 20.42
      float2  : 21.53
      float4  : 24.12
      float8  : 23.64
      float16 : 19.85

    Single-precision compute (GFLOPS)
      float   : 375.23
      float2  : 373.11
      float4  : 369.89
      float8  : 374.57
      float16 : 365.48

    half-precision compute (GFLOPS)
      half   : 623.30
      half2  : 633.73
      half4  : 633.57
      half8  : 628.67
      half16 : 624.59

    Double-precision compute (GFLOPS)
      double   : 95.02
      double2  : 94.82
      double4  : 94.86
      double8  : 94.12
      double16 : 93.20

    Transfer bandwidth (GBPS)
      enqueueWriteBuffer         : 26.22
      enqueueReadBuffer          : 10.96
      enqueueMapBuffer(for read) : 7158278.50
        memcpy from mapped ptr   : 11.74
      enqueueUnmap(after write)  : inf
        memcpy to mapped ptr     : 11.78

    Kernel launch latency : 28.77 us

khazakar@dellprecision5510:~> lsb-release -a
LSB Version:	core-2.0-noarch:core-3.2-noarch:core-4.0-noarch:core-2.0-x86_64:core-3.2-x86_64:core-4.0-x86_64:desktop-4.0-amd64:desktop-4.0-noarch:graphics-2.0-amd64:graphics-2.0-noarch:graphics-3.2-amd64:graphics-3.2-noarch:graphics-4.0-amd64:graphics-4.0-noarch
Distributor ID:	openSUSE
Description:	openSUSE Leap 15.0
Release:	15.0
Codename:	n/a
khazakar@dellprecision5510:~> `

KhazAkar, 2018-09-12:
`khazakar@dellprecision5510:~> lsb-release -a
LSB Version:	core-2.0-noarch:core-3.2-noarch:core-4.0-noarch:core-2.0-x86_64:core-3.2-x86_64:core-4.0-x86_64:desktop-4.0-amd64:desktop-4.0-noarch:graphics-2.0-amd64:graphics-2.0-noarch:graphics-3.2-amd64:graphics-3.2-noarch:graphics-4.0-amd64:graphics-4.0-noarch
Distributor ID:	openSUSE
Description:	openSUSE Leap 15.0
Release:	15.0
Codename:	n/a
khazakar@dellprecision5510:~> `

JacekDanecki, 2018-09-13:
As I can see in /opt/resolve/logs/ResolveDebug.txt, Davinci skips Neo driver
```
[0x7fa307de7700] | GPUManager           | INFO  | 2018-09-13 13:16:17,350 | Start detecting OpenCL boards
[0x7fa307de7700] | GPUManager           | INFO  | 2018-09-13 13:16:17,350 | Skipping unqualified OpenCL platform: Intel(R) OpenCL HD Graphics
[0x7fa307de7700] | GPUManager           | INFO  | 2018-09-13 13:16:17,350 | 0 OpenCL boards detected
Urgent message: CRITICAL_PREF: DaVinci Resolve could not find any OpenCL capable GPUs. Would you like to review and update your DaVinci Resolve Hardware Configuration? ||Please ensure that
[0x7fa307de7700] | GPUManager           | ERROR | 2018-09-13 13:16:17,350 | CRITICAL_PREF: DaVinci Resolve could not find any OpenCL capable GPUs. Would you like to review and update your D
[0x7fa307de7700] | IP                   | ERROR | 2018-09-13 13:16:17,350 | Failed to initialize OpenCL boards
LS data handler thread starts: 07DE7700
```

although: oclBandwidthTest and  oclDeviceQuery work fine.

Looking at: https://documents.blackmagicdesign.com/DaVinciResolve/20180407-79c607/DaVinci_Resolve_15_Configuration_Guide.pdf Intel GPU with OpenCL should be supported.

KhazAkar, 2018-09-13:
I'm worried about that mine setup don't detect Clock Frequency (shows 0 MHz), mayby that's the problem?

KhazAkar, 2018-09-17:
Ok, now I have Base Clock Speed for OpenCL, but Indigo Benchmark Crash with segmentation fault when trying to use Intel OpenCL device.

JacekDanecki, 2018-09-17:
I've reproduced segmentation fault in Indigo Benchmark on my setup. Let me investigate it.

KhazAkar, 2018-10-04:
Sure, what do you need for that?

JacekDanecki, 2018-10-05:
In case of Indigo one problem was fixed on common clang side, we discovered another issue on IGC side, and IGC team is looking at this.
In case of Davinci, I'm waiting for feedback from team who is working with Black Magic Design.

JacekDanecki, 2019-01-10:
I've not received any feedback about Davinci support under Linux for Intel GPU's, but I've found a few threads in Black Magic Design forum. See following:

- https://forum.blackmagicdesign.com/viewtopic.php?f=21&t=56878&sid=036302c6a2cfde88478cd3289b04fe90&start=1050#p443378
- https://forum.blackmagicdesign.com/viewtopic.php?f=21&t=56878&start=1100#p451134
- https://forum.blackmagicdesign.com/viewtopic.php?f=21&t=56878&start=1150#p456924
- https://forum.blackmagicdesign.com/viewtopic.php?f=21&t=81579

IGC is still working on Indigo issues. I'd recommend create new issue in [IGC](https://github.com/intel/intel-graphics-compiler) repository to track progress.

I'll close this issue, as another problems were fixed. Feel free to create new issue when you find them.

closed 2019-01-10 by JacekDanecki

JacekDanecki, 2019-04-12:
@KhazAkar With this commit 22be2f677d500641d46cf602732a19335dd35ec8 Indigo Benchmark is working.