intel/compute-runtime #87: Changing cl_cache path to non-UAC path?

mrsabhar, 2018-09-19:
Is there way to change cl_cache path to requested directory instead of default location of .exe file.  We are using clDNN and need binaries to optimize initialization time. 
App takes 10-20 seconds to initialize all kernels (clBuildprogram)

MichalMrozek, 2018-09-19:
Can you try to modify #define CL_CACHE_LOCATION "cl_cache" to see if it helps?
vpg-compute-neo\config.h.in

This variable controls where the cache directory is.

mrsabhar, 2018-09-19:
Yes that works. Is it possible to do on release Windows driver?

MichalMrozek, 2018-09-20:
We are exploring possible options to make it more convenient.

AdamCetnerowski, 2019-02-15:
We have updated our [documentation](https://github.com/intel/compute-runtime/blob/master/documentation/FAQ.md) to describe how to set the cl_cache directory. 

This feature is available on:
Linux - since release [18.49.11968](https://github.com/intel/compute-runtime/releases/tag/18.49.11968)
Windows – available in [25.20.100.6519]( https://downloadcenter.intel.com/download/28515/Intel-Graphics-Driver-for-Windows-10?product=80939) 

This should be considered experimental. Feedback is appreciated.

closed 2019-03-08 by ArturHarasimiuk