intel/compute-runtime #89: i7-5600U not found on Ubuntu 16.04.5 LTS

Sec42, 2018-09-23:
I have build and installed according to the documentation, but clinfo does not show anything.

I cross-checked the steps from #57:

lspci |grep VGA:
`00:02.0 VGA compatible controller: Intel Corporation Broadwell-U Integrated Graphics (rev 09)`

LD_DEBUG=files clinfo
```
     15090:
     15090:     file=libOpenCL.so.1 [0];  needed by clinfo [0]
     15090:     file=libOpenCL.so.1 [0];  generating link map
     15090:       dynamic: 0x00007f937682adb0  base: 0x00007f9376621000   size: 0x000000000020a138
     15090:         entry: 0x00007f93766252a0  phdr: 0x00007f9376621040  phnum:                  8
     15090:
     15090:
     15090:     file=libdl.so.2 [0];  needed by clinfo [0]
     15090:     file=libdl.so.2 [0];  generating link map
     15090:       dynamic: 0x00007f937661fd88  base: 0x00007f937641d000   size: 0x00000000002030f0
     15090:         entry: 0x00007f937641dda0  phdr: 0x00007f937641d040  phnum:                  7
     15090:
     15090:
     15090:     file=libc.so.6 [0];  needed by clinfo [0]
     15090:     file=libc.so.6 [0];  generating link map
     15090:       dynamic: 0x00007f9376416ba0  base: 0x00007f9376053000   size: 0x00000000003c99a0
     15090:         entry: 0x00007f9376073950  phdr: 0x00007f9376053040  phnum:                 10
     15090:
     15090:
     15090:     calling init: /lib/x86_64-linux-gnu/libc.so.6
     15090:
     15090:
     15090:     calling init: /lib/x86_64-linux-gnu/libdl.so.2
     15090:
     15090:
     15090:     calling init: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1
     15090:
     15090:
     15090:     initialize program: clinfo
     15090:
     15090:
     15090:     transferring control: clinfo
     15090:
     15090:
     15090:     file=/usr/local/lib/libigdrcl.so [0];  dynamically loaded by /usr/lib/x86_64-linux-gnu/libOpenCL.so.1 [0]
     15090:     file=/usr/local/lib/libigdrcl.so [0];  generating link map
     15090:       dynamic: 0x00007f9375fb83b0  base: 0x00007f9375bbf000   size: 0x00000000004935e8
     15090:         entry: 0x00007f9375bd5350  phdr: 0x00007f9375bbf040  phnum:                  8
     15090:
     15090:
     15090:     file=libpthread.so.0 [0];  needed by /usr/local/lib/libigdrcl.so [0]
     15090:     file=libpthread.so.0 [0];  generating link map
     15090:       dynamic: 0x00007f9375bb9d50  base: 0x00007f93759a2000   size: 0x000000000021c428
     15090:         entry: 0x00007f93759a80b0  phdr: 0x00007f93759a2040  phnum:                  9
     15090:
     15090:
     15090:     file=libstdc++.so.6 [0];  needed by /usr/local/lib/libigdrcl.so [0]
     15090:     file=libstdc++.so.6 [0];  generating link map
     15090:       dynamic: 0x00007f937599ad50  base: 0x00007f9375620000   size: 0x0000000000381400
     15090:         entry: 0x00007f93756a9d20  phdr: 0x00007f9375620040  phnum:                  8
     15090:
     15090:
     15090:     file=libm.so.6 [0];  needed by /usr/local/lib/libigdrcl.so [0]
     15090:     file=libm.so.6 [0];  generating link map
     15090:       dynamic: 0x00007f937561ed88  base: 0x00007f9375317000   size: 0x00000000003080f8
     15090:         entry: 0x00007f937531c600  phdr: 0x00007f9375317040  phnum:                  7
     15090:
     15090:
     15090:     file=libgcc_s.so.1 [0];  needed by /usr/local/lib/libigdrcl.so [0]
     15090:     file=libgcc_s.so.1 [0];  generating link map
     15090:       dynamic: 0x00007f93753162b0  base: 0x00007f9375101000   size: 0x0000000000215910
     15090:         entry: 0x00007f9375103a70  phdr: 0x00007f9375101040  phnum:                  6
     15090:
     15090:
     15090:     calling init: /lib/x86_64-linux-gnu/libpthread.so.0
     15090:
     15090:
     15090:     calling init: /lib/x86_64-linux-gnu/libgcc_s.so.1
     15090:
     15090:
     15090:     calling init: /lib/x86_64-linux-gnu/libm.so.6
     15090:
     15090:
     15090:     calling init: /usr/lib/x86_64-linux-gnu/libstdc++.so.6
     15090:
     15090:
     15090:     calling init: /usr/local/lib/libigdrcl.so
     15090:
     15090:     opening file=/usr/local/lib/libigdrcl.so [0]; direct_opencount=1
     15090:
Number of platforms                               0
     15090:
     15090:     calling fini: clinfo [0]
     15090:
     15090:
     15090:     calling fini: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1 [0]
     15090:
     15090:
     15090:     calling fini: /usr/local/lib/libigdrcl.so [0]
     15090:
     15090:
     15090:     calling fini: /lib/x86_64-linux-gnu/libdl.so.2 [0]
     15090:
     15090:
     15090:     calling fini: /lib/x86_64-linux-gnu/libpthread.so.0 [0]
     15090:
     15090:
     15090:     calling fini: /usr/lib/x86_64-linux-gnu/libstdc++.so.6 [0]
     15090:
     15090:
     15090:     calling fini: /lib/x86_64-linux-gnu/libm.so.6 [0]
     15090:
     15090:
     15090:     calling fini: /lib/x86_64-linux-gnu/libgcc_s.so.1 [0]
     15090:
```

I was previously using the beignet driver which worked fine.

ArturHarasimiuk, 2018-09-23:
could you share output from following commands: 
```lspci -s 00:02.0 -nnn -vvv```
```lsmod | grep i915```
```dmesg | grep i915```
```dmesg | grep drm```

and output from strace, please attach log.txt created by command:
```strace -f -o log.txt clinfo```

Sec42, 2018-09-24:
I'm running Ubuntu
```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.5 LTS"
```
Kernel is
```
Linux karoshi 4.4.0-130-generic #156-Ubuntu SMP Thu Jun 14 08:53:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

ArturHarasimiuk, 2018-09-25:
I was able to reproduce issue on my side. It seems that kernel 4.4.0-130 from Ubuntu does not support Soft-Pin feature on BDW (mentioned earlier). Since this is required by Neo thus driver refuses to start.

can you use more recent kernel? I tried ```4.15.0-34``` and ```4.13.0-16``` and it worked.

Sec42, 2018-10-11:
Just as a followup. I got it to build and run on ubuntu 18.04 (kernel `4.15.0-34`).
A quick comparison with the old beignet driver with luxmark seems to show comparable performance (up to a 6% speed increase).

closed 2018-10-11 by Sec42