intel/compute-runtime #95: Plaidbench failure

bartwojcik, 2018-09-29:
Plaidbench is a benchmark for PlaidML framework, one of few frameworks that support OpenCL. It fails the correctness test on my Skylake machine (Intel HD Graphics 520 / i7-6500U CPU) while passing on both 780GTX and HD7770 (windows, sadly) I tested. So I guess this is the OpenCL implementation's issue. Results:
```
(plaidml-NrXtVPHn) [bw@xps13 ~]$ plaidbench keras mobilenet
Running 1024 examples with mobilenet, batch size 1
INFO:plaidml:Opening device "opencl_intel_intel(r)_gen9_hd_graphics_neo.0"
Downloading data from https://github.com/fchollet/deep-learning-models/releases/download/v0.6/mobilenet_1_0_224_tf.h5
17227776/17225924 [==============================] - 18s 1us/step
Model loaded.
Compiling network...
Warming up ...
Main timing
Example finished, elapsed: 3.561319351196289 (compile), 34.086427450180054 (execution), 0.03328752680681646 (execution per example)
Correctness: FAIL, max_error: 0.6279370188713074, max_abs_error: 0.05687180161476135, fail_ratio: 0.665
```

Driver version: 18.38.11535-1

AdamCetnerowski, 2018-10-12:
Thank you for the report. We added it to our debug queue and hope to have more information for you soon.

JacekDanecki, 2019-01-16:
@bartwojcik This issue was fixed on IGC side. 
With intel/intel-graphics-compiler@64632faa3cdad142b809f63274a5ce1f315489d6 (igc_release_2019-01-15) and Neo from intel/compute-runtime@ 19.01.12103, plaidbench works in keras mobilenet test
```
$ plaidbench keras mobilenet
/usr/lib/python2.7/site-packages/requests/__init__.py:80: RequestsDependencyWarning: urllib3 (1.24.1) or chardet (3.0.4) doesn't match a supported version!
RequestsDependencyWarning)
Running 1024 examples with mobilenet, batch size 1
INFO:plaidml:Opening device "opencl_intel_intel(r)_gen9_hd_graphics_neo.0"
Model loaded.
Compiling network...
Warming up ...
Main timing
Example finished, elapsed: 3.118227005 (compile), 25.8001768589 (execution), 0.0251954852138 (execution per example)
Correctness: PASS, max_error: 1.96229684661e-05, max_abs_error: 1.33365392685e-06, fail_ratio: 0.0
```

JacekDanecki, 2019-02-04:
Closing as mobilenet works with fixes from IGC. Please reopen this issue or create new one, if you find any new problems.

closed 2019-02-04 by JacekDanecki