intel/compute-runtime #96: Slow performance on Ubuntu

lqhuy93, 2018-10-01:
My intel chip specs:

> ** GPU Info:
> - Device: Intel(R) Gen9 HD Graphics NEO
> - Hardware version: OpenCL 2.1 NEO 
> - Software version: 18.37.11481
> - Parallel compute units: 24
> - Local mem size: 65536
> 

The same code pipeline takes ~12ms on Windows but ~28ms on Ubuntu. Did I miss something (like additional flags or linking libraries)?
ps: I also have 1080ti GPU on this PC.

lqhuy93, 2018-10-01:
@JacekDanecki Sorry, I can't share the code here. I'm implementing various kernels (OpenCL) for machine learning purpose (GEMM, convolution kernels, upsampling, downsampling,...) which use _cl_intel_subgroups_ and few more Intel's extensions for optimization.
My OS is Ubuntu 16.04. The above times are the average times after inferring my model (~10000 times)

MichaelRoyceCarroll, 2018-10-01:
Hello @lqhuy93 

Thanks for your interest.

Intel has a portal for priority engagements that facilitates keeping srcs confidential. Would you be willing to file it through the Intel service center? https://supporttickets.intel.com/servicecenter?lang=en-US

Please use the Intel System Studio or Intel Media category buckets if asked.

The portal should ask these two questions:
_Do any of your questions relate to the use of Intel products in a military, defense, intelligence, nuclear/biochemical or space item? [Yes/No]
Will you be providing any technical data that is related to a defense article or any military, defense, intelligence, nuclear/biochemical or space end use? [Yes/No]_

-MichaelC

MichalMrozek, 2018-10-02:
Can you check what is slower using https://github.com/intel/opencl-intercept-layer. ?
You can check whether slowness comes from longer kernel execution or host overhead, there are appropriate toggles to dump various logs. I am particularly interested in logs from DevicePerformanceTiming  and HostPerformanceTiming ,

lqhuy93, 2018-10-04:
Thanks for tips. I'll check again and get back to you guys if I find out the problems.

closed 2018-10-04 by lqhuy93