intel/compute-runtime #97: Can't build on ubuntu 18.04

Sec42, 2018-10-04:
Hi,

as a result of Issue #89 I upgraded to ubuntu 18.04 which has a 4.15.0-34 kernel, but since then I can't even get it to build anymore.

I tried a completely fresh re-checkout, but it always fails at the same spot.

```
[ 26%] Built target copy_compiler_files
Scanning dependencies of target builtins_Gen8core
[ 26%] Generating ../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.bc, ../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.bin, ../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.cpp, ../../../bin/built_ins/x64/gen8/aux_translation_Gen8core.gen
[0]: /home/sec/Projects/IntelV2/build/bin/cloc(_ZN16SafetyGuardLinux9sigActionEiP9siginfo_tPv+0x30) [0x56106fe992d0]
[1]: /lib/x86_64-linux-gnu/libpthread.so.0(+0x12890) [0x7fd705c21890]
[2]: /lib64/ld-linux-x86-64.so.2(+0x16a31) [0x7fd706048a31]
[3]: /lib/x86_64-linux-gnu/libc.so.6(_dl_catch_exception+0x6f) [0x7fd7053e42df]
[4]: /lib/x86_64-linux-gnu/libc.so.6(_dl_catch_error+0x2f) [0x7fd7053e436f]
[5]: /lib/x86_64-linux-gnu/libdl.so.2(+0x1735) [0x7fd705e2f735]
[6]: /lib/x86_64-linux-gnu/libdl.so.2(dlclose+0x33) [0x7fd705e2f0b3]
[7]: /usr/lib/libigdfcl.so(+0x17897) [0x7fd704bf2897]
[8]: /usr/lib/libigdfcl.so(+0x27c40) [0x7fd704c02c40]
[9]: /home/sec/Projects/IntelV2/build/bin/cloc(_ZN5OCLRT15OfflineCompiler15buildSourceCodeEv+0x39d) [0x56106fe8df1d]
[10]: /home/sec/Projects/IntelV2/build/bin/cloc(_ZN5OCLRT15OfflineCompiler5buildEv+0xe) [0x56106fe953de]
[11]: /home/sec/Projects/IntelV2/build/bin/cloc(_ZN16SafetyGuardLinux4callIiN5OCLRT15OfflineCompilerEMS2_FivEEET_PT0_T1_S5_+0x46) [0x56106fe99376]
[12]: /home/sec/Projects/IntelV2/build/bin/cloc(_Z20buildWithSafetyGuardPN5OCLRT15OfflineCompilerE+0x7b) [0x56106fe9927b]
[13]: /home/sec/Projects/IntelV2/build/bin/cloc(main+0x84) [0x56106fe8c494]
[14]: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7fd70529eb97]
[15]: /home/sec/Projects/IntelV2/build/bin/cloc(_start+0x2a) [0x56106fe8c5fa]
Aborted
igdrcl_lib_release/built_ins/kernels/CMakeFiles/builtins_Gen8core.dir/build.make:72: recipe for target 'bin/built_ins/x64/gen8/aux_translation_Gen8core.bc' failed
make[2]: *** [bin/built_ins/x64/gen8/aux_translation_Gen8core.bc] Error 134
CMakeFiles/Makefile2:1214: recipe for target 'igdrcl_lib_release/built_ins/kernels/CMakeFiles/builtins_Gen8core.dir/all' failed
make[1]: *** [igdrcl_lib_release/built_ins/kernels/CMakeFiles/builtins_Gen8core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

I'm a bit at a loss right now.

JacekDanecki, 2018-10-04:
Can you provide output from cmake command?

Sec42, 2018-10-06:
Here you go:
```
karoshi:~/Projects/IntelV2/build>cmake -DBUILD_TYPE=Release -DCMAKE_BUILD_TYPE=Release ../neo
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Release build configuration
-- i915 includes dir: /home/sec/Projects/IntelV2/neo/third_party/uapi
-- Khronos OpenCL headers dir: /home/sec/Projects/IntelV2/neo/third_party/opencl_headers
-- Khronos OpenGL headers dir: /home/sec/Projects/IntelV2/neo/third_party/opengl_headers
-- Third party dir: /home/sec/Projects/IntelV2/third_party
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'igc-opencl'
--   Found igc-opencl, version 0.1.0
-- IGC include dirs: /usr/include/igc;/usr/include/igc/cif;/usr/include/igc/ocl_igc_shared/executable_format;/usr/include/igc/ocl_igc_shared/device_enqueue
-- Checking for module 'igdgmm'
--   No package 'igdgmm' found
-- GmmLib source dir is: /home/sec/Projects/IntelV2/gmmlib
platform: Linux
source_dir: /home/sec/Projects/IntelV2/gmmlib/Source/GmmLib
arch: 64
build type: Release
SourceConfiguration:
CommonDir: /home/sec/Projects/IntelV2/gmmlib/Source/Common
IncDir: /home/sec/Projects/IntelV2/gmmlib/Source/inc
MOCS table: static
-- Computed version major is: 18
-- Computed version minor is: 40
-- All supported platforms:  CNL SKL KBL GLK CFL BXT BDW
-- All tested platforms:  CNL SKL KBL GLK CFL BXT BDW
-- Default supported platform: SKL
-- Default tested platform: SKL
-- Default tested family name: SKLFamily
-- Found ccache: /usr/bin/ccache
-- Checking for module 'libva>=1.0.0'
--   No package 'libva' found
-- GTest repeat count set to 1
-- GTest shuffle set to --gtest_shuffle;--gtest_random_seed=0
-- GTest exception options set to --gtest_catch_exceptions=1
-- GTest filter for mt tests:-AsyncGPUoperations.MapBufferAfterWriteBuffer
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sec/Projects/IntelV2/build
```

JacekDanecki, 2018-10-08:
As I can see you have installed IGC 0.1.0 on your system. Did you compile compatible IGC version?  You can see [releases](https://github.com/intel/compute-runtime/releases) page to check what IGC commit is required by Neo. 
You can also install deb packages from [launchpad repository]( https://launchpad.net/~intel-opencl/+archive/ubuntu/intel-opencl). I've updated them today to the latest release 18.40.11622.
There are also docker images on [docker hub](https://hub.docker.com/r/intelopencl/intel-opencl) with Neo packages and [clinfo](https://github.com/Oblomov/clinfo) already installed under different OS's, so you can use this image to verify whether your kernel and hardware is correctly configured to run Neo.

Sec42, 2018-10-10:
You were right. While trying to debug the previous problem I had run make install on it. I removed all traces of the install, and now it builds fine.
It doesn't run, but I will open another issue for that.

closed 2018-10-10 by Sec42