intel/compute-runtime #99: driver abort()s on Ubuntu 18.04

Sec42, 2018-10-10:
Hi,
I built the driver as per readme, but when installed, `clinfo` dies:

```
Abort was called at 22 line in file:
/home/sec/Projects/IntelV2/neo/runtime/gmm_helper/gmm_interface.cpp
```

The source line is
```
    UNRECOVERABLE_IF(!gmmLib);
```

it seems that the lib is there:
```
karoshi:~>l /usr/local/lib|grep mm
lrwxrwxrwx  1 root root         18 Oct 10 21:37 libigdgmm.so.1 -> libigdgmm.so.1.0.0
-rw-r--r--  1 root root    1744264 Oct  9 22:42 libigdgmm.so.1.0.0
```

JacekDanecki, 2018-10-11:
Did you update gmmlib to this [commit](https://github.com/intel/gmmlib/commit/456b7c0354a40ac2de89f71d7fc7cc47f7de1eee)?

Ruyk, 2018-10-11:
Hello, 

I had the same issue, but updating to the latest gmmlib solved the problem.

Thanks!

Sec42, 2018-10-11:
Thanks for pointing it out. A fresh git pull of the gmmlib repo got it to build (and run). Thanks!

closed 2018-10-11 by Sec42

enihcam, 2018-12-11:
@JacekDanecki I got the same error here, but I had already upgrade gmmlib to 18.4.

```
~ $ clinfo
Abort was called at 22 line in file:
/opt/src/neo/runtime/gmm_helper/gmm_interface.cpp

~ $ pacman -Qs intel-gmmlib 
local/intel-gmmlib 18.4.0-1
    Intel Graphics Memory Management Library

~ $ pacman -Ql intel-gmmlib 
intel-gmmlib /usr/
intel-gmmlib /usr/include/
intel-gmmlib /usr/include/igdgmm/
intel-gmmlib /usr/include/igdgmm/GmmLib/
intel-gmmlib /usr/include/igdgmm/GmmLib/CachePolicy/
intel-gmmlib /usr/include/igdgmm/GmmLib/CachePolicy/GmmCachePolicyConditionals.h
intel-gmmlib /usr/include/igdgmm/GmmLib/CachePolicy/GmmCachePolicyResourceUsageDefinitions.h
intel-gmmlib /usr/include/igdgmm/GmmLib/CachePolicy/GmmCachePolicyUndefineConditionals.h
intel-gmmlib /usr/include/igdgmm/GmmLib/CachePolicy/GmmGen10CachePolicy.h
intel-gmmlib /usr/include/igdgmm/GmmLib/CachePolicy/GmmGen11CachePolicy.h
intel-gmmlib /usr/include/igdgmm/GmmLib/CachePolicy/GmmGen8CachePolicy.h
intel-gmmlib /usr/include/igdgmm/GmmLib/CachePolicy/GmmGen9CachePolicy.h
intel-gmmlib /usr/include/igdgmm/GmmLib/Platform/
intel-gmmlib /usr/include/igdgmm/GmmLib/Platform/GmmPlatforms.h
intel-gmmlib /usr/include/igdgmm/GmmLib/Texture/
intel-gmmlib /usr/include/igdgmm/GmmLib/Texture/GmmTexture.h
intel-gmmlib /usr/include/igdgmm/GmmLib/Utility/
intel-gmmlib /usr/include/igdgmm/GmmLib/Utility/CpuSwizzleBlt/
intel-gmmlib /usr/include/igdgmm/GmmLib/Utility/CpuSwizzleBlt/CpuSwizzleBlt.c
intel-gmmlib /usr/include/igdgmm/GmmLib/Utility/CpuSwizzleBlt/assert.h
intel-gmmlib /usr/include/igdgmm/GmmLib/Utility/GmmLog/
intel-gmmlib /usr/include/igdgmm/GmmLib/Utility/GmmLog/GmmLog.h
intel-gmmlib /usr/include/igdgmm/GmmLib/Utility/GmmUtility.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/CachePolicy/
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/CachePolicy/GmmCachePolicyGen10.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/CachePolicy/GmmCachePolicyGen11.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/CachePolicy/GmmCachePolicyGen8.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/CachePolicy/GmmCachePolicyGen9.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmCachePolicy.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmCachePolicyCommon.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmCachePolicyExt.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmClientContext.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmCommonExt.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmConst.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmDebug.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmFormatTable.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmGttExt.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfo.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmInfoExt.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmInternal.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmLibDll.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmLibDllName.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmMemAllocator.hpp
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmPlatformExt.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmProto.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmResourceFlags.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmResourceInfo.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmResourceInfoCommon.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmResourceInfoExt.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmTextureExt.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Common/GmmUtil.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Linux/
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/External/Linux/GmmResourceInfoLin.h
intel-gmmlib /usr/include/igdgmm/GmmLib/inc/GmmLib.h
intel-gmmlib /usr/include/igdgmm/igdgmm.h
intel-gmmlib /usr/include/igdgmm/inc/
intel-gmmlib /usr/include/igdgmm/inc/common/
intel-gmmlib /usr/include/igdgmm/inc/common/gfxmacro.h
intel-gmmlib /usr/include/igdgmm/inc/common/gfxplatform.h
intel-gmmlib /usr/include/igdgmm/inc/common/gtsysinfo.h
intel-gmmlib /usr/include/igdgmm/inc/common/igfxfmid.h
intel-gmmlib /usr/include/igdgmm/inc/common/sku_wa.h
intel-gmmlib /usr/include/igdgmm/inc/portable_compiler.h
intel-gmmlib /usr/include/igdgmm/inc/umKmInc/
intel-gmmlib /usr/include/igdgmm/inc/umKmInc/UmKmDmaPerfTimer.h
intel-gmmlib /usr/include/igdgmm/inc/umKmInc/UmKmEnum.h
intel-gmmlib /usr/include/igdgmm/inc/umKmInc/sharedata.h
intel-gmmlib /usr/include/igdgmm/util/
intel-gmmlib /usr/include/igdgmm/util/g_gfxDebug.h
intel-gmmlib /usr/include/igdgmm/util/gfxDebug.h
intel-gmmlib /usr/lib/
intel-gmmlib /usr/lib/libgmm_umd.a
intel-gmmlib /usr/lib/libigdgmm.so
intel-gmmlib /usr/lib/libigdgmm.so.5
intel-gmmlib /usr/lib/libigdgmm.so.5.0.0
intel-gmmlib /usr/lib/pkgconfig/
intel-gmmlib /usr/lib/pkgconfig/igdgmm.pc
intel-gmmlib /usr/share/
intel-gmmlib /usr/share/licenses/
intel-gmmlib /usr/share/licenses/intel-gmmlib/
intel-gmmlib /usr/share/licenses/intel-gmmlib/LICENSE.md

```

alalek, 2018-12-11:
You should re-build OpenCL runtime yourself (or via AUR) with new dependencies.
Old existed OpenCL package expects old libigdgmm.so library (something like `libigdgmm.so.3`). You can check list of loaded libraries via `LD_DEBUG=files clinfo`.

enihcam, 2018-12-11:
Then please remove the digit from the filename. Thanks.