intel/compute-runtime #105: Latest Runtime and Fine Grain SVM

dgerzhoy, 2018-11-01:
Hello,

I've been using the depreciated closed-source release of the OpenCL SDK/Runtime for a while now (I have the latest version of that).

Have been trying to use vtune-amp to do some performance analysis and it pointed out that I don't have the latest runtime installed. Does it expect to have your version of the runtime?

I'd love to switch to the latest and greatest runtime, but I rely pretty heavily on CL_MEM_SVM_FINE_GRAIN_BUFFER and CL_MEM_SVM_ATOMICS for my SVM buffers.

I saw that its not supported in this release. Any idea when that might be added? 

Thanks

Daniel Gerzhoy

BartoszDunajski, 2018-11-06:
Hello Daniel,
Unfortunately, driver can't report SVM capabilities because of missing support in Linux kernel.
Appropriate patch is already submitted, but not yet up-streamed.

You can always add required patch manually: https://patchwork.kernel.org/patch/10638787/

Additional changes in OCL driver are also required: https://github.com/intel/compute-runtime/pull/53

dgerzhoy, 2018-12-05:
Thanks! Appreciate the guidance.

You wouldn't by any chance know of a way to flush or invalidate GPU Cache lines. Either from the the runtime api or from the kernel itself.

I'm finding that when data is in the GPU cache, the overhead of transferring it is pretty high (which makes sense from the standpoint of tracking coherence in the GPU- just like the patch documentation mentions)

If I could flush the data from the GPU Kernel this might fix the problem.

Thanks!

BartoszDunajski, 2018-12-06:
Could you give an example of your use case and more details on your observations?

dgerzhoy, 2018-12-17:
I'm running what amounts to a virtual machine or daemon on the GPU that persists after each kernel launch. Like this: https://www.iwocl.org/wp-content/uploads/iwocl-2016-gpu-daemon.pdf

I need this because I have kernels whose runtime would otherwise be overpowered by the GPU kernel launch overhead.

I have a set of arrays all allocated with clSVMAlloc and CL_MEM_SVM_FINE_GRAIN_BUFFER. One is Read Only the other 2 are RW. (I’ve done them all as RW and it doesn’t affect performance)

The code is 2 loops and I have 2 versions. 
Loop 1 runs on the GPU (as a kernel) or the CPU depending on the version.
Loop 2 always runs on the CPU. 
Loop 1 (GPU|CPU) works on the 3 arrays, then the Loop 2 (CPU) operates on them. This happens several thousand times.

In the case where both run on the CPU, the data footprint of the arrays is small enough to fit in the L1 cache, so there is virtually no data transfer overhead. 

With the first half on GPU, data needs to be transferred to the CPU cache, so it must first be invalidated in the GPU cache. Which occurs when the CPU tries to access the data. 
I've noticed that if I flush the cache lines on the CPU side before loop 2, loop 2 takes a lot less time than if I let it get the data from the GPU global cache, making me think that invalidation from the GPU cache is very expensive.

I’m wondering if there is a way to invalidate the cache lines on the GPU side when the kernel is done with the data? Or in a non-blocking fashion from the runtime?

zzdanowicz, 2018-12-18:
Hi,
I'm a co-author of the GPU Daemon paper. For Daemon to work we need to have SVM Fine Grain support. Without it, there will be no data exchange possible between CPU and GPU, while there are live GPU threads.
Also if you run two separate GPU contexts (like one daemon and one regular GPU workload) you might run into problems of rescheduling and preemption.

As for invalidation of the GPU side, I think you could use clEnqueueMarker/clWaitForEvents, but please note this command will not execute until GPU threads (like Daemon) are still in flight.

You could experiment with Daemon concepts on Windows, where Fine Grain SVM is supported.
As for Linux, you need to have a i915 KMD support, which is beyond control of OpenCL UMD driver.
Please use this patch: https://patchwork.kernel.org/patch/10638787/
And then apply OpenCL patch from #53 as Bartosz explained here: https://github.com/intel/compute-runtime/issues/105#issuecomment-436240899

zzdanowicz, 2018-12-18:
Hi Daniel,

Let me know, if you are interested in all different modes of GPU Daemon. Do you use "Instant" or "Enqueue" mode?

regards,
Zbigniew

dgerzhoy, 2018-12-24:
Hi Zbigniew,

Thanks for getting back to me, sorry for the delay responding back- I must not have notifications turned on. 

I am using the older OpenCL SDK/Runtime and applied the patch that the documentation mentioned to get Fine Grain SVM working (and they do work). So far I haven't switched to the new SDK/Runtime

I am using instant mode, so my Daemon persists for the duration of the program I am running on all threads, which is why I haven't tried using any event based synchronization. Enqueue mode launch latencies were too large for my purposes.

On another note: I don't have any other GPU contexts, but I think I am still getting preemtion issues for instances where the Daemon Persists for a long time. I've found a workaround: restarting it every so often. Not certain whats causing it, but I assumed it was the OS kicking me off the GPU so it can render.

zzdanowicz, 2019-01-03:
Hi Daniel,

In case of Instant mode, there is no other way synchronize data between CPU and GPU, but to use Fine Grain SVM (cache coherent memory).
OS GPU watchdog might kick-in for various reasons and preempt GPU context (for example it can detect timeout condition) - I wonder how often it happens and what is the behavior (context termination or delay on specific daemon operation?) and does your OS support preemption?

regards,
Zbigniew

zzdanowicz, 2019-01-03:
@JacekDanecki - do you know how to manually setup Linux GPU timeouts and/or preemption timings? Eventually how to disable timeouts?

dgerzhoy, 2019-01-03:
Using Ubuntu 16.04, if that helps. Will try to find out details on preemption with this release.

JacekDanecki, 2019-01-10:
If you want to change GPU watchdog timeout, you need to change i915 driver source, see [DRM_I915_HANGCHECK_PERIOD](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/gpu/drm/i915/i915_gpu_error.h#n200), recompile i915 module, and install it.
To disable GPU watchdog you can run as root below command:
```
echo N > /sys/module/i915/parameters/enable_hangcheck
```

dgerzhoy, 2019-01-10:
Thanks Jacek, that was helpful. Changing the source is a little beyond me right now, but I disabled the watchdog successfully.

I'm now running into the problem that when my program does not exit cleanly, the kernel on the GPU continues to run. To kill it I've been turning the watchdog timer back on, waiting, and then turning it back off. Is there a better way to kill it?

JacekDanecki, 2019-01-14:
To manually reset GPU you can use below commands as root:
```
cd /sys/kernel/debug/dri/0
echo 0 > i915_error_state
echo 1 > i915_wedged
```

BartoszDunajski, 2019-02-04:
@dgerzhoy can we close the issue?

dgerzhoy, 2019-02-04:
Yes! Sorry. 

Thanks for the reply @JacekDanecki  and thanks everyone for the help.

tarlovsky, 2019-02-07:
Good afternoon. Which version of the compute-runtime do I apply the patch from #53 to? 
Neo is currently at version 19.04 but I do not see how that is possible. Ty

BartoszDunajski, 2019-02-07:
PR is based on e47344ec0e42f48a0f914f62ea8a3d79cce10943

closed 2019-02-07 by MichalMrozek