intel/compute-runtime #106: intel-opencl*.deb installation problem

rollerpunk, 2018-11-04:
Hi,

I''m trying to install openCL by following step-by-step with all linked dependencies:
https://github.com/intel/compute-runtime/blob/master/documentation/BUILD_Ubuntu.md

I've compiled everything successfully but got error at the last step (*.deb installation):

>  **$ sudo dpkg -i intel-opencl_18.44-0_amd64-opencl.deb**
> (Reading database ... 246075 files and directories currently installed.)
> Preparing to unpack intel-opencl_18.44-0_amd64-opencl.deb ...
> Unpacking intel-opencl (18.44.0) over (18.44.0) ...
> dpkg: dependency problems prevent configuration of intel-opencl:
> intel-opencl depends on intel-gmmlib; however:
>  Package **intel-gmmlib** is not installed.
> intel-opencl depends on intel-igc-opencl; however:
>  Package **intel-igc-opencl** is not installed.
>
>dpkg: error processing package intel-opencl (--install):
> dependency problems - leaving unconfigured
>Processing triggers for libc-bin (2.23-0ubuntu9) ...
>Errors were encountered while processing:
> intel-opencl

Looks like gmmlib and igc packages need to be propagated somehow.
Please help me to make thing right.
Thank you in advance !

<hr>

Here is packing result:
>**$ make -j`nproc` package**
> :
> :
> [100%] Built target run_unit_tests
> Run CPack packaging tool...
> CPack: Create package using DEB
> CPack: Install projects
> CPack: - Run preinstall target for: igdrcl
> CPack: - Install project: igdrcl
> CPack: -   Install component: opencl
> CPack: Create package
> CPackDeb: - Generating dependency list
> CPack: - package: ~/intelopenCL/neo/build/intel-opencl_18.44-0_amd64-opencl.deb generated.

workspace structure:
> **~/intelopenCL $ ls -la**
>total 52
>drwxrwxr-x 13 z z 4096 Nov  4 10:52 .
>drwxr-xr-x 11 z z 4096 Nov  4 11:27 ..
>drwxrwxr-x 16 z z 4096 Nov  4 01:28 build
>drwxrwxr-x 16 z z 4096 Nov  4 01:05 clang_source
>drwxrwxr-x  6 z z 4096 Nov  4 01:06 common_clang
>drwxrwxr-x  6 z z 4096 Nov  4 10:53 gmmlib
>drwxrwxr-x  9 z z 4096 Nov  4 01:28 igc
>drwxrwxr-x 24 z z 4096 Nov  4 10:31 libdrm
>drwxrwxr-x  9 z z 4096 Nov  4 10:32 libva
>drwxrwxr-x  4 z z 4096 Nov  4 01:06 llvm_patches
>drwxrwxr-x 16 z z 4096 Nov  4 01:25 llvm_source
>drwxrwxr-x 14 z z 4096 Nov  4 10:21 neo
>drwxrwxr-x  4 z z 4096 Nov  4 01:27 opencl_headers

>**~/intelopenCL/neo/build $ ls -l**
>total 3076
>drwxrwxr-x 16 z z    4096 Nov  4 13:51 .
>drwxrwxr-x 14 z z    4096 Nov  4 10:21 ..
>-rw-rw-r--  1 z z   61873 Nov  4 10:55 CMakeCache.txt
>drwxrwxr-x  4 z z    4096 Nov  4 13:51 CMakeFiles
>-rw-r--r--  1 z z    4629 Nov  4 10:55 CPackConfig.cmake
>-rw-r--r--  1 z z    5060 Nov  4 10:55 CPackSourceConfig.cmake
>-rw-rw-r--  1 z z  184466 Nov  4 10:55 Makefile
>drwxrwxr-x  3 z z    4096 Nov  4 13:50 _CPack_Packages
>drwxrwxr-x 22 z z    4096 Nov  4 13:51 bin
>-rw-rw-r--  1 z z    5338 Nov  4 10:55 cmake_install.cmake
>-rw-rw-r--  1 z z 1791453 Nov  4 10:55 compile_commands.json
>-rw-rw-r--  1 z z     353 Nov  4 10:55 config.h
>-rw-rw-r--  1 z z     203 Nov  4 10:55 driver_version.h
>drwxrwxr-x  3 z z    4096 Nov  4 10:55 elf
>drwxrwxr-x  4 z z    4096 Nov  4 10:55 gmmlib
>drwxrwxr-x  8 z z    4096 Nov  4 13:14 igc
>-rw-rw-r--  1 z z     245 Nov  4 10:55 igc.opencl.h
>-rw-rw-r--  1 z z    1217 Nov  4 10:55 igdgmm.h
>-rw-rw-r--  1 z z     367 Nov  4 10:55 igdgmm.pc
>drwxrwxr-x 38 z z    4096 Nov  4 10:55 igdrcl_lib_mockable
>drwxrwxr-x 40 z z    4096 Nov  4 10:55 igdrcl_lib_release
>-rw-rw-r--  1 z z      95 Nov  4 13:51 install_manifest_opencl.txt
>-rw-rw-r--  1 z z  963508 Nov  4 13:51 intel-opencl_18.44-0_amd64-opencl.deb
>-rw-rw-r--  1 z z      28 Nov  4 13:51 intel.icd
>drwxrwxr-x  2 z z    4096 Nov  4 13:21 lib
>-rw-rw-r--  1 z z     147 Nov  4 10:55 lib_names.h
>-rw-rw-r--  1 z z      15 Nov  4 13:51 libintelopencl.conf
>drwxrwxr-x  4 z z    4096 Nov  4 10:55 offline_compiler
>-rwxr-xr-x  1 z z      15 Nov  4 13:51 postinst
>-rwxr-xr-x  1 z z      15 Nov  4 13:51 postrm
>drwxrwxr-x  3 z z    4096 Nov  4 10:55 scripts
>drwxrwxr-x  3 z z    4096 Nov  4 10:55 third_party
>drwxrwxr-x  2 z z    4096 Nov  4 13:51 tmp
>drwxrwxr-x 56 z z    4096 Nov  4 10:55 unit_tests

**$ sudo lshw -C display -C CPU**
>  *-cpu                   
>       description: CPU
>       product: Intel(R) Core(TM) i3-4005U CPU @ 1.70GHz
>       vendor: Intel Corp.
>       physical id: 42
>       bus info: cpu@0
>       version: Intel(R) Core(TM) i3-4005U CPU @ 1.70GHz
>       slot: SOCKET 0
>       size: 1599MHz
>       capacity: 1700MHz
>       width: 64 bits
>       clock: 100MHz
>       capabilities: x86-64 fpu fpu_exception wp vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm epb tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid xsaveopt dtherm arat pln pts cpufreq
>       configuration: cores=2 enabledcores=2 threads=4
>  *-display
>       description: VGA compatible controller
>       product: Haswell-ULT Integrated Graphics Controller
>       vendor: Intel Corporation
>       physical id: 2
>       bus info: pci@0000:00:02.0
>       version: 0b
>       width: 64 bits
>       clock: 33MHz
>       capabilities: msi pm vga_controller bus_master cap_list rom
>       configuration: driver=i915 latency=0
>       resources: irq:43 memory:f7800000-f7bfffff memory:e0000000-efffffff ioport:f000(size=64) memory:c0000-dffff

TaihuLight, 2018-11-10:
I can successfully install all *deb for intel compute runtime which are download from release region of this repo, but when I run the command clinfo, it can not find Intel GPU device. 
OS: Ubuntu 16.04.3
CPU: Intel i7-8700k
How can I do to successfully use the released *deb on my computer?

jdanecki, 2018-11-13:
@rollerpunk Were you able to  install  Neo correctly?