intel/compute-runtime #113: Out of order queue causes gpu driver reset

ChrisLundquist, 2018-11-14:
Device info:
```
nuc6i7kyk: i7 6770hq + Intel® Iris™ Pro Graphics 580 (Gen9)

ubuntu 18.04
Linux nuc 4.15.0-36-generic #39-Ubuntu SMP Mon Sep 24 16:19:09 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

Number of devices                                 1
  Device Name                                     Intel(R) Gen9 HD Graphics NEO
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 2.1 NEO 
  Driver Version                                  18.45.11804
```

When creating a command queue like:
```
 // Create a command commands
 const cl_queue_properties props[] = {CL_QUEUE_PROPERTIES, CL_QUEUE_OUT_OF_ORDER_EXEC_MODE_ENABLE, 0};
 engine->commands =
     clCreateCommandQueueWithProperties(engine->context, engine->device_id, &props, &err);
```

and then enqueueing many instances of the same kernel:
```
for (size_t i = step_size * 0; i < global; i += step_size) {
  /* TODO readback chunks of results after a kernel finishes */
  printf(
      "enqueue kernel global: %ld, step_size: %ld, local: %ld, offset: %ld\n",
      global, step_size, local, i);
  err = clEnqueueNDRangeKernel(codec->engine->commands, codec->kernel, 1, &i,
                               &step_size, &local, 0, NULL, NULL);
  if (err) {
    printf("Error: Failed to execute kernel! Error: %d\n", err);
    return -5;
  }
}
clEnqueueBarrier(codec->engine->commands);
```

I see the following in /var/log/syslog:
```
2671 Nov 14 13:31:06 nuc kernel: [3246406.763310] i915 0000:00:02.0: Resetting rcs0 after gpu hang
```

*When using IN ORDER QUEUES* This happens when the watchdog timer (or something like it) times out because a kernel is taking too long to finish.
This is why I have to enqueue many smaller work items when working with in order queues.

However the same work sizes or smaller fail when using out-of-order queues.
I've even tried emulating an in-order queue by enqueueing a barrier after each kernel (kernel, barrier, kernel, barrier) and the linux kernel still resets the gpu / i915 driver.

Out-of-order queues appear to work correctly when using the CPU OpenCL device (as the kernel doesn't watch dog user processes).

Is this a bug? Is there an error in my code? 
What is the best way to use out-of-order queues with Gen9?

bfliflet, 2018-11-14:
Your code looks reasonable and we encourage usage of OOQs.  How many enqueues are you doing in the loop?  Does this fail if you insert a clFlush inside the loop?  With OOQs, the driver attempts to batch up as many NDRangeKernels to maximize GPU utilization prior to submitting to i915.  I suspect the timeout mechanism is per-submission.  Various API such as clFlush and clFinish among others force a submission.  Let me know what happens when you try to break your submissions into smaller chunks.

ChrisLundquist, 2018-11-14:
The exact source is https://github.com/ChrisLundquist/libpz/blob/master/opencl/engine.c#L193-L210

> How many enqueues are you doing in the loop? 
A few hundred to a few thousand. 
I make a work item for each byte in a file. The file is ~4MB. When In-Order I get the best performance around MAX_WORK_GROUP_SIZE * 32 and MWGS is 256. In that case I have ~500 enqueues.
(For some reason the Video Front End (VFE) is the best indicator of work being done via `intel_gpu_top` as it relates to copying bytes in and out VS Command Stream which is usually just 100% or 0%)

I tried using 256 work group sizes with the OOQ, which would be a few thousand enqueues (4MB/256).

>  I suspect the timeout mechanism is per-submission. 
That would make sense!
In the IIQ case sometimes the enqueue will block the host thread. in the OOQ case the host thread doesn't block suggesting batches which are too big. (I hit this in the IIQ case which caused the need to break the kernels into different ranges)

> Various API such as clFlush and clFinish among others force a submission.
Yep! I added a `clFlush` every 8 items with an OOQ and I see a ~30% performance gain.
Thank you!

It is worth noting that
https://software.intel.com/en-us/articles/opencl-out-of-order-queue-on-intel-processor-graphics
says:
> Note: For each stream of commands avoid flushing or blocking operations such as clFlush, clFinish, clWaitForEvents or blocking enqueue commands and manage dependencies with event waitlists.

Which makes sense as you want the largest OoO window possible. However, if the gpu resets, it doesn't matter much.

closed 2018-11-14 by ChrisLundquist

bfliflet, 2018-11-15:
I spoke with an i915 expert and he mentions that reset happens after 12s.  A timeout fires after 4s and if no forward progress is detected it gets nuked.  If forward progress is detected, it gets 2 more chances and then it's nuked anyway.  This timeout is per submission.  You could check if your reset is related to this by setting '**i915.enable_hangcheck=0**'.  If you find the hang still happens, there might be a bug here.  Btw, keep exploiting OOQs, they'll only get better!

ChrisLundquist, 2018-11-15:
Thank you for the pointers forward, you really helped me out!

> This timeout is per submission.

Yeah, I think by default, OOQ batches all or close to all of them into one submission (which then times out). For this particular file / workload / kernel / clock speed. I can flush the EnqueueNDRangeKernel calls in batches of 32 or 64 without error. batches of 128 hit the timeout ( at least when the GPU is downclocked via `intel_gpu_frequency -e`)

Without the flush calls and OOQ, I see the render space is 56/max. This is a magic number that happens when single group is flushed. (I think GEN9 has 7 threads per slice, and I have 8 slices? 7 * 8 = 56 or something along those lines.) 

The workload takes ~60 seconds at 350mhz so it times out when done as a single target.

So I guess what I'm getting at:
- The default OOQ behavior seems a little optimistic at the flush interval. It seems like it should take some hinting if I'm purposely chunking the workload into hundreds of calls? Detecting that hint is understandably tricky on the driver side.

- Is there a way for my app to recover when the gpu resets? Usually it hangs waiting for the kernels to finish, but they never will. Prevention is only strategy I can see. (by making a queue that has profiling, doing a test run, and finding out how long that takes) Sadly at the moment my worst case performance is pretty far from my normal/best case.

bfliflet, 2018-11-15:
> It seems like it should take some hinting if I'm purposely chunking the workload into hundreds of calls? Detecting that hint is understandably tricky on the driver side.

It is tricky.  An application can write a single looping kernel that takes more than 12s.  An application can also send 1024 NDRangeKernels that in aggregate only take 1ms.  Having said that, the motivation for aggregation was to amortize the cost of the cache flushes added by i915 around our submissions.  There is diminishing return and I'll push the team to implicitly flush after a reasonable number.  It will handle most but not all cases because of the difficulty in estimating the duration of a single submission.

> Is there a way for my app to recover when the gpu resets?

You're solution, a profiling run of sorts, is what I'd recommend as well.  Most production applications define a list of supported hardware and establish a content budget to adhere to.  Anyone outside that window is not supported.