intel/compute-runtime #121: Can't find igc-opencl with default configuration

chivakker, 2018-12-12:
Can compute-runtime operate without needing pkg-config files 

If so, can this option be disabled by default and/or add an option for it?

cmake -DCMAKE_INSTALL_PREFIX=/<my-prefi>x

Checking for module 'igc-opencl'
--   Package 'igc-opencl', required by 'virtual:world', not found

ArturHarasimiuk, 2018-12-13:
Hi @chivakker 

Could you please share more details about your ask? why is pkg-config problematic

chivakker, 2018-12-13:
I would like to use compute-runtime without igc-opencl-devel package.  Is that possible?  Maybe some features won't be available but that is fine.  I suggest moving those features inside a configuration i.e.

cmake -DVIRTUAL_WORLD=OFF

and have something that can be used.

ArturHarasimiuk, 2018-12-14:
igc-opencl-devel is build time dependency. this package contains header files required to build compute-runtime.

ArturHarasimiuk, 2019-01-11:
closing due to no activity. feel free to open new issue if anything appear.

closed 2019-01-11 by ArturHarasimiuk