intel/compute-runtime #122: Coexistence problems between LLVM versions

awherr, 2018-12-14:
There is a bug in the Neo driver that prevents the OpenCL runtime from working under application/languages that link to different versions of LLVM than the version support by the OpenCL compiler. This bug shows up in domain-specific languages like Halide (http://halide-lang.org) and TVM (https://github.com/dmlc/tvm) that use LLVM for Jitting, as well as tools like OCLGrind (https://github.com/jrprice/Oclgrind). It’s possible that there are co-existence problems where multiple OpenCL vendors have conflicting LLVM versions under the hood, but this has not been observed. 

Reproducing the problem:
Install the latest Neo driver, download Halide and run the GPU sample from here: http://halide-lang.org/tutorials/tutorial_lesson_12_using_the_gpu.html using an OpenCL target.

You will see an error similar to this: 
CommandLine Error: Option 'disable-inlined-alloca-merging' registered more than once!

Root Cause: 
Halide is built against LLVM 6 and IGC uses LLVM 4. LLVM uses static initializers to register options. When Halide is linked it execution the static initializers in LLVM6 and when Neo->IGC->LLVM4 are dynamically loaded the LLVM4 static initializers are called causing the above error. Commenting out the error trap still results in runtime errors, which is likely caused by symbol preempted from the later version of LLVM (e.g. same symbol names but with LLVM6 bits instead of LLVM4 bits). 

Possible fix:
	Both the OpenCL and IGC are late bound using dlopen. Adding the RTLD_DEEPBIND flag to the dlopen calls in Neo (which load IGC) and to IGC (where it loads LLVM) should fix the problem. The deep bind scopes symbol resolutions to the local module, separating the LLVM versions and initializers. I’m not sure if there are any problematic side-effects to a deep bind.

alalek, 2018-12-15:
I can confirm this problem.

In OpenCV we using both OpenCL and Halide (which can use OpenCL too) and observing crashes caused by LLVM symbols mess.
Some our improperly configured [builds](http://pullrequest.opencv.org/buildbot/builders/3_4_halide-skl-lin64/builds/109) crash with messages like this:

> halide/llvm/lib/IR/Value.cpp:169: llvm::ValueName* llvm::Value::getValueName() const: Assertion `I != Ctx.pImpl->ValueNames.end() && "No name entry found!"' failed.

This weird behavior can be "fixed" (workarounded) from the other side too.
We using this scheme to avoid mess of Halide's LLVM symbols:
- build Halide's LLVM as (1) static library with (2) hidden visibility
- build (3) shared libHalide.so with static LLVM binaries using (4) `-Bsymbolic` or (5) `-Wl,--exclude-libs,ALL` (or replace "ALL" to LLVM .a libs)

Using this scheme, the most part of LLVM symbols should gone from symbol tables of Halide shared library.
Use `LD_DEBUG=files,symbols <your app>`, "nm -D", "nm -g", "objdump" to confirm that symbols mess is gone.

---

This is Linux specific behavior. On Windows there is no such global lookup table with symbols - symbols are bound to .DLLs on Windows platform.
There is document with description of Linux shared libraries loader: [dsohowto.pdf](https://software.intel.com/sites/default/files/m/a/1/e/dsohowto.pdf) (interested section is 1.5.4).

IMHO, good practice is hiding or renaming of own/customized symbols that may cause mess/conflict with symbols from system packages.

dliang0406, 2019-03-08:
I met the same issue with TVM. Is there any ETA for the fix?

JacekDanecki, 2019-03-08:
This issue is on my todo list. You can find quick fix in my [fork](https://github.com/JacekDanecki/compute-runtime/commit/42a47ab539362b87f26862e3ac21b18676d463d5). With such change halide lesson 12 passed
```
/home/ocl_dev/Halide/bin/tutorial_lesson_12_using_the_gpu              
Running pipeline on CPU:
Running pipeline on GPU:
Target: x86-64-linux-avx-avx2-f16c-fma-opencl-sse41
Testing GPU correctness:
Testing performance on CPU:
0.8554 milliseconds
Testing performance on GPU:
1.1793 milliseconds
```

JacekDanecki, 2019-04-12:
Fixed with this commit: 22be2f677d500641d46cf602732a19335dd35ec8
Closing this issue, if you see new problem, please reopen it, or create new issue.

closed 2019-04-12 by JacekDanecki

Glavnokoman, 2019-04-24:
`==186==You are trying to dlopen a libopencl_clang.so shared library with RTLD_DEEPBIND flag which is incompatibe with sanitizer runtime (see https://github.com/google/sanitizers/issues/611 for details). If you want to run libopencl_clang.so library under sanitizers please remove RTLD_DEEPBIND from dlopen flags`
Is there a way to load this lib now in a project using sanitizers? I failed to pick a suppression rule fixing this.

alalek, 2019-04-24:
Take a look on `USE_ASAN` build option.

Glavnokoman, 2019-04-24:
Aha, so no other way apart from rebuilding. Got it, thanks!